strongtz/i915-sriov-dkms #214: dkms status: i915-sriov-dkms/6.8.12-2, 6.8.12-2-pve, x86_64: installed，but not ok

cffycls, 2024-10-25:
the project Readme.md file show test support 6.8.12-1-pve, and my local version is 6.8.12-2-pve
`:~# dkms status
i915-sriov-dkms/6.8.12-2, 6.8.12-2-pve, x86_64: installed
:~# pveversion
pve-manager/8.2.7/3e0176e6bb2ade3b (running kernel: 6.8.12-2-pve)`

it looks like install success.
`:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
# rebooted, cpu is intel j4215
:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation GeminiLake [UHD Graphics 600] (rev 06)`

uhd list only one.

admarty, 2024-10-26:
I don't think your iGPU (GeminiLake UHD Graphics 600) supports SR-IOV.

ChaimHong, 2024-10-26:
> I don't think your iGPU (GeminiLake UHD Graphics 600) supports SR-IOV.

https://www.intel.com/content/www/us/en/products/sku/128983/intel-celeron-processor-n4100-4m-cache-up-to-2-40-ghz/specifications.html

kaiseu, 2024-10-26:
root@pve:~# lspci |grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)

I also had that problem with 6.8.12-2.

tried proxmox-boot-tool kernel pin 6.8.12-1-pve, the pve failed to boot...

cffycls, 2024-10-29:
> I also had that problem with 6.8.12-2. I had to pin 6.8.12-1 and everything worked.
> 
> proxmox-boot-tool kernel pin 6.8.12-1-pve

Your cpu is also j4215 ? how to do this. 
I tried  6.8.12-1-pve, but system start show is  6.8.12-2-pve too.

pasbec, 2024-10-29:
With J4215 you shoud check for a GVT-g guide, not SR-IOV!

yduanBioinfo, 2024-10-29:
My host version is 6.8.12-2-pve, and my CPU is intel N100. I suppose I made it.
```
:~# dkms status
i915-sriov-dkms/2024.09.21, 6.8.12-2-pve, x86_64: installed
:~# pveversion 
pve-manager/8.2.7/3e0176e6bb2ade3b (running kernel: 6.8.12-2-pve)
:~# cat /etc/sysfs.conf 
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3

:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]

:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=3
[    0.026768] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=3
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[    5.629168] i915: module verification failed: signature and/or required key missing - tainting kernel
[    6.345235] i915 0000:00:02.0: Running in SR-IOV PF mode
......
......
[    7.878545] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    7.878901] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    7.878904] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[    7.878907] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[    7.879000] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    7.879023] i915 0000:00:02.3: Running in SR-IOV VF mode
[    7.879321] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    7.879359] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    7.879787] i915 0000:00:02.3: [drm] VT-d active for gfx access
[    7.879824] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[    7.880355] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    7.880393] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    7.880724] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    7.880728] i915 0000:00:02.3: HuC firmware PRELOADED
[    7.881007] i915 0000:00:02.3: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[    7.881009] i915 0000:00:02.3: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[    7.882921] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.882929] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[    7.883023] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[    7.883240] i915 0000:00:02.0: Enabled 3 VFs
```

By the way, I failed at the first try because I hadn't run "update-grub".

pasbec, 2024-10-29:
You could get rid of the firmware mismatch but the mmio remains. But the VFs seem to work despite the error messages. Just ignore the errors. Got the same output on my N100 and VFs are working fine with 3D and video accelleration in Windows RDP. My N100 host is up and running for 60 days without issues.

purefalse, 2024-10-30:
I encountered the same problem
But I want to share the graphics card function to OMV hard decoding HEVC
But when starting OMV i915 crashes on boot

Video card captured in OMV
But there is no such thing in /dev/dri

Is there any solution to this?

pasbec, 2024-10-30:
Have you installed the drivers in omv aswell?

yduanBioinfo, 2024-10-31:
> I encountered the same problem But I want to share the graphics card function to OMV VM hard decoding HEVC But when starting OMV VM i915 crashes on boot
> 
> Video card captured in OMV But there is no such thing in /dev/dri
> 
> Is there any solution to this?

Actually, I have the same demand as you. However, I'm still trying to make VF work in  OMV VM guest. Now I'm stuck in the "make deb-pkg" step, It's hard.

purefalse, 2024-10-31:
> Actually, I have the same demand as you. However, I'm still trying to make VF work in OMV VM guest. Now I'm stuck in the "make deb-pkg" step, It's hard.

I never thought about installing the driver Because I directly pass through the graphics card omv it can be caught but i will try i just think omv It should also be able to capture my sliced ​​graphics card. But it seems it can't And although the slicing is successful in pve But i915 reports that it has an error But I saw that everyone in win10 can directly ignore the error and output successfully. That's why I want to ask if I did something wrong somewhere

yduanBioinfo, 2024-10-31:
@purefalse According to the README of this Driver, Linux platform geust should also install this driver to enable VF pass through. While on the Windows platform the driver can be installed smoothly since it have already supported the intel VF driver (I guess).

Zackptg5, 2024-10-31:
I got it working on 6.8.12-2-pve but I had to essentially reinstall i915 kernel module:
```
rm -rf /usr/src/i915-sriov-dkms-*
rm -rf /var/lib/dkms/i915-sriov-dkms
find /lib/modules -regex ".*/updates/dkms/i915.ko" -delete
cd ~/i915-sriov-dkms
git pull
dkms add .
dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
update-initramfs -u -k all
```

yduanBioinfo, 2024-11-01:
@purefalse I built an Ubuntu VM server on my PVE  host. After installing i915 driver inside the VM machine, the VF worked, and hardware transcoding was successfully enabled.

purefalse, 2024-11-02:
[image]

I't OK.
But not working.
```
~#  lspci -nn | grep -Ei "3d|display|vga"
00:10.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [UHD Graphics] [8086:46d1]
``` 

```
~# ls /dev/dri
ls: cannot access '/dev/dri': No such file or directory

``` 

I use (intel-media-va-driver-non-free) and (intel / media-driver) 
reboot 

ls /dev/dri  Neither

purefalse, 2024-11-02:
Later I installed the OMV plug-in
Replace PVE core
After installing the i915 driver it seems OK
To be tested

purefalse, 2024-11-02:
Although /dev/dir is read
But there is no function
I don’t know if there is any solution

```
OMV:~# ls /dev/dri
by-path  card0  renderD128

``` 

```
OMV ~# sudo dmesg | grep i915
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[   12.052762] i915: loading out-of-tree module taints kernel.
[   12.052814] i915: module verification failed: signature and/or required key missing - tainting kernel
[   12.420743] i915 0000:00:10.0: Running in SR-IOV VF mode
[   12.421357] i915 0000:00:10.0: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   12.421403] i915 0000:00:10.0: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   12.422086] i915 0000:00:10.0: vgaarb: deactivate vga console
[   12.422120] i915 0000:00:10.0: [drm] Using Transparent Hugepages
[   12.423790] i915 0000:00:10.0: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   12.423834] i915 0000:00:10.0: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   12.424230] i915 0000:00:10.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   12.424232] i915 0000:00:10.0: HuC firmware PRELOADED
[   12.425294] i915 0000:00:10.0: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   12.425300] i915 0000:00:10.0: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   12.428232] i915 0000:00:10.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   12.428250] i915 0000:00:10.0: [drm] PMU not supported for this GPU.
[   12.428836] [drm] Initialized i915 1.6.0 20201103 for 0000:00:10.0 on minor 0
[   68.659006] i915 0000:00:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [1434]
[   68.659014] i915 0000:00:10.0: [drm] ffmpeg[1434] context reset due to GPU hang
[   89.008013] Fence expiration time out i915-0000:00:10.0:ffmpeg[1439]:2!
[   93.683313] i915 0000:00:10.0: [drm] GPU HANG: ecode 12:1:00000000, in ffmpeg [1439]
[   93.683323] i915 0000:00:10.0: [drm] ffmpeg[1439] context reset due to GPU hang

``` 

```
OMV # sudo /usr/lib/jellyfin-ffmpeg/vainfo --display drm --device /dev/dri/renderD128
sudo: /usr/lib/jellyfin-ffmpeg/vainfo: command not found

```

sacredx72, 2025-01-18:
@purefalse 

> Although /dev/dir is read But there is no function I don’t know if there is any solution
>
> ```
> OMV:~# ls /dev/dri
> by-path  card0  renderD128
> ```
>
> ```
> OMV ~# sudo dmesg | grep i915
>                use [...]

Good morning, have you found a solution so that the Intel driver works correctly and does not generate errors "*ERROR* GT0: GUC: mmio request 0x4100: failure 201/0"?

michael-pptf, 2025-01-27:
@purefalse 
Here are a few steps for troubleshooting.

1. Proxmox version, kernel, dkms module version
2. OMV version, kernel, dkms module version

If OMV is using a kernel older than 6.8 the `.deb` package won't work. You will have to go back to the manual build.

Here's a quick tip: If you are running OMV in an LXC then a lot of frustration can be avoided since you are just passing the host functions to the container. Why run a full size VM if you are running another Linux-base web app is beyond me. 

Somethings to consider:
1. If OMV kernel is old (<6.8), then you will have to build the driver manually on *BOTH* the host and the guest. I have not tried to mismatch the drivers but I don't thing anything good will come out of it.
2. Did you check if you have all the firmware needed? Specifically the GuC firmware.

Topper69, 2025-06-10:
Hi,

Having issue also to make it work.
Proxmox : 8.4.1
`pveversion` pve-manager/8.4.1/2a5fa54a8503f96d (running kernel: 6.8.12-11-pve)`
`dkms status i915-sriov-dkms/2025.05.18, 6.8.12-11-pve, x86_64: installed`
build & install from source (last commit 70fc8be) without any error.

`uname -a Linux pve2 6.8.12-11-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-11 (2025-05-22T09:39Z) x86_64 GNU/Linux`

`dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-11-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.080455] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-11-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    2.606428] i915: loading out-of-tree module taints kernel.
[    2.606456] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.817921] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.`

grub & initramfs are uptodate with the dkms kernel module.

After some troubleshoots, the kernel module doesn't load due to signature error.

Any help?

johntdavis84, 2025-06-11:
> Topper69

It did load. See that “installed” message? That means the DKMS module is loaded for your kernel.

The "verification failed” message is only a problem if you’ve got secure boot enabled.
Don’t worry about the kernel being tainted. That’s just a warning to make sure you know what you’re doing. For example, the ZFS module also taints the kernel, but works fine.

Here’s mine: 
***@***.***:~# dmesg -T | grep -i taint
[Wed Jun  4 19:40:43 2025] spl: loading out-of-tree module taints kernel.
[Wed Jun  4 19:40:43 2025] zfs: module license 'CDDL' taints kernel.
[Wed Jun  4 19:40:43 2025] Disabling lock debugging due to kernel taint
[Wed Jun  4 19:40:43 2025] zfs: module license taints kernel.
[Wed Jun  4 19:40:47 2025] i915: module verification failed: signature and/or required key missing - tainting kernel
[Wed Jun  4 19:40:47 2025] Setting dangerous option enable_guc - tainting kernel

SR-IOV passthrough is working fine on mine.

Are you actually not seeing VFs available to your virtual machines?

Topper69, 2025-06-11:
> It did load. See that “installed” message? That means the DKMS module is loaded for your kernel.
>
> The "verification failed” message is only a problem if you’ve got secure boot enabled.
> Don’t worry [...]

No VFs is availible 

`cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
0`

and when i try to set it, it got an error : 

`echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory`

My Host machine : ASUS NUC13ANH (brand new) with i7-1360P cpu

johntdavis84, 2025-06-11:
> … <x-msg://4/#>

Did you install sysfsutils from the apt repos?

johntdavis84, 2025-06-11:
> … <x-msg://5/#>

Strange.

What does `cat /etc/sysfs.conf` show when you run it?

Mine has a single line:
`devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7`

Topper69, 2025-06-11:
> Strange.
> 
> What does `cat /etc/sysfs.conf` show when you run it?
> 
> Mine has a single line:
> `devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7`
> […](#)

same as yours : 
`cat /etc/sysfs.conf 
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7`

Topper69, 2025-09-14:
> [@Topper69](https://github.com/Topper69) is everything still working well?
> 
> This is still marked as an open issue. You can close it with the link at the bottom of the page if you're all set.

Hi @johntdavis84 , sorry for this late feedback. Yes everything work smoothly. I have upgraded to trixie and still work fine. No issues so far.

Many thanks!

I think we can close out.