strongtz/i915-sriov-dkms #219: dkms install failure, kernel 6.8.0-48

alapins, 2024-11-01:
I'm getting the following error(s) when trying to compile the module:

/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_disable_dp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:542:43: warning: unused variable ‘new_payload’ [-Wunused-variable]
  542 |         struct drm_dp_mst_atomic_payload *new_payload =
      |                                           ^~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:540:49: warning: unused variable ‘old_payload’ [-Wunused-variable]
  540 |         const struct drm_dp_mst_atomic_payload *old_payload =
      |                                                 ^~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:43:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_enable_dp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:48: error: passing argument 2 of ‘drm_dp_add_payload_part2’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |                                                ^~~~~~~~~~~~
      |                                                |
      |                                                struct drm_atomic_state *
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:751:9: note: in expansion of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
  751 |         DRM_DP_ADD_PAYLOAD_PART2(intel_dp, state,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:854:64: note: expected ‘struct drm_dp_mst_atomic_payload *’ but argument is of type ‘struct drm_atomic_state *’
  854 |                              struct drm_dp_mst_atomic_payload *payload);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: error: too many arguments to function ‘drm_dp_add_payload_part2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:751:9: note: in expansion of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
  751 |         DRM_DP_ADD_PAYLOAD_PART2(intel_dp, state,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:853:5: note: declared here
  853 | int drm_dp_add_payload_part2(struct drm_dp_mst_topology_mgr *mgr,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.0-48-generic/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-48-generic'

I've followed the instructions on the site to this point, but I'm not sure what could be causing this.

bbaa-bbaa, 2024-11-02:
try PR #207

alapins, 2024-11-02:
I did (before filing the ticket, actually); same error.  I'm running Linux Mint 22 (based on Ubuntu 22.04), and it looks like it didn't pick up on that and fell through to the default definitions, where the needed definition is gated by kernel version < 6.10.0.  I added `-DRELEASE_UBUNTU=1 \` to EXTRA_CFLAGS in the Makefile and was able to compile successfully.  Not sure whether the RELEASE_UBUNTU and RELEASE_DEBIAN flags are normally autodetermined, or whether you must set them manually in all cases (don't have a straight Ubuntu instance to test on).  It would be nice to have some indication that this is necessary in the readme -- I've submitted a PR with a basic indication to the PR repo.

Thanks for all your work putting this together!

closed 2024-11-02 by alapins

bbaa-bbaa, 2024-11-03:
The lsb_releasedoes not indicate that Mint is based on Ubuntu, so the macro definition is not automatically determined. Perhaps we need a `./configure` script to check which kernel function signatures are actually used, but that might be too complicated :D

alapins, 2024-11-03:
That would be ideal, since it has to be built for the client VMs as well as
the host, but as you say, a bunch of work, and if Intel pushes their stuff
into the kernel and makes backports, it's probably valuable for a
relatively short period of time -- might not be worth it.

johntdavis84, 2024-11-25:
Thanks for posting this. I'm wrestling with this issue now on DragonOS (an Ubuntu Jammy-based distro with, among other things, a custom `lsb_release` string).

I tried to modify my Makefile as indicated, and am still getting a build error. I tried it both with the `DRELEASE` flag at the front of the list of `CFLAGS`, and at the end.

I modified the makefile before running `dkms add`.

[image]

```
root@DragonOS-DualDisk:~/i915-sriov-dkms# dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
dkms status

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=6.8.0-49-generic -C /lib/modules/6.8.0-49-generic/build M=/var/lib/dkms/i915-sriov-dkms/2024.09.21/build............(bad exit status: 2)
ERROR (dkms apport): binary package for i915-sriov-dkms: 2024.09.21 not found
Error! Bad return status for module build on kernel: 6.8.0-49-generic (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/make.log for more information.
i915-sriov-dkms/2024.09.21: added
```

Here's the log:
```
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dsi_dcs_backlight.o
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_disable_dp’:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:542:43: warning: unused variable ‘new_payload’ [-Wunused-variable]
  542 |         struct drm_dp_mst_atomic_payload *new_payload =
      |                                           ^~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:540:49: warning: unused variable ‘old_payload’ [-Wunused-variable]
  540 |         const struct drm_dp_mst_atomic_payload *old_payload =
      |                                                 ^~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:43:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_enable_dp’:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:48: error: passing argument 2 of ‘drm_dp_add_payload_part2’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |                                                ^~~~~~~~~~~~
      |                                                |
      |                                                struct drm_atomic_state *
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:751:9: note: in expansion of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
  751 |         DRM_DP_ADD_PAYLOAD_PART2(intel_dp, state,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:854:64: note: expected ‘struct drm_dp_mst_atomic_payload *’ but argument is of type ‘struct drm_atomic_state *’
  854 |                              struct drm_dp_mst_atomic_payload *payload);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: error: too many arguments to function ‘drm_dp_add_payload_part2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: note: in definition of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:853:5: note: declared here
  853 | int drm_dp_add_payload_part2(struct drm_dp_mst_topology_mgr *mgr,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.8.0-49-generic/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-49-generic'
```

bbaa-bbaa, 2024-12-01:
@johntdavis84
Are you interested in trying this branch? https://github.com/bbaa-bbaa/i915-sriov-dkms/tree/configure_script

I add a configure script for minimal attempt compilation to test kernel function signatures.

The usage is the same as the repo, with no additional manual intervention required. The configure script will be automatically executed during `dkms install`. You just need to replace the `git clone` command with `git clone https://github.com/bbaa-bbaa/i915-sriov-dkms -b configure_script`.

johntdavis84, 2024-12-01:
Hello!

I’m happy to test this. However, I’m an idiot and didn’t run it with the `-b configure_script` flag.

However, it worked. :)

```
:~# git clone https://github.com/bbaa-bbaa/i915-sriov-dkms
Cloning into 'i915-sriov-dkms'...
remote: Enumerating objects: 3549, done.
remote: Counting objects: 100% (946/946), done.
remote: Compressing objects: 100% (423/423), done.
remote: Total 3549 (delta 604), reused 573 (delta 483), pack-reused 2603 (from 1)
Receiving objects: 100% (3549/3549), 4.88 MiB | 18.05 MiB/s, done.
Resolving deltas: 100% (2132/2132), done.

…
***@***.***:~# cd ~/i915-sriov-dkms
dkms add .
Creating symlink /var/lib/dkms/i915-sriov-dkms/2024.11.30/source -> /usr/src/i915-sriov-dkms-2024.11.30
***@***.***:~/i915-sriov-dkms# dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
dkms status

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=6.8.0-49-generic -C /lib/modules/6.8.0-49-generic/build M=/var/lib/dkms/i915-sriov-dkms/2024.11.30/build...............
Signing module:
 - /var/lib/dkms/i915-sriov-dkms/2024.11.30/6.8.0-49-generic/x86_64/module/i915.ko
Secure Boot not enabled on this system.
cleaning build area...

i915.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.8.0-49-generic/updates/dkms/

depmod...
i915-sriov-dkms/2024.11.30, 6.8.0-49-generic, x86_64: installed
```

I blasted that installation out and tried again with the `-b configure_script` flag and got the same successful result.

I did notice that at some point my VM got updated to 6.8.0-49.
So, either that fixed it, or something else in your repo fixed it so as not to require the configure_script.

In either case, thanks for putting extra work into keeping the Ubuntu LTS kernel working with this. :)

N.B.: This still isn’t enough to get the driver working in the latest LTS version of Ubuntu, at least from my experience. Everything came up as expected, but I was still stuck with software rendering. After a bit of troubleshooting—or rather, after remembering how to actually troubleshoot this, as I’ve had it working for months—I verified that the PCIe iGPU SR-IOV slice was showing up in `lspci` (it was) and checked to see which drivers were being used.

It was showing both `xe` and `i915` as being loaded, which surprised the crap out of me as I thought `xe` wasn’t a thing in LTS 6.8 kernels. Ubuntu must be running ahead there. So, I think all I need to do is blacklist the xe driver, as it doesn’t currently support SR-IOV at all, even though the hardware does.

I’ll try that later and report back.

johntdavis84, 2024-12-01:
I have returned.

The xe driver does need to be blacklisted, via a the boot command line (just trying to `blacklist xe` with a text file in `modprobe.d` had no effect.

Under kernel 6.8 in Ubuntu Jammy LTS VM, my working kernel command line looks like this in the relevant portion:
`modprobe.blacklist=xe intel_iommu=on iommu=pt i915.enable_guc=3`

I'd suggest adding the xe blacklist to the Ubuntu instructions in the docs.