strongtz/i915-sriov-dkms #220: sysctl table check failed: dev/i915/(null) procname is null

dmilewski, 2024-11-02:
After update kernel to 6.11 have error 

[    4.470340] i915 0000:00:02.0: 7 VFs could be associated with this PF
**[    4.470395] sysctl table check failed: dev/i915/(null) procname is null
[    4.470397] sysctl table check failed: dev/i915/(null) No proc_handler**
[    4.500538] fbcon: i915drmfb (fb0) is primary device
[    4.551556] Console: switching to colour frame buffer device 240x67
[    4.569028] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device

But device seams work

[    7.134426] i915 0000:00:02.7: enabling device (0000 -> 0002)
[    7.134438] i915 0000:00:02.7: Running in SR-IOV VF mode
[    7.134580] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    7.134593] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    7.134819] i915 0000:00:02.7: [drm] VT-d active for gfx access
[    7.134838] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[    7.135175] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    7.135186] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    7.135332] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    7.135334] i915 0000:00:02.7: HuC firmware PRELOADED
[    7.137054] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.137059] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[    7.137147] [drm] Initialized i915 1.6.0 for 0000:00:02.7 on minor 7
[    7.137395] i915 0000:00:02.0: Enabled 7 VFs

bbaa-bbaa, 2024-11-03:
try PR #207

dmilewski, 2024-11-03:
@bbaa-bbaa thx work without errors

closed 2024-12-29 by strongtz