strongtz/i915-sriov-dkms #223: QSV Hardware accleration not supported in PVE

lq-anu, 2024-11-22:
Hi,

I'm running Jellyfin in an LXC container on Proxmox VE (PVE) with an Intel Core i9-13900H processor. I have set up SR-IOV on the host. The environment is configured to use VA-API for hardware acceleration, and the GPU driver appears to work correctly as vainfo shows valid output. However, when attempting to enable Intel Quick Sync Video (QSV) in Jellyfin, FFmpeg fails with the following error

```
ffmpeg -analyzeduration 200M -probesize 1G -f mov,mp4,m4a,3gp,3g2,mj2 -init_hw_device vaapi=va:,vendor_id=0x8086,driver=iHD -init_hw_device qsv=qs@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -i file:"/mnt/win11smb/Big Buck Bunny - FULL HD 60FPS.mp4" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 hevc_qsv -tag:v:0 hvc1 -preset veryfast -b:v 2874404 -maxrate 2874404 -bufsize 5748808 -profile:v:0 main -level 50 -g:v:0 180 -vf "setparams=color_primaries=bt709:color_trc=bt709:colorspace=bt709,scale_vaapi=w=960:h=540:format=nv12:extra_hw_frames=24,hwmap=derive_device=qsv,format=qsv" -codec:a:0 copy -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename "e21a243f96d30c29688446f3941bd23e-1.mp4" -start_number 0 -hls_segment_filename "/var/cache/jellyfin/transcodes/e21a243f96d30c29688446f3941bd23e%d.mp4" -hls_playlist_type vod -hls_list_size 0 -y "/var/cache/jellyfin/transcodes/e21a243f96d30c29688446f3941bd23e.m3u8"
ffmpeg version 5.1.6-0+deb12u1 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+deb12u1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-sndio --enable-libjxl --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
[AVHWDeviceContext @ 0x6519fb0f10c0] Error initializing an MFX session: -3.
Device creation failed: -1313558101.
Failed to set value 'qsv=qs@va' for option 'init_hw_device': Unknown error occurred
Error parsing global options: Unknown error occurred
```

VaInfo as follows
```
vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP
```

I am not sure if it's because of driver, I have another PVE machine without install the driver, QSV works normally. Did anyone know if the driver supports QSV?

FabioEight, 2025-01-02:
Are you using a MS-01? I noticed the same.

johntdavis84, 2025-01-02:
This has been a known issue for a while, I think. I don't think it's necessarily restricted to the MS-01.

Not sure when it was added to this post (which has been updated several times).
https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

KylinDemons, 2025-01-14:
i5-14250H, Same issue here. Only VAAPI hardware acceleration worked fine for me.

bbaa-bbaa, 2025-01-14:
Windows Guest can use QSV properly. It seems that some part of the Linux software stack is not yet ready to support QSV.

bbaa-bbaa, 2025-01-14:
After test, I has been confirmed that the latest software stack on an Arch Linux guest can works properly with Intel QSV on VF GPU.

Maybe some library on container need update.

KylinDemons, 2025-01-14:
> After test, I has been confirmed that the latest software stack on an Arch Linux guest can works properly with Intel QSV on VF GPU.
> 
> Maybe some library on container need update.

It's wired, I test QSV on ProxmoxVE LXC container(Ubuntu 24.04.1) which passthrough the host's /dev/dri in it, NOT use the PCI SRIOV passthrough in a Linux VM. When I tried to transcode a video by Jellyfin QSV method, the ffmpeg report an error:
```
[08:23:32] [INF] [65] MediaBrowser.MediaEncoding.Transcoding.TranscodeManager: /usr/lib/jellyfin-ffmpeg/ffmpeg -analyzeduration 200M -probesize 1G -f matroska -init_hw_device vaapi=va:/dev/dri/renderD133,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -canvas_size 1920x1080 -i file:"/media1/Movie/荒野机器人 (2024)/荒野机器人 (2024) - 2160p.mkv" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:2 -map -0:0 -codec:v:0 hevc_qsv -tag:v:0 hvc1 -preset veryfast -b:v 10937963 -maxrate 10937963 -bufsize 21875926 -profile:v:0 main -level 50 -g:v:0 72 -keyint_min:v:0 72 -filter_complex "[0:6]scale,scale=1920:1080:fast_bilinear,format=bgra,hwupload=derive_device=qsv:extra_hw_frames=64[sub];[0:0]setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=2560:h=1440:extra_hw_frames=24,procamp_vaapi=b=16,tonemap_vaapi=format=nv12:p=bt709:t=bt709:m=bt709:extra_hw_frames=32,hwmap=derive_device=qsv,format=qsv[main];[main][sub]overlay_qsv=eof_action=pass:repeatlast=0:w=2560:h=1440" -start_at_zero -codec:a:0 ac3 -ac 6 -ab 448000 -ar 48000 -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename "0d645d048bdd21013386d4b3d903ac01-1.mp4" -start_number 0 -hls_segment_filename "/cache/transcodes/0d645d048bdd21013386d4b3d903ac01%d.mp4" -hls_playlist_type vod -hls_list_size 0 -y "/cache/transcodes/0d645d048bdd21013386d4b3d903ac01.m3u8"
[08:23:32] [ERR] [49] MediaBrowser.MediaEncoding.Transcoding.TranscodeManager: FFmpeg exited with code 171
[08:23:32] [ERR] [49] Jellyfin.Api.Middleware.ExceptionMiddleware: Error processing request. URL GET /videos/cbb399ff-2494-d467-cdb4-9e5a7bb66346/hls1/main/-1.mp4.
MediaBrowser.Common.FfmpegException: FFmpeg exited with code 171
```
I have set jellyfin to use /dev/dri/renderD133 which belong to a unused VF and passthrough to the LXC container,  but no luck.

trablin45, 2025-01-19:
For me I got:
`[AVHWDeviceContext @ 0x59e72f435740] Error setting child device handle: -17
Device creation failed: -1313558101.
Failed to set value 'qsv=qs@va' for option 'init_hw_device': Unknown error occurred
Error parsing global options: Unknown error [occurred`]
Test with privileged and unprivileged lxc containers both same error. i5-12400

bbaa-bbaa, 2025-01-20:
> Maybe some library on container need update.

It seems that the vendor FFmpeg bundled with Jellyfin is still using libmfx.

```
[AVHWDeviceContext @ 0x6519fb0f10c0] Error initializing an MFX session: -3.
```

The Intel Media SDK was archived by Intel in 2023, and the code -3 appears to indicate a feature not yet supported by libmfx.

https://github.com/Intel-Media-SDK/MediaSDK/blob/7a72de33a15d6e7cdb842b12b901a003f7154f0a/api/include/mfxdefs.h#L198

Perhaps updating FFmpeg to use the libvpl library could resolve this issue.

bbaa-bbaa, 2025-01-20:
> Perhaps updating FFmpeg to use the libvpl library could resolve this issue.

I can confirm that Arch Linux with the software stack libvpl 2.14.0, vpl-gpu-rt 24.4.4, and FFmpeg 7.1 can use Intel QSV properly within an LXC container.

However, it seems that even with Ubuntu 24.04, the included libraries still too old to make Intel QSV work correctly.

KylinDemons, 2025-01-20:
> I can confirm that Arch Linux with the software stack libvpl 2.14.0, vpl-gpu-rt 24.4.4, and FFmpeg 7.1 can use Intel QSV properly within an LXC container.
>
> However, it seems that even with Ubuntu [...]

Thx for your reply! I test the offical jellyfin docker image in  Arch LXC container but QSV is still not work correctly, so it's seem that the jellyfin-ffmpeg is too old.

bbaa-bbaa, 2025-01-20:
It seems that using the latest version of jellyfin-ffmpeg (https://github.com/jellyfin/jellyfin-ffmpeg/releases/tag/v7.0.2-8) on Ubuntu 24.04 allows QSV to work properly via the command line. But I have not yet tested its integration with Jellyfin. I will tests it later.

trablin45, 2025-01-20:
>  Perhaps updating FFmpeg to use the libvpl library could resolve this issue.
> However, it seems that even with Ubuntu 24.04, the included libraries still too old to make Intel QSV work correctly.
Does software packages differs from installations on lxc vs vms ? I've fallback to a ubuntu 24.04 lts vm (until this gets solved) im using the same intel drivers as my host and my other jellyfin ubuntu 24 lxc, same jellyfin 10.3.0 with the ffmpeg that ships with 7.0.2-jellyfin, 6.8 kernel, no other packages installed and QSV works (but damn 4 times more ram usage at cost) I think there's something lost in translation when lxc's communicate to the host kernel.

bbaa-bbaa, 2025-01-20:
@KylinDemons Which docker image are you using? In my case  jellyfin/jellyfin work out of box.

bbaa-bbaa, 2025-01-20:
@trablin45 Can you run `/usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD128 -init_hw_device qsv@va` in the LXC container and paste the output?

KylinDemons, 2025-01-20:
> [@KylinDemons](https://github.com/KylinDemons) Which docker image are you using? In my case jellyfin/jellyfin work out of box.

The offical jellyfin/jellyfin:latest. SRIOV ubuntu 24.04 VM QSV works fine, But Ubuntu 24.04 LXC does not work, only vaapi works.

KylinDemons, 2025-01-20:
I will test LXC+Docker Jellyfin, LXC+Deb Jellyfin based Ubuntu 24.04.1 CT and post the logs later.

KylinDemons, 2025-01-21:
I have jellyfin installed by pacman in Arch Linux within Proxmox VE LXC Container.
/dev/dri/renderD133 is the VF which I passthrough to the container (not used by any other vm or lxc).

This is the lxc container conf file: 
```properties
arch: amd64
cores: 8
features: fuse=1,mount=nfs;cifs,nesting=1
hostname: ArchLinux
memory: 4096
mp1: /mnt/SSD,mp=/mnt/SSD
net0: name=eth0,bridge=vmbr0,firewall=1,hwaddr=BC:24:11:B3:4A:06,ip=dhcp,ip6=auto,type=veth
ostype: archlinux
rootfs: local-lvm:vm-103-disk-0,size=64G
swap: 512
lxc.cgroup2.devices.allow: c 226:0 rwm
lxc.cgroup2.devices.allow: c 226:128 rwm
lxc.cgroup2.devices.allow: c 29:0 rwm
lxc.mount.entry: /dev/dri dev/dri none bind,optional,create=dir
lxc.mount.entry: /dev/dri/renderD133 dev/dri/renderD133 none bind,optional,create=file
lxc.mount.entry: /dev/fb0 dev/fb0 none bind,optional,create=file
lxc.apparmor.profile: unconfined
lxc.cgroup2.devices.allow: a
lxc.cap.drop: 
```

After install jellyfin-server and jellyfin-web, the output of vainfo and ffmpeg is here: 
```bash
[root@ArchLinux ~]# /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD133 -init_hw_device opencl@va
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 14.2.1 (GCC) 20240910
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-shared --disable-libxcb --disable-sdl2 --disable-xlib --enable-gpl --enable-version3 --enable-static --enable-gmp --enable-gnutls --enable-chromaprint --enable-libfontconfig --enable-libass --enable-libbluray --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libmp3lame --enable-libopus --enable-libopenmpt --enable-libtheora --enable-libvorbis --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-opencl --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x5684c14272c0] libva: VA-API version 1.22.0
[AVHWDeviceContext @ 0x5684c14272c0] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5684c14272c0] libva: Found init function __vaDriverInit_1_22
[AVHWDeviceContext @ 0x5684c14272c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5684c14272c0] Initialised VAAPI connection: version 1.22
[AVHWDeviceContext @ 0x5684c14272c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.4.4 ().
[AVHWDeviceContext @ 0x5684c14272c0] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x5684c1462940] 0.0: Intel(R) OpenCL Graphics / Intel(R) UHD Graphics
[AVHWDeviceContext @ 0x5684c1462940] Intel QSV to OpenCL mapping function found (clCreateFromVA_APIMediaSurfaceINTEL).
[AVHWDeviceContext @ 0x5684c1462940] Intel QSV in OpenCL acquire function found (clEnqueueAcquireVA_APIMediaSurfacesINTEL).
[AVHWDeviceContext @ 0x5684c1462940] Intel QSV in OpenCL release function found (clEnqueueReleaseVA_APIMediaSurfacesINTEL).
Universal media converter
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
```

```bash
[root@ArchLinux ~]# vainfo --display drm --device /dev/dri/renderD133
Trying display: drm
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.4.4 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
[... 53 lines cut ...]

It seems all right......But jellyfin transcode I got the error:
```bash
/usr/lib/jellyfin-ffmpeg/ffmpeg -analyzeduration 200M -probesize 1G -f matroska -init_hw_device vaapi=va:/dev/dri/renderD133,driver=iHD -init_hw_device qsv=qs@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -i file:"/mnt/SSD/Media/TVShow/比翼之吻 (2018)/Season 1/比翼之吻 - S01E01 - 第 1 集.mkv" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 hevc_qsv -tag:v:0 hvc1 -preset veryfast -b:v 292000 -maxrate 292000 -bufsize 584000 -profile:v:0 main -level 50 -g:v:0 90 -keyint_min:v:0 90 -vf "setparams=color_primaries=bt709:color_trc=bt709:colorspace=bt709,scale_vaapi=w=960:h=540:format=nv12:extra_hw_frames=24,hwmap=derive_device=qsv,format=qsv" -codec:a:0 aac -ac 2 -ab 128000 -ar 44100 -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename "9f3e8351a876007cbc1728dab8f26d22-1.mp4" -start_number 0 -hls_segment_filename "/var/cache/jellyfi [... cut ...]
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x64f06f5add00] No VA display found for device /dev/dri/renderD133.
Device creation failed: -22.
Failed to set value 'vaapi=va:/dev/dri/renderD133,driver=iHD' for option 'init_hw_device': Invalid argument
Error parsing global options: Invalid argument
```

I have no idea, but it seems that jellyfin can't recognise the renderD133.

[image]

bbaa-bbaa, 2025-01-21:
Can you run /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD128 -init_hw_device qsv@va and upload the output?
I can't see any log about the intel sdk.

KylinDemons, 2025-01-21:
> Can you run `/usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD128 -init_hw_device qsv@va` and upload the output? I can't see any log about the intel sdk.

here is
```bash
[root@ArchLinux ~]# /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD128 -init_hw_device qsv@va
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 14.2.1 (GCC) 20240910
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-shared --disable-libxcb --disable-sdl2 --disable-xlib --enable-gpl --enable-version3 --enable-static --enable-gmp --enable-gnutls --enable-chromaprint --enable-libfontconfig --enable-libass --enable-libbluray --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libmp3lame --enable-libopus --enable-libopenmpt --enable-libtheora --enable-libvorbis --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-opencl --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x5e960e6172c0] libva: VA-API version 1.22.0
[AVHWDeviceContext @ 0x5e960e6172c0] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5e960e6172c0] libva: Found init function __vaDriverInit_1_22
[AVHWDeviceContext @ 0x5e960e6172c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5e960e6172c0] Initialised VAAPI connection: version 1.22
[AVHWDeviceContext @ 0x5e960e6172c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.4.4 ().
[AVHWDeviceContext @ 0x5e960e6172c0] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x5e960e652940] Use Intel(R) oneVPL to create MFX session, API version is 2.14, the required implementation version is 1.3
[AVHWDeviceContext @ 0x5e960e652940] Initialize MFX session: implementation version is 2.13
Universal media converter
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
```

bbaa-bbaa, 2025-01-21:
```
lxc.cgroup2.devices.allow: c 226:0 rwm
lxc.cgroup2.devices.allow: c 226:128 rwm
lxc.cgroup2.devices.allow: c 29:0 rwm
lxc.mount.entry: /dev/dri dev/dri none bind,optional,create=dir
lxc.mount.entry: /dev/dri/renderD133 dev/dri/renderD133 none bind,optional,create=file
```

You only allow `226:128` dev node in your lxc config.
The `/dev/dri/renderD133` should be "226:133"
You can verify that with `ls -l /dev/dri/renderD133` in your host system (PVE Shell).
Besides, I think the /dev/dri/cardX is also need to passthrough to your container.

KylinDemons, 2025-01-21:
@bbaa-bbaa Tested with docker inside LXC(Same as the environment above).
unfortunately the same issue with @trablin45 :
```bash
/usr/lib/jellyfin-ffmpeg/ffmpeg -analyzeduration 200M -probesize 1G -f matroska -init_hw_device vaapi=va:/dev/dri/renderD133,driver=iHD -init_hw_device qsv=qs@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -i file:"/media1/TVShow/老友记 (1994)/Season 1/老友记 - S01E01 - 第 1 集.mkv" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 hevc_qsv -tag:v:0 hvc1 -preset veryfast -b:v 292000 -maxrate 292000 -bufsize 584000 -profile:v:0 main -level 50 -g:v:0 72 -keyint_min:v:0 72 -vf "setparams=color_primaries=bt709:color_trc=bt709:colorspace=bt709,scale_vaapi=w=960:h=540:format=nv12:extra_hw_frames=24,hwmap=derive_device=qsv,format=qsv" -codec:a:0 ac3 -ac 6 -ab 128000 -ar 48000 -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename "738ddcaede1e48a16873cc9a30349d23-1.mp4" -start_number 0 -hls_segment_filename "/cache/transcodes/738ddcaede1e48a16873cc9a30349d23%d.mp4" -hls_playlist_type vod -hls_list_size 0 -y "/cache/transcodes/738ddcaede1e48a16873cc9a30349d23.m3u8"

ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=amd64 --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x6467e88b60c0] Error setting child device handle: -17
Device creation failed: -1313558101.
Failed to set value 'qsv=qs@va' for option 'init_hw_device': Unknown error occurred
Error parsing global options: Unknown error occurred
```

Here is the output of vainfo and ffmpeg:
```bash
[root@ArchLinux ~]# docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/vainfo --display drm --device /dev/dri/renderD133
Trying display: drm
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.4.2 (b6b411f)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
[... cut ...]
[AVHWDeviceContext @ 0x586e406d6980] Intel QSV to OpenCL mapping function found (clCreateFromVA_APIMediaSurfaceINTEL).
[AVHWDeviceContext @ 0x586e406d6980] Intel QSV in OpenCL acquire function found (clEnqueueAcquireVA_APIMediaSurfacesINTEL).
[AVHWDeviceContext @ 0x586e406d6980] Intel QSV in OpenCL release function found (clEnqueueReleaseVA_APIMediaSurfacesINTEL).
Universal media converter
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
```

bbaa-bbaa, 2025-01-21:
> [AVHWDeviceContext @ 0x6467e88b60c0] Error setting child device handle: -17
> Device creation failed: -1313558101.
> Failed to set value 'qsv=qs@va' for option 'init_hw_device': Unknown error occurred
```
 MFX_ERR_DEVICE_FAILED               = -17,  /* device operation failure */
```
https://github.com/intel/vpl-gpu-rt/blob/6a21db7af1aad95914f8ff5f181661a1ed92fec6/api/vpl/mfxdefs.h#L288

Can you install `onevpl-tools` (`apt install onevpl-tools`), and paste the output of `vpl-inspect` ?

KylinDemons, 2025-01-21:
> vpl-inspect

This is the host output:
```bash
# root @ pve in ~ [18:57:47] 
$ vpl-inspect

Warning - no implementations found by MFXEnumImplementations()
```

This is the LXC container output:
```bash
[root@ArchLinux ~]# vpl-inspect

Implementation #0: mfx-gen
  Library path: /usr/lib/libmfx-gen.so.1.2.13
  AccelerationMode: MFX_ACCEL_MODE_VIA_VAAPI
  ApiVersion: 2.13
  Impl: MFX_IMPL_TYPE_HARDWARE
  VendorImplID: 0x0000
  ImplName: mfx-gen
  License: MIT License
  Version: 1.2
  Keywords:
  VendorID: 0x8086
  mfxAccelerationModeDescription:
    Version: 1.0
    Mode: MFX_ACCEL_MODE_VIA_VAAPI
  mfxPoolPolicyDescription:
    Version: 1.0
    Policy: MFX_ALLOCATION_OPTIMAL
    Policy: MFX_ALLOCATION_UNLIMITED
    Policy: MFX_ALLOCATION_LIMITED
  mfxDeviceDescription:
    MediaAdapterType: MFX_MEDIA_INTEGRATED
    DeviceID: 46a3/0
    Version: 1.1
  mfxDecoderDescription:
    Version: 1.0
    CodecID: VC1
    MaxcodecLevel: 5
      Profile: MFX_PROFILE_VC1_SIMPLE
        MemHandleType: MFX_RESOURCE_SYSTEM_SURFACE
          Width Min: 16
          Width Max: 4096
          Width Step: 16
          Height Min: 16
          Height Max: 4096
          Height Step: 16
          ColorFormats: NV12
        MemHandleType: MFX_RESOURCE_VA_SURFACE_PTR
          Width Min: 16
          Width Max: 4096
          Width Step: 16
          Height Min: 16
          Height Max: 4096
          Height Step: 16
          ColorFormats: NV12
      Profile: MFX_PROFILE_VC1_MAIN
        MemHandleType: MFX_RESOURCE_SYSTEM_SURFACE
          Width Min: 16
          Width Max: 4096
          Width Step: 16
          Height Min: 16
          Height Max: 4096
          Height Step: 16
          ColorFormats: NV12
        MemHandleType: MFX_RESOURCE_VA_SURFACE_PTR
          Width Min: 16
          Width Max: 4096
[... 1727 lines cut ...]
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_COPY
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_COPY
      SurfaceComponent: MFX_SURFACE_COMPONENT_VPP_INPUT
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_COPY
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_COPY
      SurfaceComponent: MFX_SURFACE_COMPONENT_VPP_OUTPUT
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_IMPORT_COPY
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_SHARED
        SurfaceFlags:     MFX_SURFACE_FLAG_EXPORT_COPY

Total number of implementations found = 1
```

By the way, this is Ubuntu24.04 VM with VF output:
```bash
# ubuntu @ Ubuntu-VM in ~ [19:02:16]
$ vpl-inspect

Warning - no implementations found by MFXEnumImplementations()
```

bbaa-bbaa, 2025-01-21:
It seems that jellyfin-ffmpeg in Docker includes its own runtime, and system scope debugging tools do not appear to provide useful information.

Perhaps this is still a permission-related issue. Have you tried using a privileged container?
Can jellyfin-ffmpeg initialize the MFX session when run as the root user? `docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD133 -init_hw_device qsv@va  `

bbaa-bbaa, 2025-01-21:
> By the way, this is Ubuntu24.04 VM with VF output
> .......
> It seems that jellyfin-ffmpeg in Docker includes its own runtime

Can you try running vpl-inspect in the Jellyfin runtime?

`LD_LIBRARY_PATH="/usr/lib/jellyfin-ffmpeg/lib/:/usr/lib" vpl-inspect`

KylinDemons, 2025-01-21:
> It seems that jellyfin-ffmpeg in Docker includes its own runtime, and system scope debugging tools do not appear to provide useful information.
>
> Perhaps this is still a permission-related issue. [...]

YOU ARE RIGHT!! I recreate the jellyfin container with `--privileged=true`, QSV works correctly!!
Here is output of unprivileged docker container:
```bash
[root@ArchLinux ~]# docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD133 -init_hw_device qsv@va
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=amd64 --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x6080eec15640] libva: VA-API version 1.22.0
[AVHWDeviceContext @ 0x6080eec15640] libva: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x6080eec15640] libva: Found init function __vaDriverInit_1_22
[AVHWDeviceContext @ 0x6080eec15640] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x6080eec15640] Initialised VAAPI connection: version 1.22
[AVHWDeviceContext @ 0x6080eec15640] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.4.2 (b6b411f).
[AVHWDeviceContext @ 0x6080eec15640] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x6080eec50a00] Use Intel(R) oneVPL to create MFX session, API version is 2.13, the required implementation version is 1.3
[AVHWDeviceContext @ 0x6080eec50a00] Initialize MFX session: implementation version is 1.35
[AVHWDeviceContext @ 0x6080eec50a00] Error setting child device handle: -17
Device creation failed: -1313558101.
Failed to set value 'qsv@va' for option 'init_hw_device': Unknown error occurred
Error parsing global options: Unknown error occurred
```

Here is the output of privileged docker container::
```bash
[root@ArchLinux ~]# docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/ffmpeg -v verbose -init_hw_device vaapi=va:/dev/dri/renderD133 -init_hw_device qsv@va
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=amd64 --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
[AVHWDeviceContext @ 0x653838f93640] libva: VA-API version 1.22.0
[AVHWDeviceContext @ 0x653838f93640] libva: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x653838f93640] libva: Found init function __vaDriverInit_1_22
[AVHWDeviceContext @ 0x653838f93640] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x653838f93640] Initialised VAAPI connection: version 1.22
[AVHWDeviceContext @ 0x653838f93640] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.4.2 (b6b411f).
[AVHWDeviceContext @ 0x653838f93640] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x653838fce9c0] Use Intel(R) oneVPL to create MFX session, API version is 2.13, the required implementation version is 1.3
libva info: VA-API version 1.22.0
[... cut ...]
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x653838fce9c0] Initialize MFX session: implementation version is 2.13
Universal media converter
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
```
[image]

It's almost certain to be a permission issue.
Maybe can close the issue. @trablin45 @lq-anu

KylinDemons, 2025-01-21:
> so the problem is solved by running chmod 666 /dev/dri/* in the host and lxc ? Has anyone tested HDR tone mapping with qsv on jellyfin lxc? According to derekseaman blog post lxc's may have an issue on plex hdr tone mapping when using vf gpus.
> 
> > If you enable HDR tone mapping on the Plex server and you are viewing HDR content on a device that needs server side HDR tone mapping, the video stream will likely be corrupted

@bbaa-bbaa @trablin45  HDR tone mapping still has problems in QSV. In jellyfin can play HDR file but the frame is bad.
QSV Like this:

[image]

VAAPI works fine for HDR tone mapping.
VAAPI:

[image]

LXC and VM has the same issue, maybe need upstream to fix it.

bbaa-bbaa, 2025-01-21:
Does this issue only occur with the VF? For LXC, perhaps you could try passing through the PF‘s render node (/dev/dri/renderD128) directly?

KylinDemons, 2025-01-21:
> Does this issue only occur with the VF? For LXC, perhaps you could try passing through the PF‘s render node (/dev/dri/renderD128) directly?

Wonderful, PF /dev/dri/renderD128 works fine for QSV HDR tone mapping, no frame bad.
But as far as I know, it seems that someone mentioned that PF cannot be used in LXC? I don't know.

bbaa-bbaa, 2025-01-21:
Passing through the PF (Actually, only the relevant DRM/DRI nodes) into a container solution like LXC works perfectly without any issues. You don’t even need to use this ported driver to pass your GPU nodes into the container.

The VF is only used when running a VM or any requiring full PCI device access.

KylinDemons, 2025-01-21:
> Passing through the PF (Actually, only the relevant DRM/DRI nodes) into a container solution like LXC is no problem. You don’t even need to use this ported driver to pass your GPU nodes into the container.
> 
> The VF is designed to used when running a VM or any requiring full PCI device access.

Got. Anyway, thank you very much for helping to solve these! I will continue to see if decoding and transcoding have any other issues.

bbaa-bbaa, 2025-01-22:
> LXC and VM has the same issue, maybe need upstream to fix it.

Upstream Issue: https://github.com/intel/media-driver/issues/1665

trablin45, 2025-01-23:
What is the solution to using vgpu in a lxc ?

bbaa-bbaa, 2025-01-23:
You can directly passthrough your PF GPU DRM node to the LXC container without using any vGPU.
Passthrough the PF's DRM node to the container will not affect the usage of other VFs.

trablin45, 2025-01-27:
thank you all for the fix

austonpramodh, 2025-04-11:
I am still having issues

lxc-conf
```
arch: amd64
cores: 2
features: mount=nfs,nesting=1
hostname: olympus
memory: 2048
net0: name=eth0,bridge=vmbr0,hwaddr=BC:24:11:1D:6A:64,ip=dhcp,ip6=auto,type=veth
onboot: 0
ostype: ubuntu
rootfs: local:100/vm-100-disk-0.raw,size=64G
swap: 512
lxc.cgroup2.devices.allow: c 226:0 rwm
lxc.cgroup2.devices.allow: c 226:128 rwm
lxc.cgroup2.devices.allow: c 226:* rwm
lxc.mount.entry: /dev/dri/card0 dev/dri/card0 none bind,optional,create=file
lxc.mount.entry: /dev/dri/renderD128 dev/dri/renderD128 none bind,optional,create=file
```

Host Permissions:
```
ls -l /dev/dri/
total 0
drw-rw-rw- 2 root root        360 Apr 11 05:03 by-path
crw-rw-rw- 1 root video  226,   0 Apr 11 05:03 card0
crw-rw-rw- 1 root video  226,   1 Apr 11 05:03 card1
crw-rw-rw- 1 root video  226,   2 Apr 11 05:03 card2
crw-rw-rw- 1 root video  226,   3 Apr 11 05:03 card3
crw-rw-rw- 1 root video  226,   4 Apr 11 05:03 card4
crw-rw-rw- 1 root video  226,   5 Apr 11 05:03 card5
crw-rw-rw- 1 root video  226,   6 Apr 11 05:03 card6
crw-rw-rw- 1 root video  226,   7 Apr 11 05:03 card7
crw-rw-rw- 1 root render 226, 128 Apr 11 05:03 renderD128
crw-rw-rw- 1 root render 226, 129 Apr 11 05:03 renderD129
crw-rw-rw- 1 root render 226, 130 Apr 11 05:03 renderD130
crw-rw-rw- 1 root render 226, 131 Apr 11 05:03 renderD131
crw-rw-rw- 1 root render 226, 132 Apr 11 05:03 renderD132
crw-rw-rw- 1 root render 226, 133 Apr 11 05:03 renderD133
crw-rw-rw- 1 root render 226, 134 Apr 11 05:03 renderD134
crw-rw-rw- 1 root render 226, 135 Apr 11 05:03 renderD135
```

Container permissions:
```
ls -lh /dev/dri                                                                                                                                                                        1 ↵
total 0
crw-rw-rw- 1 root video 226,   0 Apr 11 14:03 card0
crw-rw-rw- 1 root _ssh  226, 128 Apr 11 14:03 renderD128
```

FFMPEG Logs
```
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.3.0-6ubuntu2~24.04)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=amd64 --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5f3acb70a9c0] Error setting child device handle: -17
Device creation failed: -1313558101.
Failed to set value 'qsv=qs@va' for option 'init_hw_device': Unknown error occurred
Error parsing global options: Unknown error occurred
```

GUID inside container
```
cat /etc/group | grep _ssh
_ssh:x:104:
cat /etc/group | grep render
render:x:993:
```

Not sure what am i missing
DockerComposefile
```
services:
  jellyfin:
    image: lscr.io/linuxserver/jellyfin:10.10.7
    #image: jellyfin/jellyfin:10.10
    container_name: jellyfin
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=America/New_York
      # - JELLYFIN_PublishedServerUrl=https://jellyfin.helium.auston.xyz #optional
      - DOCKER_MODS=linuxserver/mods:jellyfin-opencl-intel-25.09.32961.7
    volumes:
      - ./data/jellyfin-config:/config
      #- nfs-media:/data/media
      - ./data/media:/data/media
    devices:
      - /dev/dri:/dev/dri # VAAPI/NVDEC/NVENC render nodes
      #- /dev/dri/renderD128:/dev/dri/renderD128
    ports:
      - 8096:8096
      - 8920:8920 #optional
      - 7359:7359/udp #optional
      - 1900:1900/udp #optional
    privileged: true
    group_add:
      - "993"
      - "44"
      - "104"
    restart: unless-stopped
```

Tried with both jellyfin, and linuxserver/jellyfin

h1ght, 2025-04-22:
i fiddled around had similiar issue, vm freezes when doing qsv transcodes and showing 100% usage in the vm. vaapi is no problem. it doesnt matter if i put renderD128 or a vf into the lxc container. vm recovers when qsv is finished
also when the vm uses a vf, qsv becomes not available and ffmpeg complains about libmfx or something. guess the problem is that intel media driver goes eol (?)