strongtz/i915-sriov-dkms #224: Proxmox 8.3 Support?

johntdavis84, 2024-11-22:
Proxmox 8.3 dropped this week.

Does the current release of the DKMS driver (the one that downloads from a `git clone` command) work with Proxmox 8.3? Has anyone tested it yet? Either with kernel 6.8.12-4 or 6.11?

Fredzer, 2024-11-22:
I came across this guide a few days ago setup on my latest node fresh install-
pve-manager 8.3.0 + Linux 6.8.12-4-pve (2024-11-06T15:04Z)
CPU with on-board GPU - 13th Gen Intel(R) Core(TM) i9-13900H 
After some fiddling it was reasonably smooth in the end and I now have the devices below to assign eventually to cluster map

`00:02.0 VGA compatible controller [0300]: Intel Corporation Raptor Lake-P [Iris Xe Graphics] [8086:a7a0] (rev 04)
        DeviceName: Onboard - Video
        Kernel driver in use: i915

00:02.1 VGA compatible controller [0300]: Intel Corporation Raptor Lake-P [Iris Xe Graphics] [8086:a7a0] (rev 04)
        Kernel driver in use: i915
        Kernel modules: xe, i915

00:02.2 VGA compatible controller [0300]: Intel Corporation Raptor Lake-P [Iris Xe Graphics] [8086:a7a0] (rev 04)
        Kernel driver in use: vfio-pci
        Kernel modules: xe, i915
...
...
...
...

00:02.7 VGA compatible controller [0300]: Intel Corporation Raptor Lake-P [Iris Xe Graphics] [8086:a7a0] (rev 04)
        Kernel driver in use: i915
        Kernel modules: xe, i915

`

I have a Windows 2022 guest that worked right out of the box. Ran benchmarks, latest drivers etc it was dream. intel_gpu_top -d was showing me usage on the host side too.
The same has unfortunately, not applied to my attempts at Linux Ubuntu guests with days of trying various things to get working.

For your question though, yes it was rather smooth for me with fresh install :)

johntdavis84, 2024-11-22:
I didn't see a link to a guide. I don't think it came through. :(

What did you mean by "reasonably smooth?" Ideally, it should just be buildable and installable using the same iprocedure as before.

Are your Linux guests still not working? I've got working Linux guests with iGPU passthrough now in Proxmox 8.2, so I'd not consider the DKMS driver working under Proxmox 8.3 if that's the case.

qqshfox, 2024-11-22:
Just upgraded to 8.3 and 6.11.

```
root@pve:~# uname -a
Linux pve 6.11.0-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.11.0-1 (2024-10-23T15:32Z) x86_64 GNU/Linux
root@pve:~# lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.4 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.5 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.6 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
00:02.7 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
```

PV passthrough to LXC works just fine. Didn't test VF tho.

johntdavis84, 2024-11-25:
> Just upgraded to 8.3 and 6.11.
> 
> ```
> root@pve:~# uname -a
> Linux pve 6.11.0-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.11.0-1 (2024-10-23T15:32Z) x86_64 GNU/Linux
> root@pve:~# lspci | grep -i vga
> 00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.1 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.2 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.3 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.4 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.5 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.6 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> 00:02.7 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
> ```
> 
> PF passthrough to LXC works just fine. ~Didn't test VF tho.~
> 
> Tested VF in a Windows VM and it works well too.

Great news. :) 

I'm curious if there's specific reason you went with the 6.11 kernel. Does it have better support for the DKMS module? (I know that Intel's been iterating their iGPU drivers through the last few kernel releases, but I wasn't sure if that had performance/stability implications for the i915 driver (or the modified one we're using here).

lichao0223, 2024-11-25:
I have tested and everything work fine with Proxmox 8.3 and kernel 6.11

feerlessleadr, 2024-11-29:
For anyone that has upgraded, can you advise on the best way to upgrade to 8.3 and preserve this patch?

qqshfox, 2024-11-29:
@johntdavis84 I don't have any specific reason. Just wanted to stay up to date. So far I haven't seen any differences between 6.11 and 6.8.

@feerlessleadr I use this to force install the module when installing new kernel.

```
echo "i915-sriov-dkms" > /usr/share/dkms/modules_to_force_install/i915
```

feerlessleadr, 2024-11-29:
> @feerlessleadr I use this to force install the module when installing new kernel.
> 
> ```
> echo "i915-sriov-dkms" > /usr/share/dkms/modules_to_force_install/i915
> ```

Thanks, and apologies in advance, I think all of the turkey yesterday is making my brain not work. 

This is the guide I followed to get the module installed on my current kernel:

https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

So the procedure I would do is unpin my current kernel (6.8.12-3-pve) and update proxmox to 8.3.

Once updated, I would run what you suggested:

echo "i915-sriov-dkms" > /usr/share/dkms/modules_to_force_install/i915

And that would be it?

qqshfox, 2024-11-29:
Thanks for sharing the guide which is well written.

For upgrading to 6.8, what I did is just `apt upgrade`. It installed all the latest packages and the latest kernel 6.8. Since I was on the latest version `2024.09.21` of this repo, it's already supported kernel 6.11 so I didn't do any specific before upgrading kernel to 6.11.

Having `echo "i915-sriov-dkms" > /usr/share/dkms/modules_to_force_install/i915` just saves you one step after installing a new kernel. Otherwise, you'll have to force install the module and reboot again after you upgrade the kernel and reboot. It should still boot even if this module is not installed.

feerlessleadr, 2024-11-29:
Oh ok, so I would run that command before I upgrade everything, and then the module will auto reinstall once the kernel is upgraded.

qqshfox, 2024-11-29:
Yea. It should boot even if the module is not successfully installed. You can always reinstall it after it boots up.

If the kernel didn't boot up, you should still be able to switch back to the previous kernel as long as you haven't removed it. And please backup important data before you proceed just in case anything unexpected happens.

feerlessleadr, 2024-11-29:
Thank you! I'm going to give this a shot tonight when my family is sleeping

RobertoCMCosta, 2024-11-29:
Guys, I've been struggling to make this work with the alder lake igpu to no avail.
My latest attempt was on proxmox 8.2 with kernel 6.8.12. But I'd failed with kernel 6.5 too.
I'm reading this thread with reports of success with 13th Gen intel processors and the latest version 8.3 with upgraded kernels.
I've followed Derek's guide as well as Upinel's and a bunch of reddit posts.
I just wanted to pass the on-board intel vgpu to windows 10 vm running alongside a windows 11 vm with a passed thru amd gpu. :(
Do you guys have any suggestions?

qqshfox, 2024-11-30:
@RobertoCMCosta I'd suggest you to follow Derek's guide, (start over if you can) and document the steps and the result for each step. Can't really give any useful suggestions without seeing what went wrong on your machine. And also would suggest to start a new thread as this thread mainly focuses on 8.3.

Ch3ck3rM0n, 2024-12-10:
There seems to be an issue with new kernel 6.8.12-5, so please stay at 6.8.12-4 till it is fixed.

ibis69-git, 2025-01-21:
Hi, new 6.11.11.1 pve kernel version not working, "unknown parameter 'max_vfs' ignored"

0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.11.11-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=4 module_blacklist=xe
[    0.096556] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.11.11-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=4 module_blacklist=xe
[    2.928702] i915: unknown parameter 'max_vfs' ignored
[    2.929108] i915 0000:00:02.0: [drm] Found ALDERLAKE_P (device ID 46a6) display version 13.00 stepping D0
[    2.930946] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.930948] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.930988] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.931418] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.937419] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    2.942199] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.36.0
[    2.942204] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    2.956298] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[    2.956664] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    2.956665] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    2.957200] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    2.957822] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.005790] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    3.016691] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.043719] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.054641] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.286308] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
[    4.286330] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs

bbaa-bbaa, 2025-01-21:
@ibis69-git 
Please make sure you have the **proxmox-headers-6.11** package installed. Otherwise, DKMS modules will not automatically rebuild after a kernel update.

ibis69-git, 2025-01-21:
Hi @bbaa-bbaa, i've rebuild dkms from scratch on both nuc and got same issue :

i915-sriov-dkms/2024.12.30, 6.11.0-2-pve, x86_64: installed
i915-sriov-dkms/2024.12.30, 6.11.11-1-pve, x86_64: installed

bbaa-bbaa, 2025-01-21:
Are you try to rebuild your initramfs?

ibis69-git, 2025-01-21:
@bbaa-bbaa working now by not using .deb package

Ch3ck3rM0n, 2025-01-21:
> Hi [@bbaa-bbaa](https://github.com/bbaa-bbaa), i've rebuild dkms from scratch on both nuc and got same issue :
> 
> i915-sriov-dkms/2024.12.30, 6.11.0-2-pve, x86_64: installed i915-sriov-dkms/2024.12.30, 6.11.11-1-pve, x86_64: installed

what kind of repository are you using?
yesterday i got kernel 6.8.12-7-pve as update of my non-prod repo.. there is no 6.11...

ibis69-git, 2025-01-21:
@Ch3ck3rM0n you can opt-in for 6.11 kernel with proxmox 8.3 if you want/need it.

https://forum.proxmox.com/threads/opt-in-linux-6-11-kernel-for-proxmox-ve-8-available-on-test-no-subscription.156818/

tmm1, 2025-01-30:
> Guys, I've been struggling to make this work with the alder lake igpu to no avail.

What issue did you have?

I'm consistently getting Code 43 in Windows 11 guests. Tried on two different Alder Lake machines, with `i915-sriov-dkms/2025.01.22` deb from releases. No change with kernel 6.8 or 6.11

closed 2025-09-05 by johntdavis84