strongtz/i915-sriov-dkms #225: iGPU VF is showing code error 43 on win11 VM

Aayush-1333, 2024-11-24:
I have installed the dims module on my Proxmox host (kernel version 6.8.12-1-pve) and it is working fine with creation of 7 VFs. I had tried to pass the VF on Ubuntu server VM for media transcoding based on the setup steps on the repo. However, I am facing issues when setting this up on Windows 11 Pro (version 23H2 OS build - 22631.4460). I had tried to install latest intel Arc and Xe graphics drivers then I tried to install the older versions as well. But no luck!! It is still showing code error 43. I need a solution on how to make my iGPU VF work properly on this VM? Even it is not showing on task manager due to that. I am using i7-13700K with intel UHD 770 Graphics iGPU.

jaycelq, 2024-11-25:
Did you tick the box next to Primary GPU? For me it works after doing that.

bbaa-bbaa, 2024-11-25:
try to set a vendor id
https://github.com/strongtz/i915-sriov-dkms/issues/8#issuecomment-1567465036
https://github.com/strongtz/i915-sriov-dkms/pull/207#issuecomment-2439406645

Aayush-1333, 2024-11-25:
> Did you tick the box next to Primary GPU? For me it works after doing that.

Yes, I did that but no luck! For a while it showed me it is fine but on opening task manager the iGPU wasn't showing there. Even the intel ARC Control application showed me that the intel API wasn't able to set properly and will run with limited capablilities. I am wondering do I have to extract rom file from BIOS? I haven't tried that but it's quite hard to extract that from my motherboard BIOS. I am using proxmox 8.2.9 as host and win11 as guest VM.

Aayush-1333, 2024-11-25:
> try to set a vendor id [#8 (comment)](https://github.com/strongtz/i915-sriov-dkms/issues/8#issuecomment-1567465036) [#207 (comment)](https://github.com/strongtz/i915-sriov-dkms/pull/207#issuecomment-2439406645)

I already set it to "GenuineIntel" as vendor id in vm config file but still showing errors.

resiliencer, 2024-11-26:
@Aayush-1333 Did you try [to pass](https://github.com/resiliencer/i915-sriov-dkms?tab=readme-ov-file#first-of-all-you-need-to-extract-intelgopdriverefi-from-your-bios)  extracted **IntelGopDriver.efi** to `romfile` parameter of your VM config?   
For me it solved all problems with error n.43 on i5-13600K (UHD 770)

Aayush-1333, 2024-11-26:
@resiliencer Ok so after reading the documentation, I'll do a live boot of ubuntu and from there I'll try to extract the IntelGop romfile because I am using proxmox which is a headless server and building UEFITool binary from source for proxmox needs dependencies which it cannot satisfy (it requires Qt5 or Qt6 library which I cannot do on a headless server like Proxmox). So, I'll work on it and give an update on this.

Aayush-1333, 2024-11-28:
@resiliencer I had already downloaded the BIOS zip file but when I extracted it there wasn't the file I needed for the UEFITool. So, it wasn't possible for me to extract it.

[image]

I am using Gigabyte B760M AORUS Elite AX (rev 1.x)

resiliencer, 2024-11-28:
Did you try to open the first file in UEFITool? Looks like what you need.

Aayush-1333, 2024-11-28:
Yes, this was the file I'll extract from it and try to pass those files as args to the vm. Thanks @resiliencer, honestly I had never tried passing romfiles to the VM for iGPU to work properly. This is my first time tweaking proxmox vm configs to pass these to make it work. Let's hope this works.

resiliencer, 2024-11-28:
@Aayush-1333   

[image]

sha256 checksum:  
`131c32cadb6716dba59d13891bb70213c6ee931dd1e8b1a5593dee6f3a4c2cbd`

sacredx72, 2024-11-29:
@resiliencer Good afternoon, I would like to know if you were able to overcome error 43 in WM Win 10?
 I've been struggling with the same problem for 3 days now =)

I don't really like the presence of errors in the log
`root@server:~# dmesg | grep i915
[    0.000000] Command line: initrd=\EFI\proxmox\6.8.12-4-pve\initrd.img-6.8.12-4-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on pcie_aspm=force pcie_aspm.policy=powersave iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.071992] Kernel command line: initrd=\EFI\proxmox\6.8.12-4-pve\initrd.img-6.8.12-4-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on pcie_aspm=force pcie_aspm.policy=powersave iommu=pt i915.enable_guc=3 i915.max_vfs=7
               use xe.force_probe='46a0' and i915.force_probe='!46a0'
[   15.738082] i915: module verification failed: signature and/or required key missing - tainting kernel
[   16.092364] i915 0000:00:02.0: Running in SR-IOV PF mode
[   16.093902] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   16.107157] i915 0000:00:02.0: vgaarb: deactivate vga console
[   16.107219] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   16.107768] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[   16.110555] i915 0000:00:02.0: [drm] *ERROR* DMC firmware has wrong CSS header length (1097158924 bytes)
[   16.110560] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/adlp_dmc.bin. Disabling runtime power management.
[   16.110563] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   16.119280] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.29.2
[   16.119294] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[   16.122280] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[   16.122893] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   16.122896] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   16.123415] i915 0000:00:02.0: [drm] GuC RC: enabled
[   16.123467] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   16.123470] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   16.124680] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   16.143609] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[   16.145755] i915 0000:00:02.0: 7 VFs could be associated with this PF
[   16.173156] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   16.174627] fbcon: i915drmfb (fb0) is primary device
[   16.245457] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   21.022486] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
               use xe.force_probe='46a0' and i915.force_probe='!46a0'
[   21.025988] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   21.026447] i915 0000:00:02.1: Running in SR-IOV VF mode
[   21.027641] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   21.028394] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   21.030484] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   21.031173] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   21.032578] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   21.033570] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   21.035345] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   21.035860] i915 0000:00:02.1: HuC firmware PRELOADED
[   21.038150] i915 0000:00:02.1: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   21.038729] i915 0000:00:02.1: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   21.042976] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   21.043601] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   21.044539] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[   21.049055] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   21.049931] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46a0' and i915.force_probe='!46a0'
[   21.055041] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   21.055672] i915 0000:00:02.2: Running in SR-IOV VF mode
[   21.056821] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   21.057780] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   21.059745] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   21.060311] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   21.061964] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   21.063515] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   21.065674] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   21.066336] i915 0000:00:02.2: HuC firmware PRELOADED
[   21.067395] i915 0000:00:02.2: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   21.068075] i915 0000:00:02.2: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[... cut ...]
00:02.4 VGA compatible controller: Intel Corporation AlderLake-P GT2
00:02.5 VGA compatible controller: Intel Corporation AlderLake-P GT2
00:02.6 VGA compatible controller: Intel Corporation AlderLake-P GT2
00:02.7 VGA compatible controller: Intel Corporation AlderLake-P GT2
01:00.0 VGA compatible controller: NVIDIA Corporation TU106 [GeForce RTX 2060 Rev. A] (rev a1)`

`root@server:/dev/dri# ls
by-path  card0  card1  card2  card3  card4  card6  card7  renderD128  renderD129  renderD130  renderD131  renderD132  renderD134  renderD135`

sacredx72, 2024-11-29:
In addition, I will add an error log during the installation of the Intel Arc driver in the guest Windows 10. The driver sees the device, reads the correct PID/VID but cannot determine “No Unknown PCI devices were found.”
`2024/11/29 12:46:54.205|INFO|Logging started on 2024/11/29 12:46:54.176 ||
2024/11/29 12:46:54.225|INFO|Installer version: 1.0.691.0 ||
2024/11/29 12:46:54.225|INFO|The application has been launched with the parameters: NULL ||
2024/11/29 12:46:54.225|INFO|Language of the user interface: [ru-RU]. ||
2024/11/29 12:46:54.225|INFO|Operating system specification: Microsoft Windows NT 10.0.19045.0 ||
2024/11/29 12:46:54.320|INFO|CPU base clock: 998Mhz. ||
2024/11/29 12:46:54.320|INFO|Test Signing: OFF. ||
2024/11/29 12:46:54.320|INFO|Windows Media Player status: Installed ||
2024/11/29 12:46:54.320|INFO|Non-Interactive mode flag: False ||
2024/11/29 12:46:54.352|INFO|No Unknown PCI devices were found. ||
2024/11/29 12:46:54.371|INFO|
List of Graphics and Other Devices in Device Manager: {
  {
    Device detected: Intel(R) Iris(R) Xe Graphics
    Version: 31.0.101.3790
    Device Instance Path: PCI\VEN_8086&DEV_46A0&SUBSYS_22128086&REV_00\4&23F93F77&0&00E0
    Device manufacturer: Intel Corporation
  }
  {
    Device detected: Microsoft Remote Display Adapter
    Version: 10.0.19041.3636
    Device Instance Path: SWD\REMOTEDISPLAYENUM\RDPIDD_INDIRECTDISPLAY&SESSIONID_0001
    Device manufacturer: Microsoft
  }} ||
2024/11/29 12:46:54.371|INFO|Any platform is DG2: False ||
2024/11/29 12:46:54.378|INFO|Any platform is Gen12: True ||
2024/11/29 12:46:54.378|INFO|Dynamic background: True ||
2024/11/29 12:46:55.310|INFO|No Unknown PCI devices were found. ||
2024/11/29 12:46:55.310|INFO|
List of Graphics and Other Devices in Device Manager: {
  {
    Device detected: Intel(R) Iris(R) Xe Graphics
    Version: 31.0.101.3790
    Device Instance Path: PCI\VEN_8086&DEV_46A0&SUBSYS_22128086&REV_00\4&23F93F77&0&00E0
    Device manufacturer: Intel Corporation
  }
  {
    Device detected: Microsoft Remote Display Adapter
    Version: 10.0.19041.3636
    Device Instance Path: SWD\REMOTEDISPLAYENUM\RDPIDD_INDIRECTDISPLAY&SESSIONID_0001
    Device manufacturer: Microsoft
  }} ||
2024/11/29 12:46:56.858|INFO|
Found Intel INF files in DriverStore location:
{
	"C:\Windows\System32\DriverStore\FileRepository\cui_dch.inf_amd64_2c633e09ed7efad2\cui_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\cui_dch.inf_amd64_ea08a995fd015b2b\cui_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_48972f09ccb3f257\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_fdd58a2337bf0052\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\iai2c.inf_amd64_a77c815b2999404d\iai2c.inf",
	"C:\Windows\System32\DriverStore\FileRepository\ialpss2i_i2c_skl.inf_amd64_9d9dbb01837eba23\iaLPSS2i_I2C_SKL.inf",
	"C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_a0594a6d5b4e7b3f\iigd_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\mshdadac.inf_amd64_ea3e622713cdff6f\mshdadac.inf",
	"C:\Windows\System32\DriverStore\FileRepository\cui_dch.inf_amd64_bdc7deb712385210\cui_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_04c4cbf8d836b6b3\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_906e69d45f9f9936\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\igcc_dch.inf_amd64_5d8a0e28acd259bd\igcc_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\igcc_dch.inf_amd64_8b07a004bf9bbf23\igcc_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\igcc_dch.inf_amd64_9b33d3c65380dd2c\igcc_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\igcc_dch.inf_amd64_dcf6d1e5af74793f\igcc_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\iigd_ext.inf_amd64_512239646935ef97\iigd_ext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\mshdadac.inf_amd64_3ba55ad295ff3fda\mshdadac.inf",
	"C:\Windows\System32\DriverStore\FileRepository\intcdaud.inf_amd64_5d859c98b498348f\intcdaud.inf",
	"C:\Windows\System32\DriverStore\FileRepository\mshdadac.inf_amd64_c7c38bbdddc4b71f\mshdadac.inf",
	"C:\Windows\System32\DriverStore\FileRepository\cui_dch.inf_amd64_8f68d4f94f46f7aa\cui_dch.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_7fb836d0f4cb8e9d\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\hdbusext.inf_amd64_bc2e5f140fc48972\hdbusext.inf",
	"C:\Windows\System32\DriverStore\FileRepository\ialpssi_i2c.inf_amd64_8e00e1aed7fbdf70\ialpssi_i2c.inf",
	"C:\Windows\System32\DriverStore\FileRepository\igcc_dch.inf_amd64_29c3904e4a4a4d95\igcc_dch.inf",
[... 52 lines cut ...]
      Device instance path: PCI\VEN_8086&DEV_46A0&SUBSYS_22128086&REV_00\4&23F93F77&0&00E0
      Device manufacturer: Intel Corporation
    }
    Matching Driver Store INF:
    {
      INF file path: "C:\Windows\System32\DriverStore\FileRepository\iigd_dch_d.inf_amd64_5ab13f0cc760181d\iigd_dch_d.inf"
      INF file date: "10/17/2022"
      INF file version: "31.0.101.3790"
      INF Models Section: "NULL"
      INF Models Entry: "NULL"
    }
    Matching Installer INF: "NULL"
  }
}
 ||
2024/11/29 12:46:56.895|INFO|No Media devices were found. ||
2024/11/29 12:46:56.895|INFO|No Unknown PCI devices were found. ||
2024/11/29 12:46:56.895|INFO|

List of found Intel Audio and Other devices with installer matching INF files:
{
}
 ||
2024/11/29 12:46:56.983|ERROR|Could not find driver file. ||
`

Aayush-1333, 2024-11-29:
@resiliencer Even after passing the romfile it is showing error: 

I am pasting the error along with my VM config

[image]

[image]

======== VM config ========
affinity: 0-5
agent: 1
args: -cpu 'host,+kvm_pv_unhalt,+kvm_pv_eoi,hv_vendor_id=GenuineIntel,kvm=off'
balloon: 0
bios: ovmf
boot: order=scsi0;net0
cores: 6
cpu: host,hidden=1,flags=+pcid
efidisk0: nvme-thinpool:vm-102-disk-1,efitype=4m,pre-enrolled-keys=1,size=4M
hostpci0: mapping=vGPU-pool,pcie=1,romfile=./IntelGopDriver.bin,x-vga=1
machine: pc-q35-9.0
memory: 12288
meta: creation-qemu=9.0.2,ctime=1723475504
name: win11-pro
net0: virtio=BC:24:11:01:69:7C,bridge=vmbr0
numa: 0
ostype: win11
scsi0: nvme-thinpool:vm-102-disk-0,cache=writeback,discard=on,iothread=1,size=650G,ssd=1
scsihw: virtio-scsi-single
smbios1: uuid=473b6948-bab3-4dc5-b18d-01a24d9055c6,manufacturer=R2lnYWJ5dGUgVGVjaG5vbG9neSBDby4sIEx0ZC4=,product=Qjc2ME0gQU9SVVMgRUxJVEUgQVg=,version=RjE2,serial=U04yNDA5NTAwNDk1NjE=,sku=RGVmYXVsdCBzdHJpbmc=,family=Qjc2MCBNQg==,base64=1
sockets: 1
tablet: 0
tags: game_and_productivity
tpmstate0: nvme-thinpool:vm-102-disk-2,size=4M,version=v2.0
usb0: mapping=bluetooth
usb1: mapping=eksa-mouse
usb2: mapping=cosmic-byte-keyboard
vmgenid: 857655b7-267b-4e4f-b66b-ea93f3f8d1d2

Note - I have tried renaming it as IntelGopDriver.efi but it didn't work. I tried to extract the body but it is getting saved as a .bin file every time not a .efi file

sacredx72, 2024-11-29:
Um.....post me if I didn't fully understand you, English is not my main language. I am currently using Linux 6.8.12-4-pve (2024-11-06T15:04Z) Proxmox 8.3.0
Compilation and installation steps were performed according to the points specified in #https://github.com/strongtz/i915-sriov-dkms?tab=readme-ov-file#pve-host-installation-steps-tested-kernel-65-and- 68
 with the indication "sriov-dkms -v $(cat VERSION) --force" or "dkms install -m i915-sriov-dkms -v 2024.09.21 --force" the result is the same in both

sacredx72, 2024-11-29:
> Note - I have tried renaming it as IntelGopDriver.efi but it didn't work. I tried to extract the body but it is getting saved as a .bin file every time not a .efi file

I have exactly the same problems/questions when using the UEFITools program, according to the instructions from https://github.com/Kethen/edk2-build-intel-gop?tab=readme-ov-file#extracting-intelgopdriverefi-and-vbtbin -from-your-bios
When retrieving a search result for IntelGopDriver.efi, the program suggests saving the result in *.bin format. And when extracting the result for Vbt.bin, the program, on the contrary, will save the file in *.efi =) it seems we are doing something wrong, or the instructions are outdated =)
 also on the ubuntu virtual machine I did built files can be found in edk2/Build/OvmfX64/DEBUG_GCC5/FV/, OVMF_CODE.fd and OVMF_VAR.fd for non-secure boot builds . And again the question is, what should I do with these two files in .fd format? For Vbios I need only one file in *.rom format =) with each step more and more questions arise and so few answers =) yes I’m a noob, but I’m trying , forgive me =)

resiliencer, 2024-11-29:
@Aayush-1333 @sacredx72 It doesn't matter what is the name of file, you can save it with absolutely any name, the main thing is to specify correct full path to it in the VM settings, for example, /root/igpu/intel_whatever_any_name.absolutely. How to extract this driver I wrote [here](https://github.com/resiliencer/i915-sriov-dkms?tab=readme-ov-file#first-of-all-you-need-to-extract-intelgopdriverefi-from-your-bios). No vbios, vbt, x-vga, etc. are needed, this is for full passthru, not sr-iov, better look at the settings and configs in my repository. No need to build OVMF, use the standard 4 MB. The Linux kernels that worked for me without problems - I indicated in the post above and in the repository, this is Arch Linux. I can't tell you about Prokmox, I don't use it. I would also recommend checking the tested working commits I linked to above.

resiliencer, 2024-11-29:
@sacredx72 `romfile=../../../usr/share/kvm/OVMF.fd` - not that one, write here full path of extracted intelgop file.

sacredx72, 2024-11-29:
> As what I see... why are you hiding cpu, I would remove cpu hiding, kvm=off, x-vga=1... And use full path for intelGOP file (your device didn't init correctly, what is the path of running that VM, did you place extracted intelgop file there?If you are using current directory path in config.

Alas, still error 34, I completely cleared the config file of unnecessary parameters
`agent: 1
##args: -cpu 'host,hv_vendor_id=GenuineIntel'
##,+invtsc,+kvm_pv_unhalt,+kvm_pv_eoi,host-phys-bits-limit=37'
bios: ovmf
boot: order=scsi0;net0
cores: 6
cpu: host
efidisk0: zNVME512:vm-103-disk-0,efitype=4m,size=1M
hostpci0: 0000:00:02.7,romfile=../../../usr/share/kvm/1280P.bin
machine: pc-q35-9.0
memory: 8048
meta: creation-qemu=8.1.5,ctime=1713876876
name: Win10F
net0: e1000=BC:24:11:54:90:1E,bridge=vmbr0
numa: 0
ostype: win10
scsi0: zNVME512:vm-103-disk-1,iothread=1,size=70G,ssd=1
scsihw: virtio-scsi-single
smbios1: uuid=14b2ef2e-6e71-4755-be52-213dd3726d88
sockets: 1
tpmstate0: zNVME512:vm-103-disk-2,size=4M,version=v2.0
vga: virtio
vmgenid: df93268e-0aad-45b6-8a1d-91653ca58426`

WM 10 starts, it is displayed that the video card is Intel XE but error 43 is active, when trying to install Intel drivers through the exe there is still error 8 =(
As for the path to IntelGopDriver.efi (after extraction it is called 1280P.bin, it is correct, if you specify /usr/share/kvm/1280P.bin then an error is thrown, the file was not found. But with "romfile=../. ./../usr/share/kvm/1280P.bin" The VM starts without errors.
I ALSO got rid of errors related to the version on the host

`dmesg | grep i915
[    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-6-pve\initrd.img-6.5.13-6-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on pcie_aspm=force pcie_aspm.policy=powersave iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.073885] Kernel command line: initrd=\EFI\proxmox\6.5.13-6-pve\initrd.img-6.5.13-6-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on pcie_aspm=force pcie_aspm.policy=powersave iommu=pt i915.enable_guc=3 i915.max_vfs=7
[   15.841954] i915_compat: module verification failed: signature and/or required key missing - tainting kernel
[   16.237211] i915 0000:00:02.0: Running in SR-IOV PF mode
[   16.329383] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   16.329856] i915 0000:00:02.0: vgaarb: deactivate vga console
[   16.329905] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   16.330493] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[   16.334838] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[   16.341444] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.29.2
[   16.341460] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[   16.344222] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[   16.344641] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   16.344643] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   16.345077] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[   16.345126] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   16.345130] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   16.376581] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[   16.378567] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   16.378868] i915 display info: display version: 13
[   16.378870] i915 display info: cursor_needs_physical: no
[   16.378871] i915 display info: has_cdclk_crawl: yes
[   16.378873] i915 display info: has_cdclk_squash: no
[   16.378874] i915 display info: has_ddi: yes
[   16.378875] i915 display info: has_dp_mst: yes
[   16.378875] i915 display info: has_dsb: yes
[   16.378876] i915 display info: has_fpga_dbg: yes
[   16.378876] i915 display info: has_gmch: no
[   16.378877] i915 display info: has_hotplug: yes
[   16.378877] i915 display info: has_hti: no
[   16.378878] i915 display info: has_ipc: yes
[   16.378878] i915 display info: has_overlay: no
[   16.378879] i915 display info: has_psr: yes
[   16.378879] i915 display info: has_psr_hw_tracking: no
[   16.378880] i915 display info: overlay_needs_physical: no
[   16.378880] i915 display info: supports_tv: no
[   16.378881] i915 display info: has_hdcp: yes
[   16.378882] i915 display info: has_dmc: yes
[... 107 lines cut ...]
[   22.232558] i915 0000:00:02.4: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   22.232951] i915 0000:00:02.5: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   22.233340] i915 0000:00:02.6: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   22.233833] i915 0000:00:02.7: enabling device (0000 -> 0002)
[   22.234297] i915 0000:00:02.7: Running in SR-IOV VF mode
[   22.235351] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.13.4
[   22.236683] i915 0000:00:02.7: [drm] VT-d active for gfx access
[   22.237398] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[   22.238172] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.13.4
[   22.238922] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.13.4
[   22.240062] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   22.240446] i915 0000:00:02.7: HuC firmware PRELOADED
[   22.241289] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[... cut ...]
But the errors that were discussed in other topics "[ 16.345126] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[ 16.345130] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT" could not be resolved =( Our information is here https://www.amrzs.net/2024/11 /03/pve_sriov_gpu/ , the same errors on the screen, and the installation helped "intel-gpu-i915-backports". I also installed it, but the errors are still there =( I don’t even know what the problem is anymore

resiliencer, 2024-11-29:
@sacredx72 
Too new firmware files...

> When retrieving a search result for IntelGopDriver.efi  

Did you search by string or by hex? The right one I mentioned [here](https://github.com/resiliencer/i915-sriov-dkms?tab=readme-ov-file#first-of-all-you-need-to-extract-intelgopdriverefi-from-your-bios)

So... try to build from **TESTED AND WORKING** commits that were mentioned [here](https://github.com/strongtz/i915-sriov-dkms/issues/225#issuecomment-2507423993)
and RTFM [here](https://github.com/resiliencer/i915-sriov-dkms) 
Try just the **SAME** configuration, check it whether it works at all and then figure out what's wrong with that proxmox.

And why copy the logs like that? They are practically unreadable anyway and take up a lot of space, and lines are broken.  And then if it works you could do investigation on proxmox kernel or something.

Aayush-1333, 2024-11-30:
Thank god! It worked!! By removing the unnecessary parameters from cpu and deleting x-vga=1 option in VM config, now it is working fine. I also mentioned the exact path (romfile=/usr/share/kvm/IntelGopDriver.bin) in the VM config file for this to work. Thanks for the advice @resiliencer @sacredx72

sacredx72, 2024-11-30:
@Aayush-1333 Congratulations =) at least you succeeded, but I still can’t do it =) 
You can write here your current output of the command "dmesg | grep i915" in full, indicating the firmware versions. I will also ask you to send me your final configuration file for VM Win 10 =) 
Another question, did you install a specific version of the Intel driver yourself in the VM, or through Windows automatic updates? =)

sacredx72, 2024-11-30:
> Too new firmware files...
Do you have the firmware version listed in 
[drm] DMC i915/adlp_dmc.bin (v2.20) firmware download completed
[drm] GT0: firmware GuC i915/adlp_guc_70.bin version 70.29.2
[drm] GT0: HuC i915/tgl_huc.bin firmware version 7.9.3
 ? 
I didn’t find anywhere instructions for downgrading or indicating the required version to flash in order to downgrade it

resiliencer, 2024-12-01:
@sacredx72 
How can you solve the problem if you didn't even try to build from another commit as advised? Or didn't mention it so we are supposed to guess. What is the reason of doing all of that? 
Firmware usually is in the package `linux-firmware`.
For your current versions of firmware I think you could try 6.11+ kernel.
Custom processor is not very good suited for custom linux kernel module...)))

sacredx72, 2024-12-01:
I'm ready to do what you ask, but I probably just don't understand what I'm doing wrong =) I initially followed all the steps indicated in the article https://www.derekseaman.com/2024/07/proxmox-ve- 8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html
but the result was not successful =) I tried to manually change the DMC i915 firmware version, but after rebooting the system said that the file length is incorrect and that I need to download a different firmware version.  Before each installation of the main version or forks, I follow the steps 
apt update && apt install git sysfsutils pve-headers mokutil -y
rm -rf /usr/src/i915-sriov-dkms-*
rm -rf /var/lib/dkms/i915-sriov-dkms
find /lib/modules -regex ".*/updates/dkms/i915.ko" -delete

resiliencer, 2024-12-01:
@sacredx72 
May be you are doing all right and it's just hardware o smth... but if you are trying to solve this issue systematically and methodically why do not try at least suggested variants? 
And you potentially will know if it is kernel/module issue or it is hardware problem. 
I mentioned many times commits that were successfully tested but if you don't want to use 6.11 kernel, may be you could use 6.1.60-lts kernel version? But you will have to downgrade firmware and take appropriate commit for that kernel. Firmware versions you can also find there in readme if you'll checkout that version. 

https://xyproblem.info/
https://github.com/strongtz/i915-sriov-dkms/issues/225#issuecomment-2508711494
https://github.com/strongtz/i915-sriov-dkms/issues/225#issuecomment-2507423993
https://github.com/resiliencer/i915-sriov-dkms
https://github.com/bbaa-bbaa/i915-sriov-dkms?tab=readme-ov-file#windows-guest
https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/

emre1393, 2024-12-29:
I was having the same problem after proxmox kernel 6.8.12-5-pve update. tried older kernel, different fork of this repo from bbaa. didn't work.
I had to dowgrade i915 firmware via pve-firmware 3.13-3 package, then installed proxmox 6.11-2 kernel and installed dkms module again today when i see the new pr merge, version 28-12-2024.

now it WORKS.   

Intel i5-12500/uhd770

windows 11 24h2 guest works,  u22 server (with 6.8-50 kernel) guest works, jellyfin/plex via docker on same u22 guest works.

[image]

[image]

```
root@proxmox:~# uname -r
6.11.0-2-pve
```

```
root@proxmox:~# dkms status
i915-sriov-dkms/2024.12.28, 6.11.0-2-pve, x86_64: installed
i915-sriov-dkms/2024.12.28, 6.8.12-2-pve, x86_64: installed
```
```
root@proxmox:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.11.0-2-pve root=/dev/mapper/pve-root ro quite module_blacklist=xe intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=5
[    0.055425] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.11.0-2-pve root=/dev/mapper/pve-root ro quite module_blacklist=xe intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=5
[    5.632293] i915: module verification failed: signature and/or required key missing - tainting kernel
[    5.964373] i915 0000:00:02.0: [drm] Found ALDERLAKE_S (device ID 4690) display version 12.00 stepping C0
[    5.964399] i915 0000:00:02.0: Running in SR-IOV PF mode
[    5.964787] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    5.964792] i915 0000:00:02.0: vgaarb: deactivate vga console
[    5.964840] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    5.965682] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    5.967846] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    5.972176] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.29.2
[    5.972183] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    5.974350] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[    5.974677] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    5.974679] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    5.975011] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    5.976231] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    5.976354] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.976359] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    5.982090] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    5.997979] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    5.998054] i915 0000:00:02.0: 5 VFs could be associated with this PF
[    6.013046] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    6.043002] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    6.600076] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[    6.600123] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    6.600146] i915 0000:00:02.1: [drm] Found ALDERLAKE_S (device ID 4690) display version 12.00 stepping C0
[    6.600167] i915 0000:00:02.1: Running in SR-IOV VF mode
[    6.601083] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    6.602073] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    6.602090] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    6.602298] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    6.602651] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    6.603085] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    6.603088] i915 0000:00:02.1: HuC firmware PRELOADED
[    6.604659] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.604662] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    6.604712] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 1
```

AnakTeka, 2025-01-02:
Hi @resiliencer thanks a lot for your guide! I followed yours and have successfully create a VM and Looking Glass.

The thing is, I occasionally have black screen (especially if a UAC screen prompt appear and other 3D needed stuff) on the Looking Glass output and have to reboot either VM or the host to get the display output back. Wondering if you have similar issue?

I'm on
- 13th Gen Intel i7-1355U
- Intel Raptor Lake-P [UHD Graphics]

resiliencer, 2025-01-04:
> Hi @resiliencer thanks a lot for your guide! I followed yours and have successfully create a VM and Looking Glass.
> 
> The thing is, I occasionally have black screen or a freeze screen (especially if a UAC screen prompt appear and other 3D needed stuff) on the Looking Glass output and have to reboot either VM or the host to get the display output back. Wondering if you have similar issue?
> 
> I'm on
> 
> * 13th Gen Intel i7-1355U
> * Intel Raptor Lake-P [UHD Graphics]

I haven't encountered this particular problem, but sometimes it happened that after a few minutes the screen went dark due to a sleep timeout(I mean situation when just only screen is turning off), you click on the keyboard buttons - but the screen doesn't turn on... it was treated quite simply: you need to turn off and turn on the monitor with the power button on the monitor itself - and everything will work :D

Perhaps it's worth looking in the repositories for such a problem, because there were problems with the host falling asleep and waking up... My host falls asleep and wakes up without problems when all virtual machines are paused.

Well, you can also look at more stable BIOS settings, in particular, the minimum power in idle mode... Because I remember there was a problem with insufficient power with the i7-2700K and the voltage on the GPU had to be raised by +100 mV to avoid such jokes.

aur3l14no, 2025-01-25:
> I was having the same problem after proxmox kernel 6.8.12-5-pve update. tried older kernel, different fork of this repo from bbaa. didn't work. I had to dowgrade i915 firmware via pve-firmware [...]

Thank you so much for sharing this. I can confirm that 6.11 does fix issues for some scenarios.

I was able to use Windows 11 VM with 6.8 until a recent driver upgrade. I re-installed latest kernel module without success. Thanks to your information, I upgraded kernel to 6.11 and fixed it!

- cpu: 12500T
- kernel: 6.11.11-1-pve
- driver in windows: 32.0.101.6127

aur3l14no, 2025-02-28:
@resiliencer Thank you for your guide. Unfortunately I could not extract vbios from vendor-provided BIOS file, as there is no match when I search for the hex strings you provided. Is there any other way to get that efi file?

resiliencer, 2025-02-28:
> [@resiliencer](https://github.com/resiliencer) Thank you for your guide. Unfortunately I could not extract vbios from vendor-provided BIOS file, as there is no match when I search for the hex strings you provided. Is there any other way to get that efi file?

Are you sure what you are looking for in the unpacked BIOS image? What is your MB and CPU?
Theoretically, you can extract from suitable other BIOS images from other motherboards, but at your own risk :-)
And compare sha256sum with those mentioned somewhere here, for example:
For desktop with UHD730 and UHD770: `131c32cadb6716dba59d13891bb70213c6ee931dd1e8b1a5593dee6f3a4c2cbd`
For ADL-N: `FA12486D93BEE383AD4D3719015EFAD09FC03352382F17C63DF10B626755954B`

aur3l14no, 2025-02-28:
> Are you sure what you are looking for in the unpacked BIOS image? What is your MB and CPU? Theoretically, you can extract from suitable other BIOS images from other motherboards, but at your own [...]

Machine is Thinkstation P3 Tiny with 14600.

Well I tried the efi file from `system76/firmware-open` repo. Checksum matches. But still got 43.

Also something strange happens. I managed to get _one_ VM working without error. This VM was migrated from another Proxmox VE, 12500T, no vbios used, latest xe driver. It works well on that 12500 node, and it works as well on 14600 node.

Could it be that I messed up with the VM that got 43 error? Maybe, but I've also tried spinning up new VMs, with different machine types and all. Those all failed. I also tried rebooting host machine at least 3 times. No luck as well.

So, it appears to me that Windows is wrongly assuming the device is malfunctioning when in fact it is working pretty fine. And in that migrated VM, some kind of state "this UHD770 is working fine, don't check again" is carried over? I'm not sure. It's just so weird.

resiliencer, 2025-02-28:
You could try to fully clone that VM but wipe its storage. And try fresh windows install) And if it works you could share you config =)
The main problem usally is incorrect initialization of GPU on stage when you get on display with bios etc. i try this to solve with rom file.
but currently i am able to boot win11 without rom file lol ) idk exactly why, but i think the thing is in what i ve been talking about the year (rom+correct config)

aur3l14no, 2025-02-28:
tldr; `ostype` has to be Windows!

> You could try to fully clone that VM but wipe its storage. And try fresh windows install) And if it works you could share you config =)

Thanks for the heads up! After diffing the qemu config, I think I find the problem. The good VM has OS Type set to Windows while the bad VM has OS Type set to Linux! It works immediately after I change the bad VM's OS Type to Windows!

I vaguely remember PVE would auto select correct type after selecting the installation ISO, somehow it didn't do it this time. And I didn't notice it... again and again, while I was creating new VMs and trying. I had **all** the OS Type set to Linux.

I thought the `ostype` was just a label or something doesn't matter. Clearly I was wrong.

PVE's wiki says

> ostype: Specify guest operating system. This is used to enable special optimization/features for specific operating systems

Hmm, pretty vague. I asked LLM and it said ostype could affect

> * ACPI/APIC Settings
>  Adjusts Advanced Configuration and Power Interface (ACPI) and Advanced Programmable Interrupt Controller (APIC) settings to ensure proper OS interaction (e.g., graceful shutdowns).
>  * SMBIOS and Hardware Identification
>  Sets SMBIOS metadata (e.g., system manufacturer) to values expected by the OS, which can affect licensing or compatibility checks.
>  * Resource Allocation
>  Windows: May allocate more VRAM for graphical use (e.g., 16MB→128MB).
>  Linux: Uses minimal VRAM unless specified.

I couldn't verify these things. But at least we know that OS Type is really important. :)

resiliencer, 2025-03-02:
> Well I tried the efi file from system76/firmware-open repo. Checksum matches. But still got 43.
Which one? And what checksum is?

@aur3l14no So do you mean that you are able to boot without err43 any windows? Had you tried it with fresh install?

And I mean libvirt or qemu config, proxmox is additional layer, not evetybody uses it.

aur3l14no, 2025-03-02:
I used the one from https://github.com/system76/firmware-open/tree/master/models/bonw15 with sha256sum = 131c32cadb6716dba59d13891bb70213c6ee931dd1e8b1a5593dee6f3a4c2cbd

My experiments have been limited to Win11 on PVE only.

1. Migrated VM, ostype=win11, no vbios - good
2. Fresh VM, ostype=l26, no vbios - bad
3. Fresh VM, ostype=l26, with vbios - bad
4. Fresh VM, ostype=win11, no vbios - good

I understand libvirt or qemu config would be useful for larger audience but I'm not sure how to get them. I could only get these from `qm showcmd <vmid>`

```
# ostype: l26 (linux 2.6 higher)
/usr/bin/kvm ... -cpu host,+kvm_pv_eoi,+kvm_pv_unhalt ...

# ostype: win11
/usr/bin/kvm ... -cpu host,hv_ipi,hv_relaxed,hv_reset,hv_runtime,hv_spinlocks=0x1fff,hv_stimer,hv_synic,hv_time,hv_vapic,hv_vendor_id=proxmox,hv_vpindex,kvm=off,+kvm_pv_eoi,+kvm_pv_unhalt ...
```

Hope this could be of some help.

jaceeeeeecaj, 2025-07-30:
Found this to be working in virsh (wthout rom):

```
...
  <features>
    <acpi/>
    <apic/>
    <hyperv mode='custom'>
      <relaxed state='on'/>
      <vapic state='on'/>
      <spinlocks state='on' retries='8191'/>
      <vpindex state='on'/>
      <runtime state='on'/>
      <synic state='on'/>
      <stimer state='on'/>
      <reset state='on'/>
      <vendor_id state='on' value='proxmox'/>
      <frequencies state='on'/>
      <reenlightenment state='on'/>
      <tlbflush state='on'/>
      <ipi state='on'/>
    </hyperv>
    <kvm>
      <hidden state='on'/>
    </kvm>
  </features>
  <cpu mode='host-passthrough' check='none' migratable='on'/>
  <clock offset='utc'>
    <timer name='rtc' tickpolicy='catchup'/>
    <timer name='pit' tickpolicy='delay'/>
    <timer name='hpet' present='no'/>
    <timer name='hypervclock' present='yes'/>
  </clock>
...
```