strongtz/i915-sriov-dkms #226: Build Failure (Error 2) on Ubuntu Jammy 22.04 6.8.0-49-generic-Based VM?

johntdavis84, 2024-11-25:
I've built the driver in a VM running this Ubuntu-based distro before, but the VM recently updated to a new version of the HWE kernel, so I suspect that might be causing a problem...

The OS:
```
root@DragonOS-DualDisk:~/i915-sriov-dkms# uname -a
Linux DragonOS-DualDisk 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
root@DragonOS-DualDisk:~/i915-sriov-dkms# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    DragonOS FocalX R37.1
Release:        22.04
Codename:       jammy

```

Here's the build log. Unfortunately, I don't understand it well enough to try to guess what's going wrong, except that something's gone sideways trying to build `intel_dp_mst` again.

```
CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dsi_dcs_backlight.o
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_disable_dp’:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:542:43: warning: unused variable ‘new_payload’ [-Wunused-variable]
  542 |         struct drm_dp_mst_atomic_payload *new_payload =
      |                                           ^~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:540:49: warning: unused variable ‘old_payload’ [-Wunused-variable]
  540 |         const struct drm_dp_mst_atomic_payload *old_payload =
      |                                                 ^~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:43:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_mst_enable_dp’:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:48: error: passing argument 2 of ‘drm_dp_add_payload_part2’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |                                                ^~~~~~~~~~~~
      |                                                |
      |                                                struct drm_atomic_state *
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:751:9: note: in expansion of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
  751 |         DRM_DP_ADD_PAYLOAD_PART2(intel_dp, state,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:854:64: note: expected ‘struct drm_dp_mst_atomic_payload *’ but argument is of type ‘struct drm_atomic_state *’
  854 |                              struct drm_dp_mst_atomic_payload *payload);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: error: too many arguments to function ‘drm_dp_add_payload_part2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: note: in definition of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:853:5: note: declared here
  853 | int drm_dp_add_payload_part2(struct drm_dp_mst_topology_mgr *mgr,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.8.0-49-generic/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-49-generic'
```

bbaa-bbaa, 2024-11-25:
add `-DRELEASE_UBUNTU=1` to Makefile
See also: https://github.com/strongtz/i915-sriov-dkms/issues/219#issuecomment-2453120468

johntdavis84, 2024-11-25:
Thanks! I tried this, and it still didn't work. I added some additional debug info over on issue #219 , as my issue is a duplicate. Would you like me to close this one?

johntdavis84, 2024-12-06:
This seems to have been resolved by some combination of the VM's version of Ubuntu Jammy 22.04 updating to the 6.8.0-49 kernel and changes made to @bbaa-bbaa 's version of the driver: 
https://github.com/strongtz/i915-sriov-dkms/issues/219#issuecomment-2509666071

johntdavis84, 2024-12-06:
Closing, solved.

closed 2024-12-06 by johntdavis84