strongtz/i915-sriov-dkms #227: Problem with proxmox kernel 6.8.12-5-pve

neroita, 2024-12-10:
I get that error trying to compile with latest proxmox kernel upgrade:
```
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/intel_runtime_pm.c:246:21: error: too many arguments to function ~@~Xpm_runtime_get_if_active~@~Y
  246 |                 if (pm_runtime_get_if_active(rpm->kdev, ignore_usecount) <= 0)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/intel_runtime_pm.c:29:
./include/linux/pm_runtime.h:75:12: note: declared here   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/intel_runtime_pm.o] Error 1
```
Is there a solution ?

carfanatic93, 2024-12-10:
I got the same error

Ch3ck3rM0n, 2024-12-10:
Same to me, i rolled back to 6.8.12-4-pve till its fixed

emrepolat7, 2024-12-10:
The suggestion from ChatGPT worked for me. I hope it works for you as well! If you encounter any issues, feel free to share them here.

The i915-sriov-dkms module might be incompatible with your current kernel version. You can resolve the issue by modifying the relevant function call in the source code.

File: ~/i915-sriov-dkms/drivers/gpu/drm/i915/intel_runtime_pm.c

Line: 246
Replace the following:
if (pm_runtime_get_if_active(rpm->kdev, ignore_usecount) <= 0)
with:
if (pm_runtime_get_if_active(rpm->kdev) <= 0)

This ensures the pm_runtime_get_if_active function uses only the struct device *dev parameter, aligning with the kernel's function definition.

KyberofChirrut, 2024-12-11:
> The suggestion from ChatGPT worked for me. I hope it works for you as well! If you encounter any issues, feel free to share them here.
>
> The i915-sriov-dkms module might be incompatible with your [...]

greatman!
it worked for me too!thank you!

Ch3ck3rM0n, 2024-12-11:
works perfectly to me, too.. so my win11 vm is running again with gpu support of my 12650h

Reginard, 2024-12-11:
Didn't work for my Alder Lake iGPU :(

FabioEight, 2024-12-12:
Please prepare a pull request with these changes.

neroita, 2024-12-12:
The problem is that with that change I won't compile with kernel lower than -5pve and detect extraversion is not that simple.

rjblake, 2024-12-12:
> Didn't work for my Alder Lake iGPU :(

Didn't work initially for me on Alder Lake (i7-12700 and i5-12500).  On boot hung loading the kernel. Realised I'd forgotten a couple of steps (duh) - don't forget to do the following after update:

```
sudo update-initramfs -u -k 6.8.12-5-pve
sudo update-grub
```

After that, all good and loaded fine

Reginard, 2024-12-12:
I did apply that. The build fails with ...

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.8.12-5-pve -C /lib/modules/6.8.12-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/2024.09.21/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.12-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/make.log for more information.
i915-sriov-dkms/2024.09.21: added

coffee0818, 2024-12-19:
> The suggestion from ChatGPT worked for me. I hope it works for you as well! If you encounter any issues, feel free to share them here.
>
> The i915-sriov-dkms module might be incompatible with your [...]

That works for me. Thank you.
pve6.8.12-5-pve, i5-12500

closed 2024-12-29 by strongtz