strongtz/i915-sriov-dkms #228: Unable to build drivers using dkms on latest proxmox kernel 6.8.12-5-pve (x86_64)

brokenbuddha, 2024-12-10:
I've been having some issues compiling the modules for the lastest proxmox kernel, I have included some information about the issue below. I'm using an older kernel (6.8.12-4-pve) until the issue is resolved, cheers.

root@proxmox:/usr/src/i915-sriov-dkms-2024.09.21# dkms install -m i915-sriov-dkms -v 2024.09.21 -k 6.8.12-5-pve --force
Sign command: /lib/modules/6.8.12-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.8.12-5-pve -C /lib/modules/6.8.12-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/2024.09.21/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.12-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/make.log for more information.
root@proxmox:/usr/src/i915-sriov-dkms-2024.09.21#

[make.log](https://github.com/user-attachments/files/18079354/make.log)

Please let me know if any further information is required.

emrepolat7, 2024-12-10:
The suggestion from ChatGPT worked for me. I hope it works for you as well! If you encounter any issues, feel free to share them here.

The i915-sriov-dkms module might be incompatible with your current kernel version. You can resolve the issue by modifying the relevant function call in the source code.

File: ~/i915-sriov-dkms/drivers/gpu/drm/i915/intel_runtime_pm.c

Line: 246
Replace the following:
if (pm_runtime_get_if_active(rpm->kdev, ignore_usecount) <= 0)
with:
if (pm_runtime_get_if_active(rpm->kdev) <= 0)

This ensures the pm_runtime_get_if_active function uses only the struct device *dev parameter, aligning with the kernel's function definition.

neroita, 2024-12-10:
The code posted works , now I wait the integration on the git to update all nodes in the cluster.

brokenbuddha, 2024-12-10:
Thank! that fixed the problem, greatly appreciated.

root@proxmox:/usr/src/i915-sriov-dkms-2024.09.21# dkms status
i915-sriov-dkms/2024.09.21, 6.8.12-4-pve, x86_64: installed
i915-sriov-dkms/2024.09.21, 6.8.12-5-pve, x86_64: installed
root@proxmox:/usr/src/i915-sriov-dkms-2024.09.21#

closed 2024-12-10 by brokenbuddha

johntdavis84, 2024-12-10:
Do we need to open a pull request for this change?

iamqk, 2024-12-27:
If you already build once the file's location should be:
/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/intel_runtime_pm.c