strongtz/i915-sriov-dkms #230: I can't split the 13900K into multiple vGPUs at Linux 6.8.12-4-pve.

calonye, 2024-12-12:
I can't split the 13900K into multiple vGPUs at Linux 6.8.12-4-pve.

My general configuration and situation is as follows,I have searched a lot of information and basically nothing is missing,but still I can't get more VGPUs,I don't understand what step is missing or wrong.
Please help me.

---
CPU：i9-13900K 
Z690
Linux 6.8.12-4-pve (2024-11-06T15:04Z)
pve-manager/8.3.1/fb48e850ef9dde27
Linux pve 6.8.12-4-pve 

---

`root@pve:~# uname -a
Linux pve 6.8.12-4-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-4 (2024-11-06T15:04Z) x86_64 GNU/Linux
root@pve:~# uname -r
6.8.12-4-pve
root@pve:~# dkms status
i915-sriov-dkms/2024.09.21, 6.8.12-4-pve, x86_64: installed (WARNING! Diff between built and installed module!)
intel-i915-dkms/1.24.1.19.240119.1.nodrm, 6.8.12-4-pve, x86_64: installed
root@pve:~# lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
01:00.0 VGA compatible controller: NVIDIA Corporation AD102 [GeForce RTX 4090] (rev a1)
root@pve:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-4-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.099795] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-4-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
root@pve:~#`

ldejavul, 2024-12-12:
Have you configured the bios for SRIOV?
Are you blacklisting the nvidia drivers from loading? I see you also have a 4090 installed.

calonye, 2024-12-13:
> Have you configured the bios for SRIOV?您是否为 SRIOV 配置了 BIOS？ Are you blacklisting the nvidia drivers from loading? I see you also have a 4090 installed.您是否将 nvidia 驱动程序列入黑名单以阻止加载？我看到您也安装了 4090。
[image]

---yes～ i did~

lichao0223, 2024-12-13:
did you close security boot in bios？

calonye, 2024-12-13:
> did you close security boot in bios？您是否在 BIOS 中关闭了安全启动？

Yes, could it be related to this secure boot?

bbaa-bbaa, 2024-12-13:
Why did you add the i915 module to the blacklist? The driver was added to the blacklist so the module was not loaded.

calonye, 2024-12-13:
> Why did you add the i915 module to the blacklist? The driver was added to the blacklist so the module was not loaded.为什么把 i915 模块加入黑名单呢？该驱动程序已添加到黑名单中，因此该模块未加载。

--
[image]
I've removed,but still no split GPUs..
[image]

closed 2024-12-13 by calonye