strongtz/i915-sriov-dkms #232: pve6.8.12-5-pve 错误

lujp, 2024-12-16:
CC [M] /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_memory_region.o
CC [M] /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_pcode.o
CC [M] /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_pm.o
CC [M] /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_region_ttm.o
CC [M] /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_runtime_pm.o
/var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_runtime_pm.c: In function ‘__intel_runtime_pm_get_if_active’:
/var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_runtime_pm.c:246:21: error: too many arguments to function ‘pm_runtime_get_if_active’
246 | if (pm_runtime_get_if_active(rpm->kdev, ignore_usecount) <= 0)
| ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_runtime_pm.c:29:
./include/linux/pm_runtime.h:75:12: note: declared here
75 | extern int pm_runtime_get_if_active(struct device *dev);
| ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build/drivers/gpu/drm/i915/intel_runtime_pm.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.12-5-pve/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/6.8.12-5/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.12-5-pve'

ifwwww, 2024-12-17:
https://github.com/strongtz/i915-sriov-dkms/issues/227#issuecomment-2532165631

lujp, 2024-12-18:
https://github.com/strongtz/i915-sriov-dkms/issues/227#issuecomment-2532165631

closed 2024-12-18 by lujp