strongtz/i915-sriov-dkms #234: [BUG] Too many arguments in pm_runtime_get_if_active

jmd2851, 2024-12-20:
Hello,
I was attempting to dkms install following [this](https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/) guide, but ran into what seems like a bug. I've attached the error from the make log here:

```/var/lib/dkms/i915-sriov-dkms/6.8/build/drivers/gpu/drm/i915/intel_runtime_pm.c: In function ‘__intel_runtime_pm_get_if_active’:
/var/lib/dkms/i915-sriov-dkms/6.8/build/drivers/gpu/drm/i915/intel_runtime_pm.c:246:21: error: too many arguments to function ‘pm_runtime_get_if_active’
  246 |                 if (pm_runtime_get_if_active(rpm->kdev, ignore_usecount) <= 0)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.8/build/drivers/gpu/drm/i915/intel_runtime_pm.c:29:
./include/linux/pm_runtime.h:75:12: note: declared here
   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.8/build/drivers/gpu/drm/i915/i915_user_extensions.o
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/6.8/build/drivers/gpu/drm/i915/intel_runtime_pm.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.8.12-5-pve/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/6.8/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.12-5-pve'
```

I am not adept at C - I hope someone else is who could take a look at this! Thank you!

jmd2851, 2024-12-20:
Noting that my kernel version is `6.8.12-5-pve`.

Roomba5, 2024-12-21:
check out this issue: https://github.com/strongtz/i915-sriov-dkms/issues/227

jmd2851, 2024-12-21:
Thanks! I did a quick search but did not see that one. I'll close this in favor of that one.

closed 2024-12-21 by jmd2851