strongtz/i915-sriov-dkms #235: Your graphics device 4680 is not officially supported

wangruiqi-git, 2024-12-25:
may I ask what's the wrong
[    2.236552] xe 0000:00:02.0: Your graphics device 4680 is not officially supported
               by xe driver in this kernel version. To force Xe probe,
               use xe.force_probe='4680' and i915.force_probe='!4680'
               module parameters or CONFIG_DRM_XE_FORCE_PROBE='4680' and
               CONFIG_DRM_I915_FORCE_PROBE='!4680' configuration options.
[    2.264452] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.482078] Setting dangerous option enable_guc - tainting kernel
[    2.482629] i915 0000:00:02.0: Running in SR-IOV PF mode
[    2.482932] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.487536] Console: switching to colour dummy device 80x25
[    2.524723] intel_tcc_cooling: TCC Offset locked
[    2.538776] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.538836] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.539434] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.540050] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    2.541282] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    2.545082] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.29.2
[    2.545085] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    2.548147] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    2.548485] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    2.548486] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    2.548837] i915 0000:00:02.0: [drm] GuC RC: enabled
[    2.549343] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    2.549445] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.605130] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    2.605892] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[    2.606240] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input10
[    2.606312] i915 0000:00:02.0: 2 VFs could be associated with this PF
[    2.633486] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.638304] fbcon: i915drmfb (fb0) is primary device
[    2.704133] Console: switching to colour frame buffer device 240x67
[    2.725428] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device

wangruiqi-git, 2024-12-25:
my kernel is 6.8.12-4-pve

closed 2024-12-25 by wangruiqi-git

maur5, 2024-12-25:
@wangruiqi-git did you sort this out? I solved for this at some point in the recent past, and i'm happy to dig up whatever boot/compile/etc. flag it was if i you need it. 

Cheers.

bao3, 2025-02-11:
Same issue

wangruiqi-git, 2025-02-11:
re

> [@wangruiqi-git](https://github.com/wangruiqi-git) did you sort this out? I solved for this at some point in the recent past, and i'm happy to dig up whatever boot/compile/etc. flag it was if i you need it.
> 
> Cheers.

resolved by using another supported kernel

wangruiqi-git, 2025-02-11:
> Same issue

it works after changing to kernel is 6.8.12-1-pve