strongtz/i915-sriov-dkms #237: Error! Bad return status for module build on kernel: 6.8.12-5-pve (x86_64)

tcscat, 2024-12-30:
Sign command: /lib/modules/6.8.12-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.8.12-5-pve -C /lib/modules/6.8.12-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/2024.12.28/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.12-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/make.log for more information.

bbaa-bbaa, 2024-12-30:
Can you upload /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/make.log?

tcscat, 2024-12-30:
> 您可以上传 /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/make.log 吗？

DKMS make.log for i915-sriov-dkms-2024.12.28 for kernel 6.8.12-5-pve (x86_64)
Mon Dec 30 05:49:38 PM CST 2024
make: Entering directory '/usr/src/linux-headers-6.8.12-5-pve'
make[2]: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/configure: Permission denied
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.9.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.9/drm_dp_tunnel.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.10.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.11.o
<command-line>: fatal error: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/include/config.h: No such file or directory
compilation terminated.
<command-line>: fatal error: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/include/config.h: No such file or directory
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.9.o] Error 1
compilation terminated.
make[2]: *** Waiting for unfinished jobs....
<command-line>: fatal error: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/include/config.h: No such file or directory
compilation terminated.
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.9/drm_dp_tunnel.o] Error 1
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.10.o] Error 1
<command-line>: fatal error: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/include/config.h: No such file or directory
compilation terminated.
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/compat/backport-6.11.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.12-5-pve/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.12-5-pve'

bbaa-bbaa, 2024-12-30:
> make[2]: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/configure: Permission denied

It seems that the executable permission for the configure script is missing. Did you get the source code via a zip file?
Can you try `chmod +x /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/configure`?

tcscat, 2024-12-30:
> > make[2]: /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/configure: Permission denied
> 
> It seems that the executable permission for the configure script is missing. Did you get the source code via a zip file? Can you try and ?`chmod +x /var/lib/dkms/i915-sriov-dkms/2024.12.28/source/configure``chmod +x /var/lib/dkms/i915-sriov-dkms/2024.12.28/build/configure`
Thank you! that fixed the problem, I greatly appreciate your help

tcscat, 2024-12-30:
But I'm having a new problem. It looks like it's working, but it's not.
[image]

tcscat, 2024-12-30:
But I'm having a new problem. It looks like it's working, but it's not.

root@pve:~# dkms status
i915-sriov-dkms/2024.12.28, 6.8.12-5-pve, x86_64: installed
root@pve:~# pveversion
pve-manager/8.3.2/3e76eec21c4a14a7 (running kernel: 6.8.12-5-pve)
root@pve:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [Intel Graphics]
root@pve:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3

tcscat, 2024-12-30:
But I'm having a new problem. It looks like it's working, but it's not.

root@pve:~# dkms status
i915-sriov-dkms/2024.12.28, 6.8.12-5-pve, x86_64: installed
root@pve:~# pveversion
pve-manager/8.3.2/3e76eec21c4a14a7 (running kernel: 6.8.12-5-pve)
root@pve:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [Intel Graphics]
root@pve:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3

[image]

tcscat, 2024-12-30:
> But I'm having a new problem. It looks like it's working, but it's not.
>
> root@pve:~# dkms status
> i915-sriov-dkms/2024.12.28, 6.8.12-5-pve, x86_64: installed
> root@pve:~# pveversion [...]

My cpu is intel N150

bbaa-bbaa, 2024-12-30:
Does any error occur when using `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`?

tcscat, 2024-12-30:
> Does any error occur when using `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`?

root@pve:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory

tcscat, 2024-12-30:
> Does any error occur when using `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`?

root@pve:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory
root@pve:~# cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
0

tcscat, 2024-12-30:
> Does any error occur when using `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`?

root@pve:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory
root@pve:~# cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
0

[image]

bbaa-bbaa, 2024-12-30:
Can you upload the output of `dmesg | grep i915` ?

tcscat, 2024-12-30:
> Can you upload the output of `dmesg | grep i915` ?

[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.038865] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe

bbaa-bbaa, 2024-12-30:
Are these the only two lines of logs related to i915? 
Can you check `lsmod | grep i915`?

tcscat, 2024-12-30:
> 这是与 i915 相关的仅有的两行日志吗？您能检查一下吗`lsmod | grep i915`

Typing `lsmod | grep i915` does not produce any output

bbaa-bbaa, 2024-12-30:
try `modprobe i915`

tcscat, 2024-12-30:
> try and check the output of `modprobe i915``dmesg | tail -n 30`

root@pve:~# modprobe i915
root@pve:~# dmesg | tail -n 30
[    5.665868] kauditd_printk_skb: 11 callbacks suppressed
[    5.665872] audit: type=1400 audit(1735558274.674:22): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="/usr/bin/lxc-copy" pid=606 comm="apparmor_parser"
[    5.673000] audit: type=1400 audit(1735558274.682:23): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="/usr/bin/lxc-start" pid=611 comm="apparmor_parser"
[    5.684430] audit: type=1400 audit(1735558274.694:24): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="lxc-container-default" pid=618 comm="apparmor_parser"
[    5.684438] audit: type=1400 audit(1735558274.694:25): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="lxc-container-default-cgns" pid=618 comm="apparmor_parser"
[    5.684441] audit: type=1400 audit(1735558274.694:26): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="lxc-container-default-with-mounting" pid=618 comm="apparmor_parser"
[    5.684444] audit: type=1400 audit(1735558274.694:27): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="lxc-container-default-with-nesting" pid=618 comm="apparmor_parser"
[   10.718279] igc 0000:02:00.0 enp2s0: PHC removed
[   10.912184] igc 0000:03:00.0 enp3s0: PHC removed
[   11.819132] tap100i0: entered promiscuous mode
[   11.858960] vmbr0: port 2(fwpr100p0) entered blocking state
[   11.858965] vmbr0: port 2(fwpr100p0) entered disabled state
[   11.858980] fwpr100p0: entered allmulticast mode
[   11.859018] fwpr100p0: entered promiscuous mode
[   11.859056] vmbr0: port 2(fwpr100p0) entered blocking state
[   11.859057] vmbr0: port 2(fwpr100p0) entered forwarding state
[   11.866776] fwbr100i0: port 1(fwln100i0) entered blocking state
[   11.866781] fwbr100i0: port 1(fwln100i0) entered disabled state
[   11.866800] fwln100i0: entered allmulticast mode
[   11.866836] fwln100i0: entered promiscuous mode
[   11.866870] fwbr100i0: port 1(fwln100i0) entered blocking state
[   11.866872] fwbr100i0: port 1(fwln100i0) entered forwarding state
[   11.874598] fwbr100i0: port 2(tap100i0) entered blocking state
[   11.874603] fwbr100i0: port 2(tap100i0) entered disabled state
[   11.874614] tap100i0: entered allmulticast mode
[   11.874673] fwbr100i0: port 2(tap100i0) entered blocking state
[   11.874675] fwbr100i0: port 2(tap100i0) entered forwarding state
[   14.415364] pci 0000:00:1f.3: deferred probe pending: snd_hda_intel: couldn't bind with audio component
[  105.250119] i915: module verification failed: signature and/or required key missing - tainting kernel
[  105.663012] Setting dangerous option enable_guc - tainting kernel

bbaa-bbaa, 2024-12-30:
try `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs` and check the output of `cat /sys/kernel/debug/dri/0000:00:02.0/i915_sriov_info`

tcscat, 2024-12-30:
> 尝试并检查`echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs``cat /sys/kernel/debug/dri/0000:00:02.0/i915_sriov_info`

root@pve:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory
root@pve:~# cat /sys/kernel/debug/dri/0000:00:02.0/i915_sriov_info
cat: '/sys/kernel/debug/dri/0000:00:02.0/i915_sriov_info': No such file or directory

[image]

bbaa-bbaa, 2024-12-30:
Can you confirm if the i915 kernel driver is loading correctly? Run `lspci -v` and check the `kernel driver in use` line for the 00:02.0 bus.

tcscat, 2024-12-30:
> Can you confirm if the i915 kernel driver is loading correctly? Run and check the line for the 00:02.0 bus.`lspci -v``kernel driver in use`

00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [Intel Graphics] (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Intel Corporation Alder Lake-N [Intel Graphics]
        Flags: bus master, fast devsel, latency 0, IRQ 11, IOMMU group 0
        Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 3000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Capabilities: [320] Single Root I/O Virtualization (SR-IOV)

root@pve:~# kernel driver in use
-bash: kernel: command not found

bbaa-bbaa, 2024-12-30:
It seems the i915 driver is not loading properly. Did you disable Secure Boot in your BIOS?
Can you try `insmod /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915*` to load the module?

tcscat, 2024-12-30:
> It seems the i915 driver is not loading properly. Did you disable Secure Boot in your BIOS? Can you try `insmod /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915*` to load the module?

Yes, I've turned off secure boot

root@pve:~# insmod /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915*
insmod: ERROR: could not insert module /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915.ko: Unknown symbol in module

bbaa-bbaa, 2024-12-30:
> Yes, I've turned off secure boot
>
> root@pve:~# insmod /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915* insmod: ERROR: could not insert module /usr/lib/modules/6.8.12-5-pve/updates/dkms/i915.ko: [...]

Are there any errors in the `dmesg` output?

tcscat, 2024-12-30:
> Are there any errors in the `dmesg` output?

root@pve:~# dmesg
[    0.000000] Linux version 6.8.12-5-pve (build@proxmox) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-5 (2024-12-03T10:26Z) ()
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000005dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000005e000-0x000000000005efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000005f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000042a7afff] usable
[    0.000000] BIOS-e820: [mem 0x0000000042a7b000-0x0000000042a7bfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000042a7c000-0x0000000042ab1fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000042ab2000-0x0000000042ab2fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000042ab3000-0x000000006f9b4fff] usable
[    0.000000] BIOS-e820: [mem 0x000000006f9b5000-0x0000000072ab4fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072ab5000-0x0000000072ba6fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000072ba7000-0x0000000072ff3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000072ff4000-0x0000000073e65fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000073e66000-0x0000000073efefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000073eff000-0x0000000073efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073f00000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000067fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.8 by American Megatrends
[    0.000000] efi: ACPI=0x72fd0000 ACPI 2.0=0x72fd0014 TPMFinalLog=0x72f9f000 SMBIOS=0x73bb9000 SMBIOS 3.0=0x73bb8000 MEMATTR=0x6b3b1018 ESRT=0x6de51918 MOKvar=0x73bea000 
[    0.000000] efi: Remove mem83: MMIO range=[0xc0000000-0xcfffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xc0000000-0xcfffffff] reserved
[    0.000000] efi: Not removing mem84: MMIO range=[0xfe000000-0xfe010fff] (68KB) from e820 map
[    0.000000] efi: Not removing mem85: MMIO range=[0xfec00000-0xfec00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem86: MMIO range=[0xfed00000-0xfed00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem88: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem89: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 3.6.0 present.
[    0.000000] DMI: Default string Default string/Default string, BIOS 5.27 11/12/2024
[    0.000000] tsc: Detected 800.000 MHz processor
[    0.000000] tsc: Detected 806.400 MHz TSC
[    0.000009] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000011] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000021] last_pfn = 0x67fc00 max_arch_pfn = 0x400000000
[    0.000024] MTRR map: 6 entries (3 fixed + 3 variable; max 23), built from 10 variable MTRRs
[    0.000026] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000125] last_pfn = 0x73f00 max_arch_pfn = 0x400000000
[    0.003948] found SMP MP-table at [mem 0x000fce90-0x000fce9f]
[    0.003956] esrt: Reserving ESRT space from 0x000000006de51918 to 0x000000006de519c8.
[    0.003961] e820: update [mem 0x6de51000-0x6de51fff] usable ==> reserved
[    0.003978] Using GB pages for direct mapping
[    0.004373] secureboot: Secure boot disabled
[    0.004374] RAMDISK: [mem 0x30ab1000-0x3454ffff]
[    0.004376] ACPI: Early table checksum verification disabled
[    0.004379] ACPI: RSDP 0x0000000072FD0014 000024 (v02 ALASKA)
[... 1188 lines cut ...]
[  123.211679] i915: Unknown symbol ttm_bo_vm_reserve (err -2)
[  123.211696] i915: Unknown symbol drm_dp_mst_get_port_malloc (err -2)
[  123.211716] i915: Unknown symbol drm_dp_dual_mode_set_tmds_output (err -2)
[  123.211728] i915: Unknown symbol ttm_bo_validate (err -2)
[  123.211921] i915: Unknown symbol drm_dp_dsc_sink_max_slice_count (err -2)
[... cut ...]
[  123.212420] i915: Unknown symbol drm_edp_backlight_set_level (err -2)
[  123.212427] i915: Unknown symbol drm_dp_set_phy_test_pattern (err -2)
[  123.212503] i915: Unknown symbol ttm_pool_alloc (err -2)
[  123.212514] i915: Unknown symbol drm_dsc_dp_pps_header_init (err -2)
[  123.212597] i915: Unknown symbol drm_dp_read_clock_recovery_delay (err -2)
[  123.212704] i915: Unknown symbol ttm_bo_unmap_virtual (err -2)
[  147.007494] hrtimer: interrupt took 25693 ns
[  206.406810] usb 3-5: USB disconnect, device number 2

bbaa-bbaa, 2024-12-30:
try `modprobe i915` and check the output `modinfo i915`

tcscat, 2024-12-30:
> try `modprobe i915` and check the output `modinfo i915`

Thank you very much, I have succeeded by installing intel-i915-dkms 🙂 `https://github.com/moetayuko/intel-gpu-i915-backports/releases`

bbaa-bbaa, 2024-12-30:
> Thank you very much, I have succeeded by installing intel-i915-dkms 🙂 `https://github.com/moetayuko/intel-gpu-i915-backports/releases`

> My cpu is intel N150

Thanks for your information. This repo has been repositioned based on the Linux 6.12 branch. The backporting strategy used did not include the PID information for the GPU in your CPU (0x46d4 was added in Linux 6.9 (https://github.com/moetayuko/intel-gpu-i915-backports/issues/34)). I will create a PR later to backport all new pci IDs to the older kernel version.

PR #238 (2024-12-31): Backport missing pciids for ADL-N CPU
Fix #237
closed 2024-12-31 by strongtz