strongtz/i915-sriov-dkms #244: vaInitialize failed with error code -1 (unknown libva error),exit

aronjanosch, 2025-01-23:
On my kali guest, I'm not able to get the VAAPI to pickup the iGPU. I get this error instead:

```
┌──(kali㉿kali)-[~]
└─$ vainfo 
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.22.0
libva error: vaGetDriverNames() failed with unknown libva error
vaInitialize failed with error code -1 (unknown libva error),exit
```

The VA gets recognized by the kernel:
┌──(kali㉿kali)-[~]
└─$ dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.11.2-amd64 root=UUID=1b5744b1-0d66-4cee-a662-e02b9d97ed08 ro i915.enable_guc=3 splash
[    0.018366] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.11.2-amd64 root=UUID=1b5744b1-0d66-4cee-a662-e02b9d97ed08 ro i915.enable_guc=3 splash
[    1.737934] i915: loading out-of-tree module taints kernel.
[    1.738002] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.033976] i915 0000:06:10.0: [drm] Found ALDERLAKE_P/ADL-N (device ID 46d0) display version 13.00 stepping D0
[    2.034013] i915 0000:06:10.0: Running in SR-IOV VF mode
[    2.034620] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.17.0
[    2.035305] i915 0000:06:10.0: vgaarb: deactivate vga console
[    2.035324] i915 0000:06:10.0: [drm] Using Transparent Hugepages
[    2.036120] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.17.0
[    2.036648] i915 0000:06:10.0: [drm] GT0: GUC: interface version 0.1.17.0
[    2.037782] i915 0000:06:10.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    2.037786] i915 0000:06:10.0: HuC firmware PRELOADED
[    2.041062] i915 0000:06:10.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.041070] i915 0000:06:10.0: [drm] PMU not supported for this GPU.
[    2.041194] [drm] Initialized i915 1.6.0 for 0000:06:10.0 on minor 0

On my proxmox host the vainfo output is as expected:

```
vainfo
error: can't connect to X server!
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

kernel is `6.8.12-1-pve`
and `6.11.2-amd64` for kali.

Any Idea why this is happening?

luxk3, 2025-02-10:
Hello everybody!
I am tagging along on this issue. I am experiencing something very similar:

### My configuration

- On the host: I installed i915-sriov-dkms following the README
- Additionally, I had to add i915.force_probe=7d67 in /etc/default/grub as suggested by an error in dmesg. This is the full string: `GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 i915.force_probe=7d67 module_blacklist=xe"`
- The drivers load correctly and I see 7 VF, and I can I can assign them to virtual machines

### My  problem
- **On the host**: vainfo returns this error (I tried different driver names, none work. The correct one should be iHD):

```root@proxmox:\~# export LIBVA_DRIVER_NAME=iHD
root@proxmox:\~#vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

I am not sure if this on the host is an actual issue though, as I don't really need hw acceleration on the host. I tried both intel-media-va-driver and intel-media-va-driver-non-free, the is no difference.

The issue I am experiencing consists in:
- **hardware acceleration in Windows VM guest works correctly.** I assigned a vgpu to a Windows VM guest and installed the Intel Arc & Iris Graphics driver.
- **hardware acceleration in Linux VM guest does not work.** I assigned a vgpu to a linux guest (I tried Debian, Ubuntu), installed the same i915-sriov-dkms driver, and the intel-media-va-driver. I used the recommended string in /etc/default/grub 
`GRUB_CMDLINE_LINUX_DEFAULT="intel_iommu=on i915.enable_guc=3 module_blacklist=xe"`.

The driver **load  correctly** as I can see this:

```#lspci -k
01:00.0 VGA compatible controller: Intel Corporation Arrow Lake-S [Intel Graphics] (rev 06)
	Subsystem: ASUSTeK Computer Inc. Arrow Lake-S [Intel Graphics]
	Kernel driver in use: i915
	Kernel modules: xe, i915
```
and
```
root@debian:~# ls /dev/dri
by-path  card0  renderD128
```

**However, vainfo fails**. This is my actual issue, as I cannot use hardware acceleration. This is the specific error (different from the one I get on the host):
```l@debian:\~$ export LIBVA_DRIVER_NAME=iHD
l@debian:\~$ vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
### My info

1. Proxmox version:
```
root@proxmox:~# pveversion
pve-manager/8.3.0/c1689ccb1065a83b (running kernel: 6.8.12-8-pve)
```
2. CPU:
```
root@proxmox:~# lscpu | grep "Model name:"
Model name:                           Intel(R) Core(TM) Ultra 5 245K
```

4. Proxmox (Host) dmesg:
```
root@proxmox:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-8-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 i915.force_probe=7d67 module_blacklist=xe
[... 333 lines cut ...]
[    0.011641] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.12.9+bpo-amd64 root=UUID=f397a952-4b26-4e3f-8bf2-a53846c1e3da ro quiet intel_iommu=on i915.enable_guc=3 i915.force_probe=7d67 module_blacklist=xe
[    1.105217] i915: loading out-of-tree module taints kernel.
[    1.105246] i915: module verification failed: signature and/or required key missing - tainting kernel
[    1.260188] i915 0000:01:00.0: Force probing unsupported Device ID 7d67, tainting kernel
[    1.260259] i915 0000:01:00.0: [drm] Cannot read valid display IP version; disabling display.
[    1.260260] i915 0000:01:00.0: [drm] Found METEORLAKE (device ID 7d67) display version 0.00 stepping N/A
[    1.260265] i915 0000:01:00.0: Running in SR-IOV VF mode
[    1.260266] i915 0000:01:00.0: [drm] preliminary graphics version 12.70 media version 13.00
[    1.260766] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.17.0
[    1.260805] i915 0000:01:00.0: [drm] GT0: GMD_ID 0x3118004 version 12.70 step B0
[    1.261787] i915 0000:01:00.0: [drm] GT1: GUC: interface version 0.1.17.0
[    1.261796] i915 0000:01:00.0: [drm] GT1: GMD_ID 0x3400008 version 13.0 step C0
[    1.262250] i915 0000:01:00.0: vgaarb: deactivate vga console
[    1.262262] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    1.263205] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.17.0
[    1.266288] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.17.0
[    1.267052] i915 0000:01:00.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    1.267053] i915 0000:01:00.0: HuC firmware N/A
[    1.271230] i915 0000:01:00.0: [drm] GT1: GUC: interface version 0.1.17.0
[    1.273136] i915 0000:01:00.0: [drm] GT1: GUC: interface version 0.1.17.0
[    1.273371] i915 0000:01:00.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    1.273372] i915 0000:01:00.0: HuC firmware PRELOADED
[    1.276154] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    1.276224] [drm] Initialized i915 1.6.0 for 0000:01:00.0 on minor 0
```

bbaa-bbaa, 2025-02-11:
Wow, you're currently using a very new CPU. It's great to see that Arrow Lake is able to work partially on this driver.

Unfortunately, Intel has shifted its focus to developing XE drivers for the new platform, and SR-IOV support is still incomplete. You might have to wait for an upstream update.

luxk3, 2025-02-15:
Thanks @bbaa-bbaa for the reply. I went for a new CPU indeed, taking a bit of risk.

It took me few days for investigation and trial and erro, but I got it working.
I wanted to update here as well in case someone else has the same issue.

I searched for a newer version of the intel driver. I found out that Debian had a newer package for the `intel-media-va-driver-non-free` in their **testing  and unstable** releases. 
I installed Debian testing and the included intel driver `intel-media-va-driver-non-free` **version 25.1.0** and vainfo does not brake anymore.

The tool now returns the profiles:
```
root@a:# export LIBVA_DRIVER_NAME=iHD
root@a:# vainfo --display x11
Trying display: x11
libva info: VA-API version 1.22.0
libva error: vaGetDriverNames() failed with unknown libva error
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 25.1.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSlice
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSlice
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSlice
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSlice
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSlice
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSlice
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSlice
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSlice
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointEncSlice
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSlice
```

Note: I noticed there is an error in the output saying it cannot get the driver name. I compared that with a healthy output and it seems it cannot programmatically get the name "iHD". I am not sure about the impact of that error. 

Thanks, I hope this could help OP as well.

cjs-github, 2025-06-19:
I'm in the same bucket. Not sure if I'm getting the expected performance from the chip but I did get it kind of working. Same steps/process but also these steps to get vainfo working in the linux VM:

**https://dgpu-docs.intel.com/driver/client/overview.html**

Even assigning all the VFs to the linux VM ffmpeg only gets me ~40fps with hardware acceleration, CPU transcoding I get 100fps+ (Intel(R) Core(TM) Ultra 7 265)