strongtz/i915-sriov-dkms #245: truenas client and unraid client  i915 error

baozaodetudou, 2025-01-24:
I use PVE 8.3 version, and the kernel version is 6.8.12-7.
➜  ~ uname -a
Linux pve 6.8.12-7-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-7 (2025-01-17T08:18Z) x86_64 GNU/Linux

I install the win10 virtual machine and install the driver https://www.intel.com/content/www/ens/download/741626/780560/intel-graphics-windows.html.
But neither my Unraid virtual machine and Truenas virtual machine cannot be used normally

⚡ root@truenas  ~  uname -a
Linux truenas 6.6.44-production+truenas #1 SMP PREEMPT_DYNAMIC Mon Dec 16 20:59:32 UTC 2024 x86_64 GNU/Linux

 ⚡ root@truenas  ~  dmesg | grep i915
[   46.021835] i915 0000:06:10.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
[   46.022610] i915 0000:06:10.0: Device initialization failed (-5)
[   46.022618] i915 0000:06:10.0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[   46.022627] i915: probe of 0000:06:10.0 failed with error -5

I don't know what the problem is.

bbaa-bbaa, 2025-01-24:
You need to install this dkms module in guest.

baozaodetudou, 2025-01-27:
> You need to install this dkms module in guest.

This configuration does not support 6.6.44 kernel

bbaa-bbaa, 2025-01-27:
Use https://github.com/strongtz/i915-sriov-dkms/tree/intel-lts-v6.1 branch instead.