strongtz/i915-sriov-dkms #246: 【需要帮助 HELP NEEDED】failed use in fnOS based on debian

Akimio521, 2025-01-25:
我使用的是飞牛OS（基于debian）：
```bash
root@AkimioNAS:~# uname -a
Linux AkimioNAS 6.6.38-trim #80 SMP PREEMPT_DYNAMIC Tue Dec 10 20:45:43 CST 2024 x86_64 GNU/Linux
```
我参考PVE和手动安装的部分，依然没有成功，希望有大佬可以帮我检查一下
我使用的是CPU是13900H es（代号 q1hy）
一下是我的操作步骤（均使用root用户）
```bash
cd /tmp
curl https://github.com/strongtz/i915-sriov-dkms/releases/download/2025.01.22/i915-sriov-dkms_2025.01.22_amd64.deb -o i915-sriov-dkms_2025.01.22_amd64.deb
apt install ./i915-sriov-dkms_2024.12.30_amd64.deb
```
然后修改 `/etc/default/grub`
`GRUB_CMDLINE_LINUX_DEFAULT="quiet i915.force_probe=7d55 intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7"`
```bash
update-grub
update-initramfs -u
apt update
apt install sysfsutils -y
echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7" > /etc/sysfs.conf
reboot
```
重启完成后0000:00:02总线上依旧只有0号功能

---------
I'm using FnOS (based on Debian):
```bash
root@AkimioNAS:~# uname -a
Linux AkimioNAS 6.6.38-trim #80 SMP PREEMPT_DYNAMIC Tue Dec 10 20:45:43 CST 2024 x86_64 GNU/Linux
I referred to the parts of PVE and manual installation, but still didn't succeed. I hope some experts can help me check.
```
The CPU I'm using is 13900H es (code name q1hy).
The following are my operation steps (all using the root user):
```bash
cd /tmp
curl https://github.com/strongtz/i915-sriov-dkms/releases/download/2025.01.22/i915-sriov-dkms_2025.01.22_amd64.deb -o i915-sriov-dkms_2025.01.22_amd64.deb
apt install./i915-sriov-dkms_2024.12.30_amd64.deb
```
Then modify `/etc/default/grub`
`GRUB_CMDLINE_LINUX_DEFAULT="quiet i915.force_probe=7d55 intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7"`
```bash
update-grub
update-initramfs -u
apt update
apt install sysfsutils -y
echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7" > /etc/sysfs.conf
reboot
```
After the reboot, there is still only function 0 on the 0000:00:02 bus.

bbaa-bbaa, 2025-01-25:
You are using kernel version `6.6.38-trim`, but the current branch only supports kernel versions 6.8 and above.
Please use the intel-6.1 branch instead.

Akimio521, 2025-01-26:
> You are using kernel version `6.6.38-trim`, but the current branch only supports kernel versions 6.8 and above. Please use the https://github.com/strongtz/i915-sriov-dkms/tree/intel-lts-v6.1 branch instead.

感谢你的提醒，我听说编译驱动可能导致污染环境（因为目前我的系统是用作NAS）所以使用了相同的镜像作为虚拟机进行编译，编译完成后在将驱动复制到我的NAS中
以下是我的编译步骤：
```bash
sudo -i
apt-mark unhold libc6:amd64
apt install -y libc6-dev build-essential git dkms
cd /tmp
git clone -b intel-lts-v6.1 https://github.com/strongtz/i915-sriov-dkms.git
cd i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v 2024.09.21 --force
```
完成编译后，我将`/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/i915.ko`复制到NAS的`/lib/modules/6.6.38-trim/updates/dkms/i915.ko`然后进行安装操作
```bash
chmod a+r /lib/modules/6.6.38-trim/updates/dkms/i915.ko
depmod
update-grub
reboot
```
并且在此之前我修改了NAS的`/etc/default/grub`、`/etc/modules`、`/etc/sysfs.conf`，重启后我通过`lspci`查看依然只有0000:00:02.0，没有出现其余的1-7

--------------------------------

Thank you for your reminder. I heard that compiling drivers might cause system pollution (since my current system is used as a NAS), so I used the same image as a virtual machine for compilation. After the compilation was completed, I copied the drivers to my NAS.

Here are my compilation steps:
```bash
sudo -i
apt-mark unhold libc6:amd64
apt install -y libc6-dev build-essential git dkms
cd /tmp
git clone -b intel-lts-v6.1 https://github.com/strongtz/i915-sriov-dkms.git
cd i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v 2024.09.21 --force
```
After the compilation was completed, I copied `/var/lib/dkms/i915-sriov-dkms/2024.09.21/build/i915.ko` to `/lib/modules/6.6.38-trim/updates/dkms/i915.ko` on the NAS and then carried out the installation operation.
```bash
chmod a+r /lib/modules/6.6.38-trim/updates/dkms/i915.ko
depmod
update-grub
reboot
```
And before that, I modified `/etc/default/grub`, `/etc/modules`, and `/etc/sysfs.conf` on the NAS. After restarting, when I checked with `lspci`, there was still only `0000:00:02.0`, and the rest from 1 - 7 did not appear.

bbaa-bbaa, 2025-01-26:
Please provide the `dmesg` log.

Akimio521, 2025-01-26:
> Please provide the `dmesg` log.

```
root@AkimioNAS:~# dmesg
[    0.000000] microcode: updated early: 0x410e -> 0x4121, date = 2023-12-07
[    0.000000] Linux version 6.6.38-trim (root@debian) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #80 SMP PREEMPT_DYNAMIC Tue Dec 10 20:45:43 CST 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.6.38-trim root=UUID=64c2b57b-a030-4ffd-96e3-77b0cbd0c116 ro modprobe.blacklist=pcspkr quiet i915.force_probe=7d55 intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.000000] x86/tme: not enabled by BIOS
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000002effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000002f000000-0x00000000320fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000032100000-0x00000000321ebfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000321ec000-0x0000000032318fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000032319000-0x0000000032e63fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000032e64000-0x0000000032efefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000032eff000-0x0000000032efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000032f00000-0x0000000038ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000039c00000-0x000000003f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000008c07fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.8 by American Megatrends
[    0.000000] efi: ACPI=0x32275000 ACPI 2.0=0x32275014 TPMFinalLog=0x32244000 SMBIOS=0x32bf7000 SMBIOS 3.0=0x32bf6000 ESRT=0x2c77a498
[    0.000000] efi: Remove mem71: MMIO range=[0xc0000000-0xcfffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xc0000000-0xcfffffff] reserved
[    0.000000] efi: Not removing mem72: MMIO range=[0xfe000000-0xfe010fff] (68KB) from e820 map
[    0.000000] efi: Not removing mem73: MMIO range=[0xfec00000-0xfec00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem74: MMIO range=[0xfed00000-0xfed00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem76: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem77: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000000] SMBIOS 3.5.0 present.
[    0.000000] DMI: Intel i9-13900HK(ES)/Z790I PLUS, BIOS Leo-Patch 2.6 08/19/2024
[    0.000000] tsc: Detected 2600.000 MHz processor
[    0.000000] tsc: Detected 2611.200 MHz TSC
[    0.000784] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000785] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000789] last_pfn = 0x8c0800 max_arch_pfn = 0x400000000
[    0.000791] MTRR map: 6 entries (3 fixed + 3 variable; max 23), built from 10 variable MTRRs
[    0.000792] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.001162] e820: update [mem 0x3b000000-0xffffffff] usable ==> reserved
[    0.001164] last_pfn = 0x32f00 max_arch_pfn = 0x400000000
[    0.011439] esrt: Reserving ESRT space from 0x000000002c77a498 to 0x000000002c77a548.
[    0.011442] e820: update [mem 0x2c77a000-0x2c77afff] usable ==> reserved
[    0.011450] Using GB pages for direct mapping
[    0.011538] Secure boot disabled
[    0.011538] RAMDISK: [mem 0x19dc2000-0x1c607fff]
[    0.011540] ACPI: Early table checksum verification disabled
[    0.011541] ACPI: RSDP 0x0000000032275014 000024 (v02 ALASKA)
[    0.011543] ACPI: XSDT 0x0000000032274728 0000F4 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.011546] ACPI: FACP 0x00000000321E9000 000114 (v06 ALASKA A M I    01072009 AMI  01000013)
[    0.011548] ACPI: DSDT 0x000000003215C000 08CC83 (v02 ALASKA A M I    01072009 INTL 20200717)
[    0.011549] ACPI: FACS 0x0000000032318000 000040
[    0.011551] ACPI: FIDT 0x000000003215B000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.011552] ACPI: SSDT 0x00000000321EB000 00038C (v02 PmaxDv Pmax_Dev 00000001 INTL 20200717)
[    0.011553] ACPI: SSDT 0x0000000032155000 005D34 (v02 CpuRef CpuSsdt  00003000 INTL 20200717)
[    0.011554] ACPI: SSDT 0x0000000032152000 002935 (v02 SaSsdt SaSsdt   00003000 INTL 20200717)
[    0.011556] ACPI: SSDT 0x000000003214E000 0033D3 (v02 INTEL  IgfxSsdt 00003000 INTL 20200717)
[... 1193 lines cut ...]
[  267.198583] veth76384bc: entered allmulticast mode
[... cut ...]
[  333.554092] vetha5970c5: entered allmulticast mode
[  333.554126] vetha5970c5: entered promiscuous mode
[  333.554190] br-b31d7bc84a3a: port 1(vetha5970c5) entered blocking state
[  333.554192] br-b31d7bc84a3a: port 1(vetha5970c5) entered forwarding state
[  333.554632] br-b31d7bc84a3a: port 1(vetha5970c5) entered disabled state
[  333.732801] eth0: renamed from veth3b88f7a
[  333.765163] br-b31d7bc84a3a: port 1(vetha5970c5) entered blocking state
[  333.765170] br-b31d7bc84a3a: port 1(vetha5970c5) entered forwarding state
```

bbaa-bbaa, 2025-01-26:
It seems that this module has been loaded correctly. Can you try manually creating a VF using `echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`

Akimio521, 2025-01-26:
It seen not good 
```bash
root@AkimioNAS:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: Cannot allocate memory
root@AkimioNAS:~# cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
0
```

bbaa-bbaa, 2025-01-26:
Please provide the dmesg log.

Akimio521, 2025-01-26:
This is the newly added content in dmesg logs
```
[ 4593.404028] i915 0000:00:02.0: not enough MMIO resources for SR-IOV
[ 4593.500174] i915 0000:00:02.0: [drm] *ERROR* Failed to enable 2 VFs (-ENOMEM)
```

ShmilyStar, 2025-01-26:
请问在fnOS中安装成功看吗

bbaa-bbaa, 2025-01-26:
The MMIO error usually occurs on ES CPUs, and I don't know how to resolve this issue .

Akimio521, 2025-01-26:
OK, thanks for you help,I failed to use sr-iov since I use unRAID. Maybe it was a hardware bug

Akimio521, 2025-01-26:
> Your MB may not support sriov or the feature does not enabled. You can
> check your bios settings and enable sriov.
> Another, you can add pci=assign-busses,realloc into the linux
> commandline.Refer the grub or your OS's bootloader.
> 
> 秋澪Akimio ***@***.***>于2025年1月26日 周日14:23写道：
> […](#)

Yes,you are right,the option VTIO in bios is disable in default. I enable it and also failed. I'm not sure whether the following options are needed.

[image]

Akimio521, 2025-01-26:
> The VTIO looks like iommu, and it is not sriov.
> You should find the sriov switch. Or add the options into linux commandline.
> 
> 秋澪Akimio ***@***.***>于2025年1月26日 周日15:38写道：
> […](#)

Under the default BIOS settings, I tried to add `pci=assign - busses pci=realloc` to GRUB, but still couldn't enable SR - IOV.

Akimio521, 2025-01-26:
It seems that the failure to properly enable SR-IOV is related to the MMIO issue. Should I enable this option?

[image]

Akimio521, 2025-01-26:
Thanks everyone. After I enable the option `Above 4GB MMIO BIOS assignment` I enable the GPU SR-IOV
```bash
$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.1 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.2 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.3 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.4 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.5 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.6 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
00:02.7 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
```

closed 2025-01-26 by Akimio521