strongtz/i915-sriov-dkms #248: Problems installing in Debian with apt

tangent360, 2025-01-26:
Using  `apt install ./i915-sriov-dkms_2024.12.30_amd64.deb` rather than building from source, I ran into problems on several machines running Debian Trixie kernel version 6.12.8 - 6.12.10 where the module would not install correctly or would not displace the existing installed i915 module.

`dksm status` and `lsmod | grep i915` and `apt list --installed | grep i915` each make it appear that the module was correctly installed but `vainfo` would fail and the following errors were shown early in the boot sequence:

`
i915 0000:01:00.0: [drm] Found ALDERLAKE_P/ADL-N (device ID 46d1) display version 13.00 stepping D0
i915 0000:01:00.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
i915 0000:01:00.0: [drm] *ERROR* Device initialization failed (-5)
i915 0000:01:00.0: probe with driver i915 failed with error -5
`

trying `apt --reinstall` does not fix the issue, nor does `apt remove` followed by `apt install`.

### To Recreate:
I'm not sure what's causing it.  It's possible that if you are missing kernel headers of the correct version before starting the `apt install` that the build will fail but not gracefully. But it has happened several times now on different machines.

### Workaround: 
The hard part is realizing what's causing the boot errors because everything looks to be working.
`modprobe -r i915` the 'wrong' module out of the kernel and then `apt remove` the SRI-OV package followed by a fresh install.
You may have to reboot in between steps.
`modinfo i915 | grep -r filename -e version` to make sure you have the correct module:

correct:

`filename:       /lib/modules/6.12.10-amd64/updates/dkms/i915.ko.xz
version:        2025.01.22-sriov.6.12.10`

legacy in my install has no version number and uses the following file:

`filename:       /lib/modules/6.12.9-amd64/kernel/drivers/gpu/drm/i915/i915.ko.xz`

Alternatively try building from source or, if not already installed, install the correct headers and the dkms system will kick of a rebuild of the module and you maybe be able to `modprobe` it into the kernel.

bbaa-bbaa, 2025-01-26:
The 2024.12.30 version lacks a `postinst` script to actively trigger an initramfs update. In some cases, initramfs may not be updated by apt, which could result in initramfs still containing the original modules from the kernel.

Should have been fixed by https://github.com/strongtz/i915-sriov-dkms/commit/000066479bdce87c3e20c5e7f2a3a2e85043c754

tangent360, 2025-01-26:
Sorry.  Mistake in my initial post … copied the wrong thing and it should have read 2025.01.22  (edited to correct). I also ran ‘update-initramfs -u -k all’ after the install script ran thinking that might be the issue, even though I could see the initramfs step being run by the installer.

bbaa-bbaa, 2025-01-27:
The dh-dkms helper script will only perform dkms install for the current kernel and any future kernels after the DKMS module installed. 
Perhaps you updated to a new kernel before installing this module, which caused dh-dkms to miss the compilation for that kernel.

> filename:       /lib/modules/6.12.10-amd64/updates/dkms/i915.ko.xz
filename:       /lib/modules/6.12.9-amd64/kernel/drivers/gpu/drm/i915/i915.ko.xz

It seems that a kernel update from 6.12.9 to 6.12.10 occurred before installing the DKMS module.

tangent360, 2025-01-27:
Kernel wasn't updated - that text is actually from two different machines on different kernel versions just to demonstrate the `modinfo` output from a successful install vs the output from an unsuccessful install.  I installed it on several different VM's and had issues on 3 of them (and they have different kernel version). 

In any event, if the kernel was updated from 6.12.9 to 6.12.10 before installing the DKMS module isn't that fine so long as I stay using 6.12.10 or later?

I can try to reproduce off an old VM snapshot if helpful but I'm not sure the conditions that were leading to failure here.  
Worked fine first time on some machines.

bbaa-bbaa, 2025-01-28:
If you do not reboot and switch to the new kernel after a kernel update and immediately install the corresponding kernel module, dkms may miss the installation for the new kernel.

The helper script only perform dkms install for the **current kernel** and any future kernels after the DKMS module installed.

Let helper script install for all kernels during the first configuration to avoid this issue. https://github.com/strongtz/i915-sriov-dkms/commit/f5db912d92a9e9b1449e0bccc6781c74cc78c4a3

tangent360, 2025-01-28:
### To reproduce:

1. Fresh Debian trixie install, latest kernel, no headers installed.  Clean reboot.
2. Download .deb package and run `sudo apt install ./i915-sriov-dkms_2025.01.22_amd64.deb`
3. I looks like it completes successfully but, if the headers are missing, it displays the following warning during the build:
```
Module build for kernel 6.12.10-amd64 was skipped since the
kernel headers for this kernel do not seem to be installed.
```
[initial_install_log.txt](https://github.com/user-attachments/files/18567565/initial_install_log.txt)

4. On reboot the error messages are displayd on startup, as per OP.   `lsmod` and `dksm status` make it look like the module is installed correctly.  Re-running installer does nothing, even with --reinstall.  `modinfo` will reveal the module is not the SIR-OV version.
5. Install the correct kernel headers and this triggers a DKMS rebuild as part of the header install process:
```
Setting up linux-headers-6.12.10-amd64 (6.12.10-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.12.10-amd64.
```
[header_install_log.txt](https://github.com/user-attachments/files/18567566/header_install_log.txt)

6. This builds the module but does not install it.  
One must run `sudo apt --reinstall install ./i915-sriov-dkms_2025.01.22_amd64.deb` to force the correct installation of the module at this point.

**Two Suggestions:**
1. Error out of the build or install the headers as pre-requisites if the correct headers for the kernel are not present.
2. Decide/implement the behavior if a new kernel is installed with/without headers.  It would be a very tricky problem to troubleshoot if the module looked like it was re-built for the new kernel yet was not correctly installed. Maybe this is not a consideration if the module is correctly installed in the kernel prior to update ... I don't know much about how the DKMS system is intended to work (is it auto rebuild/install of the modules with the install of both a new kernel and matching headers occurs?) but the most undesirable outcome is when it looks like it did everything automatically but didn't work.

bbaa-bbaa, 2025-01-28:
This is the normal behavior of the dkms package. 

The kernel header dependencies required for building need to be manually satisfied by the user, which is why all the installation guides in the readme.md let users to manually install the kernel headers before installation. 

The dkms package is designed for various kernels, not a specific one, and the package manager cannot resolve the header file dependencies for the specific kernel you are using.

tangent360, 2025-01-28:
ok, understood.   Thanks for the follow up.

PR #249 (2025-01-27): Install the module for all kernels during the first configuration
Fixed #248