strongtz/i915-sriov-dkms #254: Host/LXC graphics tasks using the iGPU with /dev/renderD128 causes VM freese

KylinDemons, 2025-02-05:
I have discovered a strange thing. 
If the host system (including LXC) uses /dev/renderD128 for tasks(like ffmpeg transcoding), the graphical interfaces of other VMs which using the VF will be freeze(seem not entirely freezed, the graphical interface of the VM seems to be very slow to response), during this if start a win guest can cause code 43 randomly(already set the IntelGopDeriver.efi from my BIOS). After stopping host's tasks with the /dev/renderD128, the VM will resume normal.

What happened? During this there is no error information was found in dmesg.

i5-12450H.

bbaa-bbaa, 2025-02-05:
Will the same issue occur when using the renderD129+ node in VF's render?

KylinDemons, 2025-02-05:
> Will the same issue occur when using the renderD129+ node in VF's render?

No, that's exactly the strange thing. Previously, due to VF's inability to use HDR Tonemapping properly, I migrated to an LXC container and used the host's renderD128 for transcoding tasks. I only recently discovered this issue.
It seems that heavily loaded graphics tasks in the host can cause VF problems.
I am currently further confirming the issue.

bbaa-bbaa, 2025-02-05:
All PF/VF share the same compute units. If one of your workloads consumes all the resources, it is normal for other workloads to be affected. 

> No, that's exactly the strange thing. 

Perhaps PF has a higher priority than VF in scheduling.
Can you try limiting the workload on PF? For ffmpeg video transcoding, you can use the [-readrate](https://ffmpeg.org/ffmpeg.html#:~:text=readrate%20speed) option.

```
ffmpeg -readrate 4 -init_hw_device qsv:hw,child_device=/dev/dri/renderD128 -i "[Your Test Video Here]" -c:v h264_qsv -f mp4 /dev/null
```
The maximum decoding speed will be limited to 4x.

LarkinZero, 2025-02-12:
I also encountered this issue, which occurs when streaming with Sunshine and transcoding with Jellyfin simultaneously.

> Virtual Machine 1 is running Windows 11 with Sunshine streaming via a virtual screen.  
> Virtual Machine 2 is running Ubuntu, with Jellyfin installed in Docker, capable of transcoding videos.

Both virtual machines use OVMF (UEFI) and i440fx, and different SR-IOV integrated graphics have been added.

When used individually, there are no issues with these two virtual machines. The problem manifests as follows:
* If transcoding starts before streaming, once Jellyfin begins transcoding, Windows 11 cannot stream.
* If streaming starts before transcoding, the Windows 11 screen freezes and becomes unresponsive while streaming through Moonlight.

I noticed that the integrated graphics load during transcoding is not high. However, streaming on Windows 11 just fails (it keeps spinning, and the screen freezes), and there are no error messages in Sunshine's troubleshooting. I would like to ask if anyone has encountered similar issues and how to resolve them?

bbaa-bbaa, 2025-02-12:
Note that all VFs are allocated with some amount of shared resources (like unlimited GPU execution and preemption times)

If one of your workloads consumes all the resources, it is normal for other workloads to be affected.
-readrate 4

In the i915 driver, we do not have the ability to control the weight of each VF. 
A possible workaround is to write a jellyfin-ffmpeg wrapper script that adds a "-readrate" parameter to limit GPU usage during each transcoding call.

FYI: The Xe driver has added a debugfs option for  tweak SRIOV parameters, but we still need to wait. https://gitlab.freedesktop.org/drm/xe/kernel/-/commit/6bb05b3631bd81227298b291b651dc7ec6ee0145

LarkinZero, 2025-02-12:
> Note that all VFs are allocated with some amount of shared resources (like unlimited GPU execution and preemption times)
> 
> If one of your workloads consumes all the resources, it is normal for other workloads to be affected.
> 
> In the i915 driver, we do not have the ability to control the weight of each VF. A possible workaround is to write a jellyfin-ffmpeg wrapper script that adds a "[-readrate](https://ffmpeg.org/ffmpeg.html#:~:text=readrate%20speed)" parameter to limit GPU usage during each transcoding call.
> 
> FYI: The Xe driver has added a debugfs option for tweak SRIOV parameters, but we still need to wait. https://gitlab.freedesktop.org/drm/xe/kernel/-/commit/6bb05b3631bd81227298b291b651dc7ec6ee0145

My issue is that the transcoding does not fully utilize all the integrated graphics resources; there are still some free resources available. However, the part of the integrated graphics responsible for streaming on Windows 11 appears to be completely unusable.
[image]

bbaa-bbaa, 2025-02-12:
The values from intel_gpu_top are not entirely accurate, 60% usage does not mean you still have 40% of the performance available.

This can be easily verified with a workload that transcoding two streams at the same time, such as watching two different Jellyfin video streams on different devices. You can observe the sum of transcoding frame rate meter is nearly a constant.

KylinDemons, 2025-02-12:
同样的问题，你可以查看下Windows虚拟机的：控制面板——安全与维护——查看存档的信息，看下里面是否有“硬件错误”的报告。我发现在Windows虚拟机出现假死情况时，会有这种错误报告，感觉可能是SRIOV驱动不完善导致的资源调度异常。

LarkinZero, 2025-02-12:
> 同样的问题，你可以查看下Windows虚拟机的：控制面板——安全与维护——查看存档的信息，看下里面是否有“硬件错误”的报告。我发现在Windows虚拟机出现假死情况时，会有这种错误报告，感觉可能是SRIOV驱动不完善导致的资源调度异常。

确实，我也看到这个报错了
Indeed, I have also encountered this error.
```
Source
Windows

Summary
Hardware Error

Date
2025/2/12 15:10

Status
Not Reported

Description
A hardware issue has caused Windows to stop functioning properly.

Problem Signature
Problem Event Name: LiveKernelEvent
Code: 117
Parameter 1: ffffb00601c05460
Parameter 2: fffff806480f2470
Parameter 3: 0
Parameter 4: 0
OS Version: 10_0_26100
Service Pack: 0_0
Product: 256_1
OS Version: 10.0.26100.2.0.0.256.48
Locale ID: 2052

Files to Help Describe the Problem
sysdata.xml
WERInternalMetadata.xml
memory.csv
sysinfo.txt
WERInternalRequest.xml
```

bbaa-bbaa, 2025-02-13:
可以把 jellyfin 的 ffmpeg 换成一个 wrapper 脚本来限制转码速度。限制在 2x 就不会卡 Win 虚拟机了，比 jellyfin 自带的那个暂停好用

```bash
#!/bin/sh
readrate=2
null_input=1
null_output=0
parse_args() {
        while [ $# -gt 0 ]; do
                case "$1" in
                        -f)
                                if [ "$2" = "null" ]; then
                                        null_output=1
                                        shift
                                fi
                                ;;
                        -i)
                                null_input=0 
                                ;;
                        *)
                                ;;
                esac
                shift
        done
}
parse_args "$@"
if [ $null_input -eq 1 ] || [ $null_output -eq 1 ]; then
        /usr/lib/jellyfin-ffmpeg/ffmpeg "$@"
else
        /usr/lib/jellyfin-ffmpeg/ffmpeg -readrate $readrate "$@"
fi
```

## 前
[image]

## 后
[image]

## 空载
[image]

LarkinZero, 2025-02-13:
> 可以把 jellyfin 的 ffmpeg 换成一个 wrapper 脚本来限制转码速度。限制在 2x 就不会卡 Win 虚拟机了，比 jellyfin 自带的那个暂停好用
> 
> #!/bin/sh
> readrate=2
> null_input=1
> null_output=0
> parse_args() {
>         while [ $# -gt 0 ]; do
>                 case "$1" in
>                         -f)
>                                 if [ "$2" = "null" ]; then
>                                         null_output=1
>                                         shift
>                                 fi
>                                 ;;
>                         -i)
>                                 null_input=0 
>                                 ;;
>                         *)
>                                 ;;
>                 esac
>                 shift
>         done
> }
> parse_args "$@"
> if [ $null_input -eq 1 ] || [ $null_output -eq 1 ]; then
>         /usr/lib/jellyfin-ffmpeg/ffmpeg "$@"
> else
>         /usr/lib/jellyfin-ffmpeg/ffmpeg -readrate $readrate "$@"
> fi
> ## 前
> [image]
> 
> ## 后
> [image]
> 
> ## 空载
> [image]

感谢你的方法，我测试后，虽然Jellyfin转码性能下降到50fps，但确实可以操作sunshine串流的虚拟机了。
Thank you for your method. After testing, although Jellyfin's transcoding performance dropped to 50fps, I can indeed operate the Sunshine streaming virtual machine now.