strongtz/i915-sriov-dkms #259: [Late February 2025--Possible Addition to Documentation?] Compatible Remote Desktop Server Solutions for Linux/Windows?

johntdavis84, 2025-02-23:
Hello,

I'm wondering what everyone is using for their remote desktop solution with guest VMs--particularly in Linux. (Microsoft's Remote Desktop works so easily and well that it's my goto for non-gaming VMs).

For my Ubuntu-based VMs, I've been using this fork of `xrdp` with hardware acceleration support for VA-API. https://github.com/neutrinolabs/xrdp/discussions/2383

Unfortunately, that's a PPA, so it's only really readily usable with Debian/Ubuntu guests.

The last time I tried Sunshine/Moonlight, it didn't like Intel GPUs or VA-API. Likewise, NoMachine wasn't compatible with Intel iGPUs or VA-API when I looked at it, either.

What are y'all using for various Arch/Debian/Ubuntu/OpenSUSE-based VMs that need Remote Desktop acceleration?

pasbec, 2025-02-23:
I'm using Ubuntu 24.04 with Wayland and Remote Login enabled. Works out of the box (similar to Windows) with VA-API and RDP. Absolutely smooth and perfect for internet browsing, 4K video and casual games and such.

[image]

johntdavis84, 2025-02-23:
@pasbec Thanks! I'm guessing Ubuntu's "Remote Login" feature is just stock `xrdp` under the hood (since you're using Windows Remote Desktop and it's using the standard RDP port)? (On my Lubuntu distro, `xrdp` is not installed out of the box, so I'd need to install something.)

What version of `xrdp` is it?

When I `apt show xrdp` under Lubuntu 20.04, It tells me 0.9.17 is available to install, which I know is too old to do VA-API hardware acceleration. The PPA I linked above is a forked version of 0.9.22 with VA-API hardware acceleration enabled for Intel iGPUs. I think there's even been a newer version of stock `xrdp` since then, which has the PPA fork's functionality merged in (except for H.264 hardware encoding).

What version do you see when you look at `apt show xrdp` on your working system? If you have a newer one than I do, it sounds like the hardware acceleration just works out of the box on 24.04, which is a spectacular improvement.

I'm excited to hear you've been playing games with it. I've thought it should be possible to do some old DOS/Windows 95-XP-era emulation with a Windows 11 VM, or maybe even a game like Shredder's Revenge, but I wasn't sure. How many VF devices have you defined? Does performance change if you define, say, 4 vs the maximum 7?

pasbec, 2025-02-23:
I' pretty sure, `xrdp` is not involved in this setup. As far as I can see, headless remote sessions are *natively* supported with **GNOME 46** using [`gnome-remote-desktop`](https://gitlab.gnome.org/GNOME/gnome-remote-desktop). It provides limited RDP (via FreeRDP) and VNC (via LibVNCServer) backends which seem to be enough for remoting into some VM with common settings. The related Epic can be found here:
https://gitlab.gnome.org/GNOME/gnome-remote-desktop/-/issues/90

The port can be selected freely and the corresponding username/password is different from the PAM/Linux username. As already mentioned, there is no additional software required at all. Everything works out of the box, similar to Windows. Also `xrdp` does not seem to support Wayland at all:
https://github.com/neutrinolabs/xrdp/issues/2637

Btw, KDE does not have a similar feature, yet:
https://www.reddit.com/r/kde/comments/1f7wyzj/does_kde_wayland_have_any_headless_remote_desktop/?sort=new

The performance with multiple VFs will defintely be worse. I have 7 VFs defined but 'm usually not using multiple simultaneously with high load. If Jellyfin e.g. is transcoding in another VM, there might be some lags but this is just some tiny iGPU and overall performance is very good for headless Ubuntu with SR-IOV.

sshaikh, 2025-02-23:
@pasbec is correct, xrdp doesn't work with wayland.

When I had this working (Proxmox running Arch), I used xrdp along with xorgxrdp-glamor (a variant of the more common xorgxrdp). It worked fine. I disabled it because this repo was moving so fast, and I thought we were close to official support.

FWIW I ran it on two, maybe three, arch VMs and it seemed to work okay. But I only really used for desktop acceleration. I presume the number of VFs doesn't matter, just how many are in use at the same time.

johntdavis84, 2025-02-24:
I'll update my OP tomorrow with these options. Thanks!

I still need to find a more generalized solution that doesn't depend on GNOME. I tend to prefer other desktop environments. And not all the pre-built Ubuntu-variant distros I use ship with GNOME.

I really wish Sunshine and Moonlight or NoMachine would work with hardware acceleration using VA-API. I'd really love to have a distro-agnostic solution.

> FWIW I ran it on two, maybe three, arch VMs and it seemed to work okay. But I only really used for desktop acceleration. I presume the number of VFs doesn't matter, just how many are in use at the same time.

@sshaikh , this is what I was wondering about. I wasn't sure if allocating VFs pre-allocated GPU resources (so four VFs would each have ¼ of the total iGPU graphics resources, fixed, even if only one VM was running using 1 VF); or if a single VF running by itself would use all available resources.

I'm glad to know it works the more useful way and I can just run one VM by itself to get maximum performance. :)

tabletseeker, 2025-02-26:
> I'm using Ubuntu 24.04 with GNOME 46 in Wayland and Remote Login enabled. Works out of the box (similar to Windows) with VA-API and RDP. 

So you're using Spice VNC display on the host?

I thought SRIOV with Linux Guest Video output rendered by the vGPU was impossible, due to Looking Glass still not fully supporting it.

Alternatives like parsec or TigerVNC have not worked for me on Debian.

Could you elaborate further what you did exactly?

johntdavis84, 2025-03-03:
Here's an example of a working configuration with Parsec and VB-Cable for audio. 
https://github.com/strongtz/i915-sriov-dkms/issues/262#issuecomment-2685665604

johntdavis84, 2025-03-03:
I updated the OP with the list of examples of working configs in this thread so far, and added a couple more that I'd still like to see.

tabletseeker, 2025-03-03:
> I updated the OP with the list of examples of working configs in this thread so far, and added a couple more that I'd still like to see.

Hey there, yes I tried your xrdp-egfx solution and it's working but I only get around 12% vGPU Utilization.
I opened an issue here: https://github.com/strongtz/i915-sriov-dkms/issues/264

johntdavis84, 2025-03-04:
> Hey there, yes I tried your xrdp-egfx solution and it's working but I only get around 12% vGPU Utilization. I opened an issue here: [#264](https://github.com/strongtz/i915-sriov-dkms/issues/264)

I saw the issue you opened on this; I'm curious to see what responses you get.

One thing to keep in mind is that the Intel GPU monitor tool isn't as sophisticated as something like `nvidia-smi`. I've seen anecdotal reports that it doesn't report usage accurately. I've seen it report consuming ~50 percent of my iGPU running a software-defined radio app.

tabletseeker, 2025-03-04:
> One thing to keep in mind is that the Intel GPU monitor tool isn't as sophisticated as something like `nvidia-smi`. I've seen anecdotal reports that it doesn't report usage accurately. I've seen it report consuming ~50 percent of my iGPU running a software-defined radio app.

Ah ok, I see. I actually get higher usage when VA-API is not even engaged sometimes. Something definitely seems odd with the display. Another way I've tried to gauge utilization was with temperature readings. Temps with and without VA-API differ around ~ 5 - 8 degrees, so it's probably working I believe.

> Just use Ubuntu 24.04 or a similar GNOME 46+ distribution (not tested) and make sure gnome-remote-desktop is installed. 

Cool, thanks for the tip. I checked it out on [packages.debian.org
](https://packages.debian.org/trixie/gnome-remote-desktop) and hardware acceleration doesn't seem to be mentioned, but I'll definitely try it, regardless.

> Afterwards the virtio display can be removed and the VM stopped/restarted. The VM ist then completely headless but still accessible via RDP.

Yes, I have already tried xrdp-egfx on Ubuntu and it worked only because with Ubuntu I _somehow_ managed to have the VM boot without a Virtio display. I could simply add the SRIOV iGPU and set Video: None. Then once it boots RDP can be connect. That worked fine.

But with a Debian VM for some reason I can't get this to work. The VM simply won't boot without a Display Output for some reason, and thus the rdp service never starts to enable a connection. I'm currently trying to figure out how to run a Debian VM headlessly to fix that. If you're interested too, I'll post my results here.

tabletseeker, 2025-03-11:
OK, all figured out. 
A few adjustments have to be made for Debian users in order to get xrdp-egfx working with SRIOV.

I created this repo that builds it and sets everything up.
https://github.com/tabletseeker/xrdp_vaapi/tree/main

The **most** important thing is this udev rule, which allows access to /dev/dri/renderD128. Drove me insane until I finally found this.
```
cat > /usr/lib/udev/rules.d/90-xorgxrdp-dri.rules
# installed by xorgxrdp. allows every user to use glamor accellerated session
SUBSYSTEM=="drm", KERNEL=="renderD*", GROUP="render", MODE="0666"
SUBSYSTEM=="kfd", GROUP="render", MODE="0666"
```

wouldntyouknow, 2025-05-14:
I can't make the Ubuntu24/25 Gnome work with this and I don't get why.
I have followed the method @pasbec shared above - don't install anything (so no intel va nonfree drivers for example), only xe blacklist and guc=3 are in GRUB.
lspci -nnk output:
```
01:00.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [UHD Graphics] [8086:46d1]
	Kernel driver in use: i915
	Kernel modules: xe, i915
```
System info output:
[image]

Vainfo fails:
```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.22.0
libva error: vaGetDriverNames() failed with unknown libva error
vaInitialize failed with error code -1 (unknown libva error),exit
```
I don't understand what I might be missing. The other solution shared by @johntdavis84 worked great on Ubuntu 24 XFCE but I really want to make this work as well.

pasbec, 2025-07-30:
Sorry about the late response @wouldntyouknow  @tristan-k

With "no additional software", I meant not strictly installing "nothing", rather no special software for remoting, etc. I have indeed installed a few packages manually:
```
>$ sudo apt-mark showmanual
bsdutils
build-essential
cifs-utils
cmake
dash
diffutils
dkms
findutils
git
glmark2-x11
grep
grub-pc
gzip
hostname
ibus-table-cangjie-big
ibus-table-cangjie3
ibus-table-cangjie5
init
language-pack-en
language-pack-en-base
language-pack-gnome-en
language-pack-gnome-en-base
libc6
libc6:i386
libchewing3
libchewing3-data
libegl1
libegl1:i386
libgbm1
libgbm1:i386
libgl1
libgl1-mesa-dri
libgl1-mesa-dri:i386
libgl1:i386
libm17n-0
libmarisa0
libopencc-data
libopencc1.1
libotf1
libpinyin-data
libpinyin15
linux-generic-hwe-24.04
login
m17n-db
mesa-utils
mokutil
ncurses-base
ncurses-bin
qemu-guest-agent
steam-launcher
steam-libs-amd64
steam-libs-i386:i386
ubuntu-desktop-minimal
ubuntu-minimal
ubuntu-restricted-addons
ubuntu-standard
ubuntu-wallpapers
vainfo
vim
wbritish
```

wouldntyouknow, 2025-08-11:
> When I had this working (Proxmox running Arch), I used xrdp along with xorgxrdp-glamor (a variant of the more common xorgxrdp). It worked fine. I disabled it because this repo was moving so fast, [...]

@sshaikh can you recommend a working recent how-to on this subject? I tried to set up xordxrdp-glamor but having issues. For example, I noticed it complaining about missing displays but recent search result on the subject revealed a dummy display was needed earlier, but should not be the case nowadays. Not sure.

wouldntyouknow, 2025-08-11:
> > [@sshaikh](https://github.com/sshaikh) can you recommend a working recent how-to on this subject? I tried to set up xordxrdp-glamor but having issues. For example, I noticed it complaining about missing displays but recent search result on the subject revealed a dummy display was needed earlier, but should not be the case nowadays. Not sure.
> 
> I put together a script based on Nexarian's gist, which fully automates the entire process. https://github.com/tabletseeker/xrdp_vaapi
> 
> It will build all the necessary drivers and packages, while setting up a working environment. SRIOV included.

Hey, yeah it's very cool, used it on an Ubuntu VM, but here the issue is with Arch specifically.

sshaikh, 2025-08-11:
> [@sshaikh](https://github.com/sshaikh) can you recommend a working recent how-to on this subject? I tried to set up xordxrdp-glamor but having issues. For example, I noticed it complaining about [...]

It's been a minute but `xorgxrdp-glamor` should work without setting up sriov as a drop in replacement for vanilla `xorgxrdp` (but without acceleration, obviously), so it may be worth trying to get that up and running first.

tabletseeker, 2025-08-11:
> Hey, yeah it's very cool, used it on an Ubuntu VM, but here the issue is with Arch specifically.

Ah ok, I mean apart from dependencies, you can easily port that script. AUR has the latest versions of most of those packages anyway.

wouldntyouknow, 2025-08-12:
> > Hey, yeah it's very cool, used it on an Ubuntu VM, but here the issue is with Arch specifically.
> 
> Ah ok, I mean apart from dependencies, you can easily port that script. AUR has the latest versions of most of those packages anyway.

Well I am yet a scrub so that's not for me :)
Though I tested the current version of your script on the new Debian Trixie release and the xrdp-sink is not working unfortunately. 
`
ciuser@Deb13-Cloud:/usr/libexec/pulseaudio-module-xrdp$ bash load_pa_modules.sh 
Failure: No such entity
? Can't load pulseaudio xrdp-sink
Failure: No such entity
? Can't load pulseaudio xrdp-source

ciuser@Deb13-Cloud:/usr/libexec/pulseaudio-module-xrdp$ ls -l $(pkg-config --variable=modlibexecdir libpulse) | grep xrdp
-rwxr-xr-x 1 root root    986 Aug 11 21:56 module-xrdp-sink.la
-rwxr-xr-x 1 root root  89288 Aug 11 21:56 module-xrdp-sink.so
-rwxr-xr-x 1 root root    998 Aug 11 21:56 module-xrdp-source.la
-rwxr-xr-x 1 root root  79984 Aug 11 21:56 module-xrdp-source.so
`