strongtz/i915-sriov-dkms #260: Kernel 6.14-rc2 still incompatible?

tabletseeker, 2025-02-23:
- Platform: Debian Trixie
- Compositor: X11
- Kernel: 6.14-rc2
- Hardware: 12th Gen Alderlake
- Install Method: i915-sriov-dkms_2025.02.03_amd64.deb 
- Kernel Parameters: `iommu=pt intel_iommu=on i915.enable_guc=3 module_blacklist=xe i915.max_vfs=7`
- Additional Kernel Modules: `kvmgt vfio_iommu_type1 mdev`
- Issue: Dragging windows causes this error: (Not an actual screenshot - just an illustration of the error)

[image]

dmesg:
```
[    1.714297] i915: loading out-of-tree module taints kernel.
[    1.714662] i915: module verification failed: signature and/or required key missing - tainting kernel
[    1.893341] i915 0000:00:02.0: [drm] Found ALDERLAKE/UY (device ID 9a49) display version 12.00 stepping C0
[    1.894147] i915 0000:00:02.0: Running in SR-IOV PF mode
[    1.894975] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.895402] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.895444] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.897971] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.899213] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[    1.943600] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    1.943604] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    1.946934] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[    1.947784] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    1.947785] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    1.948292] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    1.950230] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.080193] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    3.098060] fbcon: i915drmfb (fb0) is primary device
[    4.257002] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    4.277557] i915 0000:00:02.0: 7 VFs could be associated with this PF
[   13.524912] kvmgt: disagrees about version of symbol i915_gem_object_set_to_cpu_domain
[   13.529514] kvmgt: Unknown symbol i915_gem_object_set_to_cpu_domain (err -22)
[   13.558132] kvmgt: disagrees about version of symbol i915_gem_object_alloc
[   13.561915] kvmgt: Unknown symbol i915_gem_object_alloc (err -22)
[   13.573386] kvmgt: disagrees about version of symbol i915_gem_object_create_shmem
[   13.576808] kvmgt: Unknown symbol i915_gem_object_create_shmem (err -22)
[   13.580812] kvmgt: disagrees about version of symbol i915_gem_object_pin_map
[   13.584769] kvmgt: Unknown symbol i915_gem_object_pin_map (err -22)
[   13.596204] kvmgt: disagrees about version of symbol i915_gem_gtt_insert
[   13.599882] kvmgt: Unknown symbol i915_gem_gtt_insert (err -22)
[   13.603406] kvmgt: disagrees about version of symbol i915_request_add
[   13.606796] kvmgt: Unknown symbol i915_request_add (err -22)
[   13.616861] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_fini
[   13.620676] kvmgt: Unknown symbol i915_gem_ww_ctx_fini (err -22)
[   13.632098] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_init
[   13.632099] kvmgt: Unknown symbol i915_gem_ww_ctx_init (err -22)
[   13.632114] kvmgt: disagrees about version of symbol i915_reserve_fence
[   13.632116] kvmgt: Unknown symbol i915_reserve_fence (err -22)
[   13.632132] kvmgt: disagrees about version of symbol i915_gem_object_ggtt_pin_ww
[   13.632134] kvmgt: Unknown symbol i915_gem_object_ggtt_pin_ww (err -22)
[   13.632238] kvmgt: disagrees about version of symbol i915_gem_object_init
[   13.632239] kvmgt: Unknown symbol i915_gem_object_init (err -22)
[   13.681002] kvmgt: disagrees about version of symbol _i915_vma_move_to_active
[   13.681004] kvmgt: Unknown symbol _i915_vma_move_to_active (err -22)
[   13.681017] kvmgt: disagrees about version of symbol i915_request_create
[   13.681018] kvmgt: Unknown symbol i915_request_create (err -22)
[   13.681051] kvmgt: disagrees about version of symbol i915_ppgtt_create
[   13.681053] kvmgt: Unknown symbol i915_ppgtt_create (err -22)
[   13.681090] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_backoff
[   13.681091] kvmgt: Unknown symbol i915_gem_ww_ctx_backoff (err -22)
[   13.681099] kvmgt: disagrees about version of symbol i915_unreserve_fence
[   13.681101] kvmgt: Unknown symbol i915_unreserve_fence (err -22)
[   13.681116] kvmgt: disagrees about version of symbol __i915_gem_object_set_pages
[   13.681117] kvmgt: Unknown symbol __i915_gem_object_set_pages (err -22)
[   13.681159] kvmgt: disagrees about version of symbol i915_request_wait
[   13.681160] kvmgt: Unknown symbol i915_request_wait (err -22)
[   13.681189] kvmgt: disagrees about version of symbol __i915_gem_object_flush_map
[   13.681190] kvmgt: Unknown symbol __i915_gem_object_flush_map (err -22)
[  459.338037] i915 0000:00:02.1: Running in SR-IOV VF mode
[  459.338448] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[  459.339145] i915 0000:00:02.1: [drm] VT-d active for gfx access
[  459.339194] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[  459.339531] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[  459.340500] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[  459.341408] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[  459.341421] i915 0000:00:02.1: HuC firmware PRELOADED
[  459.346060] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[  459.346070] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[  459.346364] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 1
```
Xorg.conf: (tried all possible options/variations + none)
```
Section "Device"
  Identifier "Intel Graphics"
  Driver "modesetting"
  Option      "AccelMethod"  "uxa"
  Option      "TearFree"        "false"
  Option      "TripleBuffer"    "false"
  Option      "SwapbuffersWait" "false"
EndSection
```

bbaa-bbaa, 2025-02-23:
The kvmgt module seems to be designed for Intel GVT-g, and we have not ported this module in this repo.
The SR-IOV should not use the kvmgt module.

tabletseeker, 2025-02-23:
> The SR-IOV should not use the kvmgt module.

Hi, thanks for the reply.

OK, I unloaded `kvmgvt` with `modprobe -r kvmgt`, but the dragging error persists.

Another thing I'm noticing is that X11 environment variables seem to be empty and XFCE specific stuff like the wallpaper are malfunctioning, as well.

Hmm, in the meantime I'll revert back to 6.12.12 and check if that solves the issue.

PR #273 (2025-03-26): Support v6.14 kernel / Merge upstream patches from linux-stable.git (tag v6.12.20)
Support for 6.14 kernel
Merge upstream patches from linux-stable.git (tag v6.12.20)
Build kvmgt agaginst the i915 module for compatibility (Fixed #260)
Remove unused ccflags (Close #261)
closed 2025-02-24 by tabletseeker