strongtz/i915-sriov-dkms #263: Code 43 after installing WSL in Windows 11 VM

nitinkmr333, 2025-03-01:
Issue:
I have been able to consistently reproduce this problem. Whenever I install WSL in windows 11 VM (host is Proxmox), I start getting code 43 for the intel graphics driver (igpu is Intel xe graphics) in device manager. Uninstalling WSL does not fix it and I have to rollback the VM from a snapshot.

Steps to reproduce:
```
1. Add a VF in win11 guest and install the intel xe graphics drivers. There is no code 43 and graphics driver works fine.
![Image](https://github.com/user-attachments/assets/567dac4d-588b-4445-872f-e87e2de4d3bf)
![Image](https://github.com/user-attachments/assets/dbf9da10-ff31-436b-8772-46487f4a5ca6)
3. Install WSL by running "wsl --install" in elevated powershell (https://learn.microsoft.com/en-us/windows/wsl/install).
![Image](https://github.com/user-attachments/assets/7fd80d83-e805-41dd-97f7-0f65e3d62e47)
4. After reboot, igpu from task manager disappears and we get code 43 in device manager.
![Image](https://github.com/user-attachments/assets/67989583-5bf4-4872-9721-0e37d9a42603)
```

Relevant VFs on Host:
```
❯ lspci
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.4 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.5 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.6 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.7 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
```

VFs are attached by creating "Resource Mapping"-
[image]
[image]
I have not added `intelgopdriver_desktop.bin` here because VM works fine without it, but I did try it. I still get code 43 even after adding the igpu rom and installing wsl.
[image]

Maybe the issue has something to do with nested virtualization?

Host Info:
```
❯ dkms status
i915-sriov-dkms/2025.02.03, 6.8.12-8-pve, x86_64: installed
❯ uname -a
Linux pve 6.8.12-8-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-8 (2025-01-24T12:32Z) x86_64 GNU/Linux
❯ cat /etc/kernel/cmdline
root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
```
I am running ZFS on proxmox boot drive.

Hardware:
```
Minisforum MS-01
CPU- i9 12900H
Graphics- Intel XE graphics (VF shows up as Intel UHD graphics in task manager inside win11 guest)
```

KylinDemons, 2025-03-03:
I have tested that, same issue. It seems that vGPU conflict with the Hyper-V. Turn on Hyper-V makes your VM in a nested virtualization  environment.

demomanca, 2025-03-26:
I narrowed down the component, it's this in Windows Components. If you remove it, driver fires up on reboot.

[image]

nitinkmr333, 2025-03-27:
@demomanca Thank you, your suggestion indeed works. I guess we cannot use WSL until microsoft or intel fixes this.

Peskedor, 2025-05-30:
Same issue here on W10 - is there no fix for this? If i enable hyper-v i cannot get the iGPU to work - with it disabled everything runs fine...