strongtz/i915-sriov-dkms #264: Is this vGPU Utilization normal?

tabletseeker, 2025-03-03:
Host:
- Platform: Debian Trixie
- Compositor: X11
- Kernel: 6.14-rc2
- Hardware: 12th Gen Alderlake
- Install Method: i915-sriov-dkms_2025.02.03_amd64.deb
- Kernel Parameters: `iommu=pt intel_iommu=on i915.enable_guc=3 module_blacklist=xe i915.max_vfs=7`

Guest:
- Platform: Ubuntu 24.2
- Compositor: X11
- Kernel: 6.11
- Install Method: i915-sriov-dkms_2025.02.03_amd64.deb
- Kernel Parameters: `intel_iommu=on i915.enable_guc=3 module_blacklist=xe`
- libvirt Video Device: None
- Using xrdp-egfx with h.264 as suggested by @johntdavis84  https://github.com/neutrinolabs/xrdp/discussions/2383

Issue: 
- Only ~12 - 15% Utilization of SRIOV vGPU (Tested with glxgears, mpv and firefox)

dmesg guest:
```
[    0.027343] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.11.0-17-generic ro intel_iommu=on i915.enable_guc=3 module_blacklist=xe quiet splash vt.handoff=7
[    5.195805] i915: loading out-of-tree module taints kernel.
[    5.195947] i915: module verification failed: signature and/or required key missing - tainting kernel
[    6.082400] i915 0000:09:00.0: [drm] Found ALDERLAKE (device ID 7a50)
[    6.082489] i915 0000:09:00.0: Running in SR-IOV VF mode
[    6.083287] i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    6.083825] i915 0000:09:00.0: [drm] VT-d active for gfx access
[    6.083943] i915 0000:09:00.0: vgaarb: deactivate vga console
[    6.084025] i915 0000:09:00.0: [drm] Using Transparent Hugepages
[    6.086702] i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    6.087744] i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    6.088262] i915 0000:09:00.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    6.088271] i915 0000:09:00.0: HuC firmware PRELOADED
[    6.093341] i915 0000:09:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.093351] i915 0000:09:00.0: [drm] PMU not supported for this GPU.
[    6.093602] [drm] Initialized i915 1.6.0 for 0000:09:00.0 on minor 0

```

xrdp.log (h264 working)
~~~
[2025-03-03T06:40:02.931-0600] [INFO ] Codec search order is H264, RFX
[2025-03-03T06:40:02.933-0600] [INFO ] Matched H264 mode
[2025-03-03T06:40:02.936-0600] [INFO ]   replying version 0x000a0701 flags 0x000000c2
[2025-03-03T06:40:02.943-0600] [INFO ] xrdp_mm_egfx_caps_advertise: xrdp_egfx_send_capsconfirm error 0 best_index 9
[2025-03-03T06:40:02.945-0600] [INFO ] xrdp_egfx_reset_graphics:
[2025-03-03T06:40:02.948-0600] [INFO ] xrdp_egfx_reset_graphics: width 1920 height 1080 monitorcount 1
[2025-03-03T06:40:02.953-0600] [INFO ] xrdp_mm_egfx_create_surfaces: map surface_id 0 left 0 top 0 width 2560 height 1440
[2025-03-03T06:40:02.956-0600] [INFO ] xrdp_encoder_create: starting h264 codec session gfx
[2025-03-03T06:40:02.959-0600] [INFO ] xrdp_encoder_create: using x264 for software encoder
[2025-03-03T06:40:02.962-0600] [INFO ] xrdp_mm_egfx_invalidate_wm_screen:
[2025-03-03T06:40:02.965-0600] [INFO ] xrdp_mm_egfx_caps_advertise: egfx created.
~~~

glxinfo guest:
~~~
glxinfo | grep renderer
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
Extended renderer info (GLX_MESA_query_renderer):
OpenGL renderer string: Mesa Intel(R) Xe Graphics (ALD GT4)
~~~

vainfo guest:

~~~
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
vainfo: Supported profile and entrypoints
(cut off all the profiles and entrypoints )
~~~

glxgears:
~~~
GL_RENDERER   = Mesa Intel(R) Xe Graphics (ALD GT4)
GL_VERSION    = 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.04.1
GL_VENDOR     = Intel
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
332 frames in 5.0 seconds = 66.391 FPS
300 frames in 5.0 seconds = 59.996 FPS
301 frames in 5.0 seconds = 59.936 FPS
301 frames in 5.0 seconds = 60.090 FPS
300 frames in 5.0 seconds = 59.968 FPS
~~~

JobberRT, 2025-03-15:
Hey, I wonder what iGPU you are using? My 12500 works perfectly with PVE8.1 and Windows 11 VM(even no need to use custom EFI rom or vendorID, just install the driver then it works). But the framerate is very low(~10fps). I allocate 4 VF.

I think it's not a VF not actually working problem, because before installing the driver, my cpu usage is 70+ at idle, it's rendering. And after installing the  driver, the cpu usage is -10 at idle.

I'm using RustDesk if that's the case, but PVE Web VNC also low fps

CrAazZyMaN21, 2025-03-15:
Hi, just a wild guess have you tried setting the `i915.max_vfs` to less than `7`. That usage would make sense if the iGPU splits the usage according to the `i915.max_vfs` value. `100% Usage / i915.max_vfs=7 = 14,29% Usage per iGPU Split`

JobberRT, 2025-03-15:
Hi @CrAazZyMaN21 , Yes, I tried `max_vfs` from `7` down to `3`, but the performance is still poor. 

I'm using `VirtIO-GPU` for display since `Standard VGA` has no 3D-acceralate(says chatGPT), and `None`/`SPLICE` will cause the VM no booting(Rustdesk server report no incomming conneciton). So I have to use `VirtIO-GPU` or `VirtGL-GPU`, both has no performance difference. 

I will start a new issue for help, no messing up this issue. Thanks!

tabletseeker, 2025-03-15:
> Hey, I wonder what iGPU you are using? 

Hi there, I have an Alderlake Mobile CPU. 

> I think it's not a VF not actually working problem, because before installing the driver, my cpu usage is 70+ at idle, it's rendering. And after installing the driver, the cpu usage is -10 at idle.

Right, I believe it's actually XRDP no fully utilizing hw acceleration which causes this.

> Hi, just a wild guess have you tried setting the `i915.max_vfs` to less than `7`. That usage would make sense if the iGPU splits the usage according to the `i915.max_vfs` value. `100% Usage / i915.max_vfs=7 = 14,29% Usage per viGPU Split`

Very interesting suggestion! I'm gonna check if that could be the reason. I use ` i915.max_vfs=7` but `sriov_numvfs=1`.
Does the performance split already happen when `max_vfs` is set or only with `num_vfs` ?

bbaa-bbaa, 2025-03-15:
Whether or not max_vfs and sriov_numvfs are set, no performance split will occur, and all VFs will fairly share the resources on your GPU.

tabletseeker, 2025-03-20:
> Whether or not max_vfs and sriov_numvfs are set, no performance split will occur, and all VFs will fairly share the resources on your GPU like your vCPU does.

Looks like it, yeah. I tested `max_vfs=1` and couldn't detect any difference from 7.

Another thing I tested was the IGPU's performance on the host. To my surprise, GPU bare metal utilization is **identical** to that of the virtualized one using SRIOV around ~ 8 - 12%. 

So now I believe the problem must be something else like tlp throttling the iGPU, thermal stuff, maybe the kernel or whatever, because that simply can't be normal.

johntdavis84, 2025-03-20:
Could be a workload issue (some software might not be able to take advantage of the iGPU as well as others).

I'm using Lubuntu 22.04 Jammy with the Microsoft RDP client (the "Windows app") on a Mac. I'm running two virtual desktops, so I can dual screen inside the VM.

At idle with nothing happening in the VM, there's 0 percent utilization. With my SDR receiver open, which is a very graphics intense application, I see the below--about 45 percent used, relatively low clock speed. I don't know enough about hardware accelerated graphics to say, but it looks like the iGPU only commits just enough resources to do what it's supposed to do and no more.

[image]

I'd actually like to see what it looks like doing light gaming, to compare.

Monitoring the host, what's your CPU/GPU temp when you're using your VM?

My mini PC gets scaldingly hot on the bottom when it's doing basically anything, so I set the entire computer on top of a 120mm USB-powered fan, set to high.

Even with the GPU usage above, the CPU is stabilized around 63-65 C.

Via `watch sensors`:

[image]

tabletseeker, 2025-03-20:
@johntdavis84  hey man, thanks for checking in.
It's really weird. I have been playing around with virt-manager a bit, adding/removing hardware and something interesting happened. I **can** achieve 40 - 50% utilization with a virtual GPU added (Virtio) and OpenGL enabled (iGPU) while the SRIOV VF is passed and used as primary renderer in the VM (as indicated by glxinfo).

With this configuration I can see `intel_gpu_top` shooting up to 50% utilization with glxgears, mpv playback using `hwdec=vaapi` etc., **however** the CPU is noticeably under higher load (70%) compared to the regular configuration without Virtio+OpenGL, and graphics are clearly less smooth.

This is extremely confusing, honestly.

When I use the regular SRIOV VF config, I disable the Video Device entirely with `Video=None` and connect via xrdp with hw accel enabled. In this case intel_gpu_top shows 8 - 12% utilization, but CPU load is around 40 - 50% and playback seems quite smooth.

Could it be possible that intel_gpu_top on the host is simply not picking up the load generated by the VM/freerdp2-x11 client ?

johntdavis84, 2025-03-20:
> Could it be possible that intel_gpu_top on the host is simply not picking up the load generated by the VM/freerdp2-x11 client ?

I'm not sure. I think it also might depend on the codec you're using. If it's encoding (to send to your RDP client) in a format that it has a specific hardware decoder for, I don't think that counts as general GPU usage?

**In Proxmox:**
**VirtIO GPU** is just a paravirtualized video driver that should perform better than the default software rendering--but it's still purely software rendering.

**VirtIO GL** if a paravirtualized GPU driver that passes through the actual GPU hardware to the VM via a paravirtualized device, so it _does_ show up in `intel_gpu_top`.

**An actual VF given to the VM via PCIe passthrough**, with the correct drivers, should outperform both of them.

> When I use the regular SRIOV VF config, I disable the Video Device entirely with Video=None and connect via xrdp with hw accel enabled. In this case intel_gpu_top shows 8 - 12% utilization, but CPU load is around 40 - 50% and playback seems quite smooth.

I suspect in this case it's using a hardware encoder in the VM that's built into the iGPU, so work is getting offloaded to that hardware and it's not being counted as general GPU workload, so the utilization of the general GPU is lower.

I am also guessing and someone smarter than me will hopefully chime in.

I think this is also a question worth asking on the GitHub for the `xrdp` fork, but yes, my money's on a specialized hardware decoder in the GPU offloading work from the part of the GPU that `intel_gpu_top` actually monitors.

tabletseeker, 2025-03-20:
@johntdavis84  If you have the time, could you maybe rdp into a Linux VM with a VF passed and check if you're still getting 40% utilization?

Would be really interesting to me if this is platform specific.

johntdavis84, 2025-03-21:
> tabletseeker
>  left a comment
> (strongtz/i915-sriov-dkms#264)
> Thanks for checking!
>
> That would actually explain everything.
>
> The reason why I can see intel_gpu_top hit 40% on the host when using virt-manager with the OpenGL option, is because it's directly engaging the iGPU on the host. Of course the opposite stands true when the VF is used to render, and intel_gpu_top on the host can only read load output of the host iGPU not the SRIOV VF, which is literally a separate entity.
>
> Thus the only true way to test whether there is any load is in the VM itself, and I haven't been able to install intel_gpu_top there yet. It produces an error message about the kernel not being compatible, but I'll hopefully figure that out.

Also (sorry for the double-post), this is, for the moment, a correctable deficiency in Intel’s utility software. 

Compare the way that `nvidia-smi` can easily and seamlessly monitor, from the host, exactly how the GPU is sliced up and exactly how it’s being used in the the guests.

I really hope that Intel gets to something like feature parity with that, at some point.

I’m very curious to see what version 1.30 of the software looks like from the Trixie repo, not to mention the version 2.0 software that just dropped into Bleeding Edge Debian 8 days ago.

closed 2025-03-23 by tabletseeker