strongtz/i915-sriov-dkms #266: only works with Ubuntu  24.4| Tested with Arch, Debian => broken

whymartha, 2025-03-08:
Everything works in Ubuntu but in Arch and Debian llvmpipe (software rendering) is used instead the VF.

For example normal Debian VM with intel-media-va-driver-non-free and firmware-intel-graphics packages gives me this. 

```
$ inxi -GSaz
System:
  Kernel: 6.12.12-amd64 arch: x86_64 bits: 64 compiler: gcc v: 14.2.0
    clocksource: tsc avail: acpi_pm
    parameters: BOOT_IMAGE=/boot/vmlinuz-6.12.12-amd64 ro intel_iommu=on
    i915.enable_guc=3 module_blacklist=xe i915.modeset=1 quiet
  Desktop: Xfce v: 4.20.1 tk: Gtk v: 3.24.48 wm: xfwm4 v: 4.20.0
    with: xfce4-panel tools: avail: light-locker dm: LightDM v: 1.32.0
    Distro: Debian GNU/Linux trixie/sid
Graphics:
  Device-1: Intel RocketLake-LF GS2 [Iris Xe Graphics] vendor: Microsoft
    driver: i915 v: 2025.02.03-sriov.6.12.11 alternate: xe arch: Xe
    process: Intel 14nm built: 2021-21 bus-ID: 09:00.0 chip-ID: 8086:9a49
    class-ID: 0300
  Display: x11 server: X.Org v: 21.1.16 compositor: xfwm4 v: 4.20.0 driver:
    X: loaded: fbdev,modesetting,vesa dri: swrast gpu: i915 display-ID: :10.0
    screens: 1
  API: EGL v: 1.5 hw: drv: intel iris platforms: gbm: drv: iris:
    drv: swrast x11: drv: swrast inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: mesa v: 25.0.1-2 glx-v: 1.4
    direct-render: yes renderer: **llvmpipe (LLVM 19.1.7 256 bits)**
  Info: Tools: api: clinfo, eglinfo, glxinfo de: xfce4-display-settings
    gpu: gputop, intel_gpu_top, lsgpu x11: xdriinfo, xdpyinfo, xprop, xrandr
```

```
$ glxinfo | grep renderer
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_gl_interop, GLX_MESA_query_renderer, GLX_SGIS_multisample, 
Extended renderer info (GLX_MESA_query_renderer):
OpenGL renderer string: **llvmpipe (LLVM 19.1.7, 256 bits)**
```

```
$ lspci | grep VGA
02:00.0 VGA compatible controller: Intel Corporation RocketLake-LF GS2 [Iris Xe Graphics]
```
```
$ vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.22.0
libva error: vaGetDriverNames() failed with unknown libva error
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 2
vaInitialize failed with error code 2 (resource allocation failed),exit
```
```
$ dmesg | grep i915
i915 0000:09:00.0: [drm] Found ROCKETLAKE/LF
i915 0000:09:00.0: Running in SR-IOV VF mode
i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.2.1
i915 0000:09:00.0: [drm] VT-d active for gfx access
i915 0000:09:00.0: vgaarb: deactivate vga console
i915 0000:09:00.0: [drm] Using Transparent Hugepages
i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.2.1
i915 0000:09:00.0: [drm] GT0: GUC: interface version 0.1.2.1
i915 0000:09:00.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
i915 0000:09:00.0: HuC firmware PRELOADED
i915 0000:09:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
i915 0000:09:00.0: [drm] PMU not supported for this GPU.
drm Initialized i915 1.6.0 for 0000:09:00.0 on minor 0

```

johntdavis84, 2025-03-08:
> whymartha
>  created an issue

What does your kernel boot command say?

whymartha, 2025-03-09:
> What does your kernel boot command say?
```
parameters: BOOT_IMAGE=/boot/vmlinuz-6.12.12-amd64 ro intel_iommu=on 
i915.enable_guc=3 module_blacklist=xe i915.modeset=1 quiet
```
I use Video=none (-vga none) option in libvirt to make sure the VM only has access to the VF and no virtual GPU.

Interesting side note. When I use a virtual GPU like Video=QXL or Video=VGA in libvirt then vainfo works if LIBVA_DRIVER_NAME=iHD is set.

tabletseeker, 2025-03-11:
Try this. Debian does not allow access by default, thus everything device related fails out of the box.
```
sudo /bin/bash -c 'cat > /usr/lib/udev/rules.d/render_access.rules << EOF
SUBSYSTEM=="drm", KERNEL=="renderD*", GROUP="render", MODE="0666"
SUBSYSTEM=="kfd", GROUP="render", MODE="0666"
EOF
'
```

whymartha, 2025-03-22:
> Try this. Debian does not allow access by default, thus everything device related fails out of the box.
> 
> ```
> sudo /bin/bash -c 'cat > /usr/lib/udev/rules.d/render_access.rules << EOF
> SUBSYSTEM=="drm", KERNEL=="renderD*", GROUP="render", MODE="0666"
> SUBSYSTEM=="kfd", GROUP="render", MODE="0666"
> EOF
> '
> ```
> 
> You can check out xrdp with hw accel here: https://github.com/tabletseeker/xrdp_vaapi

Why I can't find this anywhere ??

All commands intel related work now.

yardern, 2025-06-11:
It works under Arch Linux with xorgxrdp-glamor.
It is mentioned here: [https://wiki.archlinux.org/title/Xrdp](url)

Somehow, I couldn't get the audio work with pipewire-module-xrdp.