strongtz/i915-sriov-dkms #267: PVE 8.1 12500 SR-IOV with Windows 11 LTSC VM comes with poor performance

JobberRT, 2025-03-15:
# Brief Problem
12500 CPU with Windows 11 VM has super poor performance, in both PVE WebVNC or Rustdesk(a remote desktop software) it can only have around 3~15fps.

# Detailed System Status
**Strange Installation Behavior(Maybe a good thing)**
My installation of Windows 11 VM has no encounted Code43 error, just use host cpu and UEFI BIOS with main-gpu checked. 

**Screenshots, click to view**

<details>
### Driver Information

[image]

### PVE Web Configuration

[image]

### Remote Desktop Metrics

[image]

### Resource Manager GPU Usage
**Desktop Window Manager has high GPU usage, but I don't know why the total usage is only 1%, and the Desktop Winow Manager usage is around 50%**
[image]

</details>

# Things I've tried
- Scale down the number of `max_vfs`
- Change different display mode. `None` and `SPLICE` will cause the VM no booting. `Standard VGA` has no 3D-Acceralate. So I have to use `VirtGL-GPU` or `VIrtIO-GPU`, both of them has no perforance difference.
- Recreate a fresh new VM
- Trying to run Furkmark but it says I only got OpenGL 1.1

### I don't know if 12500 iGPU can handle graphics rendering inside a VM through SR-IOV(3VFs), but to my acknowledge, I think UDH770 should handle this eaisly like breathing..

closed 2025-03-15 by JobberRT

CrAazZyMaN21, 2025-03-15:
Hi, can you please post your steps, so others who have the same problem can solve that too. Thx

JobberRT, 2025-03-15:
@CrAazZyMaN21 Sure! Let me do a full step explaination(They are all from the README). I will edit this comment after I finish writing