strongtz/i915-sriov-dkms #271: intel_gpu_top does not support VFs

tabletseeker, 2025-03-23:
`intel_gpu_top` can only be launched successfully with the main card (PF) on the host, via the following commands:

```
intel_gpu_top -d drm:/dev/dri/card0
intel_gpu_top -d drm:/dev/dri/renderD128
intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0
```

This works, but does not monitor the load of a VF that is passed to a VM. It can only display the utilization of card0 on the host.

Using these commands with a VF on the host fails:

```
intel_gpu_top -d drm:/dev/dri/card1
intel_gpu_top -d drm:/dev/dri/renderD129
intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.1
```
```
Requested device drm:/dev/dri/card1 not found!
Requested device drm:/dev/dri/renderD129 not found!
Requested device sys:/sys/devices/pci0000:00/0000:00:02.1 not found!
```

Executing these commands against the passed VF inside the VM fails, as well.
```
intel_gpu_top -d drm:/dev/dri/card0
intel_gpu_top -d drm:/dev/dri/renderD128
intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0
```
```
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
```

`dmesg | grep i915` shows that apart from card0, any SRIOV VF that is initialized displays the following kernel message:
```
i915 0000:09:00.0: [drm] PMU not supported for this GPU.
```

Is there any way to enable PMU support for the VF?
Is VF PMU support only offered on specific Gen12+ architectures?

This Proxmox guide seems to indicate that monitoring the VF with `intel_gpu_top` works:
>Check you can monitor the VFs - if not you have issues
monitor any VF renderer in real time with intel_gpu_top -d drm:/dev/dri/renderD128 there is one per VF - to see them all use ls -l /dev/dri

https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3?permalink_comment_id=4774387

johntdavis84, 2025-03-25:
> Is there any way to enable PMU support for the VF?

This is a good question, and I'm not sure. That's a not-minor-at-all feature. One of the project maintainers and contributors would have confirm, but I have a sneaking suspicion that we'll have to wait for something to happen upstream … or at least for Proxmox to move to a version of Debian with a version of `intel_gpu_top` that isn't over two years old. There's been a number of updates to the tool since Debian 12 was released, and the Debian 12 repo isn't keeping up with those.

That said, a Windows VM has no problems at all measuring GPU usage inside the VM (take a look at Task Manager's tab where it shows individual CPU cores and the GPU). It's definitely possible. We just can't do it with the Linux toolchain right now. The Windows Xe driver package is really rock solid for VF-enabled guests, which makes the lack of official support from Intel for Linux that much more frustrating.

Thanks for testing all of these commands in one place.
I've always used `intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0`; I wasn't even sure how to use some of the other options.
Apparently, `intel_gpu_top -d sriov` is also supposed to work.

> This Proxmox guide seems to indicate that monitoring the VF with `intel_gpu_top` works

This is really frustrating, because for the 1.5 years I've been using this DKMS driver, for all that it's not a perfect solution, `intel_gpu_top` _has_ worked just as advertised, and even given useful data when a VM is using a VF. Not as much as I'd like--it still can't see exactly what the VM is doing and has always been kind of glitchy to the point that I don't completely trust the actual values of the numbers it gives me, but enough to verify that the GPU is actually doing something.

But sometime in the last couple of months, it just stopped working and doesn't show utilization bars at all. It's still showing wattage and some other live data at the top, so it's partially working, but something has gone wrong.

I don't really get it, either. Updating the kernel without recompiling the DKMS driver would break the driver, but I haven't done that. I need to try fully uninstalling the driver on the Proxmox host and reinstalling it, but for the moment it's been more of an annoyance than anything else and I've had too many other things to troubleshoot. It keeps getting pushed down the list.

tomalta, 2025-03-28:
> This is really frustrating, because for the 1.5 years I've been using this DKMS driver, for all that it's not a perfect solution, `intel_gpu_top` _has_ worked just as advertised, and even given [...]

Unfortunately this has never been my experience. As tabletseeker reported I haven't found a way either to monitor VFs for any Linux guest installations I've ever tried. Inside the guest OS I always get the "failed to detect engines" error, no matter how I try to address the device, through names or PCIe addresses.

Despite this, when it's not throwing a code 43 error, I can measure the VF usage under Windows without issues. It shows in Task Manager, GPU-Z and Furmark.

johntdavis84, 2025-03-28:
> tomalta

> Despite this, when it's not throwing a code 43 error, I can measure the VF usage under Windows without issues. It shows in Task Manager, GPU-Z and Furmark.

This has been my experience as well. At least we know it’s working. :P

johntdavis84, 2025-04-11:
Following up. I've poked at this a bit more and at least on my system, not all of the commands in OP work.

On the Proxmox host: `intel_gpu_top -d drm:/dev/dri/card0` throws an error.
```
root@andromeda2:~# intel_gpu_top -d drm:/dev/dri/card0
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
Segmentation fault
```

These commands both seem to load `/dev/dri/card1`.
```
intel_gpu_top -d drm:/dev/dri/renderD128
intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0
```

And today, over on the Intel LTS Mainline GitHub thread for SR-IOV it was recommended that I use this command, which, again, loads `/dev/dri/card1`.
`intel_gpu_top -d pci`                # I assume this one just grabs whichever Intel card it finds on the PCIe bus?
Source: https://github.com/intel/linux-intel-lts/issues/33#issuecomment-2795771887

tabletseeker, 2025-04-12:
Is that inside a Linux VM?

johntdavis84, 2025-04-12:
> Is that inside a Linux VM?

My last post?

Proxmox host.

I've given up on monitoring inside a Linux VM.

johntdavis84, 2025-09-05:
Update:

I've been using `-d sriov` without issues for a few months now.

`nvtop` also works (available from `apt`), though it's not fully compatible with i915 driver-based cards, so a lot of information it could show isn't there. It works much better with Xe driver-based setups.