strongtz/i915-sriov-dkms #272: RAM Ballooning not working?

isc30, 2025-03-23:
Hi, after installing this dkms module, it seems like RAM ballooning stopped working for that VM.
Is it a known issue?

isc30, 2025-03-23:
Ignore it, it seems like passing a device in proxmox disables balooning

closed 2025-03-23 by isc30

johntdavis84, 2025-03-23:
PCIe device passthrough is generally not compatible with memory ballooning, for GPUs or otherwise. PCIe passthrough works in part by mapping the I/O hooks (I’m paraphrasing, not sure what these are actually called, but they’re the pathway for the VM to talk to the real physical hardware) for a PCIe device into a memory location in the VM. 

If ballooning is on, the memory map inside the VM can change enough at runtime that the PCIe device is no longer in the place in RAM where the guest OS thinks it is, which will lead to instability/crashes/worse.

Disable memory ballooning for that VM.