strongtz/i915-sriov-dkms #274: dkms status gives an error: `WARNING! Diff between built and installed module!`

luckman212, 2025-03-26:
Hi I had been running the previous 2025.02.03 release on Proxmox 8.3 with the optional 6.11 kernel successfully for some time. Then it stopped working (I think due to a minor kernel update) so I decided to try today's new release.

I again followed the steps from https://github.com/strongtz/i915-sriov-dkms/releases/download/2025.03.26/i915-sriov-dkms_2025.03.26_amd64.deb

And, it was successful 👍 

But, after rebooting and checking the status, I see this warning:

```
root@pve01:~# dkms status -m i915-sriov-dkms
i915-sriov-dkms/2025.03.26, 6.11.11-2-pve, x86_64: installed (WARNING! Diff between built and installed module!)
```

```
root@pve01:~# uname -r
6.11.11-2-pve
```

I googled around a bit but can't figure out how to solve this error? Or if it even needs to be "solved"?

bbaa-bbaa, 2025-03-27:
Can you try reinstalling it?
(use dpkg -P i915-sriov-dkms to purge it first)

luckman212, 2025-03-27:
Sure, I just did that. Purged and reinstalled, and rebooted.

Now I have this:

```
pve01:~# dkms status
i915-sriov-dkms/2025.03.26, 6.11.11-1-pve, x86_64: installed
i915-sriov-dkms/2025.03.26, 6.11.11-2-pve, x86_64: installed (WARNING! Diff between built and installed module!)
i915-sriov-dkms/2025.03.26, 6.8.12-9-pve, x86_64: installed (WARNING! Diff between built and installed module!)
root@pve01:~# uname -r
6.11.11-2-pve
```

bbaa-bbaa, 2025-03-27:
Try #275
Github Actions artifacts are available here: https://github.com/bbaa-bbaa/i915-sriov-dkms/actions/runs/14098864373

PR #275 (2025-03-27): Fix the module cannot be installed via deb package
Fix #274
closed 2025-03-27 by strongtz