strongtz/i915-sriov-dkms #276: Intel-GPU-Tools -- Setting up a PPA for Ubuntu/Debian for the Latest Version?

johntdavis84, 2025-03-27:
Hello,

This is a bit far afield of the actual driver dev, but I wanted to bring it up here in case someone with more experience in Debian/Ubuntu PPA maintenance might have some input on whether it's worth pursing ancillary to this project.

`intel-gpu-tools` is an essential toolset for this project--in particular, `intel_gpu_top`. It's the bet way to monitor actual VF activity from the host to see if the DKMS driver is actually working.

Right now, This entire package is tied to the repo of whatever version of Debian/Ubuntu you're using. Proxmox 8 is Debian 12, so it's using the Debian 12 user land repos--which have a 2 year old version of `intel_gpu_tools`. The current version is only a month or two old, and is currently in the Debian 13/Trixie repo, which won't be a part of Proxmox until Proxmox 9.

It would be really great if we could add a PPA repo to our host systems that contains the latest version of the `intel-gpu-tools` package, for use both in the host and in VMs (maybe the new version enables looking at VF status inside the VM?).

Like I said, I know that's out of scope for this project, but since I lack the technical skill to contribute here, trying to set up a PPA for the Intel tools is something I'm really interested in.

Does anyone have any advice on how we might get started with something like this?

lichao0223, 2025-03-28:
I manually compiled the Intel GPU Tools version 2.0 and ran it in a virtual machine, but the result remained the same - only PF (Physical Function) is supported while VF (Virtual Function) support is absent. After reviewing the changelogs for version 2.0 and prior releases, most updates appear to focus on supporting Xe-series graphics cards.

johntdavis84, 2025-03-28:
How did you install and compile the version 2.0 tools? I'd like to try this, but I've never done it before and have no idea how to get started.

Also, did you experience this issue with the 2.0 package?
https://github.com/strongtz/i915-sriov-dkms/issues/271

Just like the OP over there, aside from showing power draw and RC6 (whatever that is), I've completely lost the usage percentages from `intel_gpu_top`.

lichao0223, 2025-03-28:
You can obtain the latest source code for intel-gpu-tools from [](https://gitlab.freedesktop.org/drm/igt-gpu-tools). I compiled it under Debian 12, and the source code includes build instructions for compilation. If you need the binary files I compiled, I can upload them for you. I suspect the VF (Virtual Function) monitoring capability hasn't been implemented in the Linux i915 driver, whereas the Windows driver implementation supports this functionality.