strongtz/i915-sriov-dkms #277: Correct GRUB flags for PVE Install and code 43 related?

tomalta, 2025-03-28:
My system is based on an Intel 12900HK and an Intel Iris Xe 96EU.
I managed to install the DKMS module somewhat successfully on the system but I'm getting this message:

`[0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-7-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7`
`[0.083357] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-7-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 use xe.force_probe='46a6' and i915.force_probe='!46a6'`

I've also noticed in the guide that it says to blacklist the xe module.
Now I'm confused on what I should do: should I blacklist the xe module or, as far as I understood, use `i915.force_probe='!46a6'` to force the use of the i915 module?

Could it be that me not using the correct GRUB flags is giving me issues in a Windows VM? I have a display dummy in the HDMI port and this way I can RDP in the system without a display configured. But trying to use Sunshine gets me a code 43 error and trying to connect to the machine through Parsec before using RDP gives me the same code 43 error.
Having no dummy or display connected to the HDMI doesn't allow me to RDP to the machine without a virtual display set up in Proxmox.

LukeMondy, 2025-04-20:
This line: `use xe.force_probe='46a6' and i915.force_probe='!46a6'`

is actually backwards - it forces using `xe` and stops `i915` from being loaded.

I was having issues getting i915 to load, even when explicitly told with `i915.force_probe='!46a6'`, and the issue ended up being because of secure boot. I had to follow the instructions related to `MOK` here: https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

tomalta, 2025-04-20:
Thanks for the clarification. My setup works and I got the MOK key enrolled for the module. So that message means nothing for me, right?
Let me know if you need more details about my setup.