strongtz/i915-sriov-dkms #278: Not working with kernel 6.12.20-amd64?

derzwen, 2025-03-29:
Hi, did a fresh debian testing installation today in a Proxmox VM. In the past I had no problems getting SR-IOV working on other debian stable VMs. But with this kernel in testing it does work. DKMS module is installed but somehow not loaded.

```shell
cat /proc/cmdline 

BOOT_IMAGE=/vmlinuz-6.12.20-amd64 root=/dev/mapper/testing--vg-root ro quiet i915.enable_guc=3 module_blacklist=xe
``` 

```shell
dkms status

i915-sriov-dkms/2025.03.27, 6.12.20-amd64, x86_64: installed (Original modules exist)
``` 

```shell
dmesg | grep -E "i915|xe|VGA|Display"

[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.12.20-amd64 root=/dev/mapper/testing--vg-root ro quiet i915.enable_guc=3 module_blacklist=xe console=tty0 console=ttyS0,115200
[    0.000000] NX (Execute Disable) protection: active
[    0.000097] MTRR map: 4 entries (2 fixed + 2 variable; max 18), built from 8 variable MTRRs
[    0.010004] PM: hibernation: Registered nosave memory: [mem 0xe0000000-0xefffffff]
[    0.014025] Kernel command line: BOOT_IMAGE=/vmlinuz-6.12.20-amd64 root=/dev/mapper/testing--vg-root ro quiet i915.enable_guc=3 module_blacklist=xe console=tty0 console=ttyS0,115200
[    0.111451] PCI: ECAM [mem 0xe0000000-0xefffffff] (base 0xe0000000) for domain 0000 [bus 00-ff]
[    0.416756] pci 0000:00:01.0: vgaarb: setting as boot VGA device
[    0.416756] pci 0000:00:01.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.416756] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    0.419733] system 00:04: [mem 0xe0000000-0xefffffff window] has been reserved
[    0.716793] fb0: EFI VGA frame buffer device
[   10.590381] systemd[1]: Set up automount proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point.
[   10.896354] RAPL PMU: API unit is 2^-32 Joules, 0 fixed counters, 10737418240 ms ovfl timer
[   10.925372] Module xe is blacklisted
``` 

```shell
lspci | grep UHD

01:00.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
``` 

What am I missing?

bbaa-bbaa, 2025-03-29:
Try `modprobe i915` and upload latest `dmesg` here.
The result of `modinfo i915`  is also useful.

derzwen, 2025-03-29:
Figured it out 🤦‍♂️ `modinfo i915` gave the hint. Quick googling revealed the underlaying problem. Secure boot was enabled in that VM. 

```shell
modinfo i915

modprobe: ERROR: could not insert 'i915': Key was rejected by service
``` 

Is this a know issue that the module does not work with secure boot enabled? If yes, I didn't find anything in the docs. Maybe add a hint there?

Thanks for the quick help anyway 💪👌 BTW: I'm very glad that this module exists and very well works!

bbaa-bbaa, 2025-03-29:
The simplest way is to disable Secure Boot.
If you need Secure Boot, you will need to maintain your own Secure Boot keys and use the correct private key to sign the module.
More information can be found here: https://github.com/dell/dkms?tab=readme-ov-file#module-signing

derzwen, 2025-03-29:
Yeah, I don't need it so I just disabled it. Issue can be closed 👍