strongtz/i915-sriov-dkms #281: Promxox Kernel 6.8.12-9 Dropped Today; Issues with i915 DKMS Driver?

johntdavis84, 2025-04-03:
Can anyone confirm that the DKMS driver works with the 6.8.12-9 kernel that Proxmox just pushed to the Enterprise repo today?

This kernel contains a fix for a potentially serious ZFS bug, so I'd like to upgrade a soon as possible.

nitinkmr333, 2025-04-07:
I am using 6.8.12-9. Haven't faced any issues yet.

w00tlarr, 2025-04-10:
Works but this, which maybe not because of Proxmox 8.4?

[   26.173139] i915 0000:00:02.0: Enabled 7 VFs
[  171.646951] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.646975] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF1 state (-EPROTO)
[  171.647320] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.647328] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF2 state (-EPROTO)
[  171.647705] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.647716] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF3 state (-EPROTO)
[  171.648402] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.648414] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF4 state (-EPROTO)
[  171.649025] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.649035] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF5 state (-EPROTO)
[  171.649525] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.649534] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF6 state (-EPROTO)
[  171.649962] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[  171.649968] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF7 state (-EPROTO)

johntdavis84, 2025-04-10:
> w00tlarr

I haven’t ever seen that one before. But, I’ve seen other errors/warnings that don’t seem to actually stop it from working.

But that looks like something fundamental isn’t working. Would you mind opening a separate issue on it? It might get buried and ignored if it’s only here in my “hey, y’all, did your Proxmox explode?” thread. ;)

johntdavis84, 2025-04-12:
> Will open a new thread.

I've updated Proxmox and my VMs. I'm seeing the same error, though I have a different `status` code. Though, I suspect that might be a memory address.

Interestingly, mine also references VF7.

I only have 4 VFs enabled.