strongtz/i915-sriov-dkms #284: Proxmox 8.4 Released Today. Compatibility Check?

johntdavis84, 2025-04-09:
Hello!

Proxmox VE 8.4 was released today, with 6.8.12-9 as the default Linux kernel, and 6.14 available as an option.

For the default config, I think we're okay, since 6.8.12-9 is already known to work?

I think I saw an issue comment earlier about 6.14 still having issues.

ratbich, 2025-04-12:
Has anyone done the 8.4 update yet and have any success to share?

johntdavis84, 2025-04-12:
I did it.

Everything seems to be working fine, though I'm seeing the error reported here: https://github.com/strongtz/i915-sriov-dkms/issues/281#issuecomment-2794915464

```
[ 26.173139] i915 0000:00:02.0: Enabled 7 VFs
[ 171.646951] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.646975] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF1 state (-EPROTO)
[ 171.647320] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.647328] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF2 state (-EPROTO)
[ 171.647705] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.647716] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF3 state (-EPROTO)
[ 171.648402] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.648414] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF4 state (-EPROTO)
[ 171.649025] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.649035] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF5 state (-EPROTO)
[ 171.649525] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.649534] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF6 state (-EPROTO)
[ 171.649962] i915 0000:00:02.0: [drm] ERROR GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[ 171.649968] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF7 state (-EPROTO)
```

Everything seems to be working, but there's definitely some odd stuff in `dmesg` I don't remember seeing before.

KylinDemons, 2025-04-12:
I have updated to the ProxmoxVE 8.4.1 with kernel Linux 6.14.0-2-pve, all things seem worked fine for me with no errors

```bash
# root @ pve in ~ [18:59:57] 
$ dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.14.0-2-pve root=/dev/mapper/pve-root ro quiet splash intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.052676] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.14.0-2-pve root=/dev/mapper/pve-root ro quiet splash intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[    1.423687] i915: loading out-of-tree module taints kernel.
[    1.688601] i915 0000:00:02.0: [drm] Found ALDERLAKE_P (device ID 46a3) display version 13.00 stepping D0
[    1.688613] i915 0000:00:02.0: Running in SR-IOV PF mode
[    1.689246] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.689248] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.689281] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.689720] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.692904] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    1.699253] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.36.0
[    1.699261] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    1.716277] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[    1.716745] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    1.716750] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    1.717216] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    1.719488] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.720270] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    1.733916] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    1.744032] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    1.753876] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    1.763883] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.041057] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   11.195169] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[   11.195278] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   11.195309] i915 0000:00:02.1: [drm] Found ALDERLAKE_P (device ID 46a3) display version 13.00 stepping D0
[   11.195333] i915 0000:00:02.1: Running in SR-IOV VF mode
[   11.195926] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.17.0
[   11.197207] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   11.197246] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   11.197977] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.17.0
[   11.198564] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.17.0
[   11.199251] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.199254] i915 0000:00:02.1: HuC firmware PRELOADED
[   11.202473] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.202478] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   11.202652] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 1
[   11.203020] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   11.203024] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[   11.203113] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   11.203132] i915 0000:00:02.2: [drm] Found ALDERLAKE_P (device ID 46a3) display version 13.00 stepping D0
[   11.203149] i915 0000:00:02.2: Running in SR-IOV VF mode
[   11.203708] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.17.0
[   11.204194] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   11.204215] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   11.204836] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.17.0
[   11.205328] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.17.0
[   11.206005] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.206008] i915 0000:00:02.2: HuC firmware PRELOADED
[   11.208576] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.208581] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[   11.208670] [drm] Initialized i915 1.6.0 for 0000:00:02.2 on minor 2
[   11.209035] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   11.209038] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   11.209041] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[   11.209098] i915 0000:00:02.3: enabling device (0000 -> 0002)
[   11.209115] i915 0000:00:02.3: [drm] Found ALDERLAKE_P (device ID 46a3) display version 13.00 stepping D0
[   11.209133] i915 0000:00:02.3: Running in SR-IOV VF mode
[   11.209708] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.17.0
[   11.210212] i915 0000:00:02.3: [drm] VT-d active for gfx access
[... cut ...]
[   11.230188] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.17.0
[   11.230505] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.230508] i915 0000:00:02.7: HuC firmware PRELOADED
[   11.232194] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.232198] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[   11.232310] [drm] Initialized i915 1.6.0 for 0000:00:02.7 on minor 7
[   11.232453] i915 0000:00:02.0: Enabled 7 VFs
```

[image]