strongtz/i915-sriov-dkms #285: not working for debian 12 with debian 6.12.12+bpo-amd64 kernel.

whjiang, 2025-04-19:
I have i915 installed. 

After successfully executed `modprobe i915`,  `dmesg |grep i915` returns following message:
```
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.12.12+bpo-amd64 root=UUID=04c80ebd-b372-4ad8-8250-5454c3e57456 ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.030571] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.12.12+bpo-amd64 root=UUID=04c80ebd-b372-4ad8-8250-5454c3e57456 ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[   12.234068] i915 0000:00:02.0: [drm] Found ALDERLAKE_P/RPL-P (device ID a7a0) display version 13.00 stepping E0
[   12.236904] i915 0000:00:02.0: Running in SR-IOV PF mode
[   12.240380] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   12.243359] i915 0000:00:02.0: vgaarb: deactivate vga console
[   12.243401] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   12.243790] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   12.243823] i915 0000:00:02.0: firmware: failed to load i915/adlp_dmc.bin (-2)
[   12.243827] i915 0000:00:02.0: firmware: failed to load i915/adlp_dmc.bin (-2)
[   12.243829] i915 0000:00:02.0: Direct firmware load for i915/adlp_dmc.bin failed with error -2
[   12.243846] i915 0000:00:02.0: firmware: failed to load i915/adlp_dmc_ver2_16.bin (-2)
[   12.243850] i915 0000:00:02.0: firmware: failed to load i915/adlp_dmc_ver2_16.bin (-2)
[   12.243851] i915 0000:00:02.0: Direct firmware load for i915/adlp_dmc_ver2_16.bin failed with error -2
[   12.243853] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/adlp_dmc.bin (-ENOENT). Disabling runtime power management.
[   12.243859] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
[   12.252215] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_70.bin (-2)
[   12.252229] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_70.bin (-2)
[   12.252242] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_70.1.1.bin (-2)
[   12.252252] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_70.1.1.bin (-2)
[   12.252264] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_69.0.3.bin (-2)
[   12.252274] i915 0000:00:02.0: firmware: failed to load i915/adlp_guc_69.0.3.bin (-2)
[   12.252278] i915 0000:00:02.0: [drm] *ERROR* GT0: GuC firmware i915/adlp_guc_70.bin: fetch failed -ENOENT
[   12.252283] i915 0000:00:02.0: [drm] GT0: GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   12.253147] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 0.0.0
[   12.253221] i915 0000:00:02.0: [drm] *ERROR* GT0: GuC initialization failed -ENOENT
[   12.253222] i915 0000:00:02.0: [drm] *ERROR* GT0: Enabling uc failed (-5)
[   12.253224] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to initialize GPU, declaring it wedged!
[   12.253629] i915 0000:00:02.0: [drm] CI tainted: 0x9 by intel_gt_init+0xd8/0x380 [i915]
[   12.323252] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[   12.488516] fbcon: i915drmfb (fb0) is primary device
[   12.513238] i915 0000:00:02.0: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[   12.579821] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   12.592865] i915 0000:00:02.0: [drm] *ERROR* GT0: IOV: Initialization failed (-EIO) GT wedged
[   12.593020] i915 0000:00:02.0: [drm] PF aborted (-EIO) i915_driver_probe+0xb73/0xdc0 [i915]
[   12.593569] i915 0000:00:02.0: No VFs could be associated with this PF!
[   13.937570] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   13.988054] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
ro
```

bbaa-bbaa, 2025-04-20:
I think you may need to install [firmware-intel-graphics](https://packages.debian.org/bookworm-backports/amd64/firmware-intel-graphics)

whjiang, 2025-04-20:
> I think you may need to install [firmware-intel-graphics](https://packages.debian.org/bookworm-backports/amd64/firmware-intel-graphics)

I installed this package. But, the same error still exists.