strongtz/i915-sriov-dkms #286: failed to enable on pve 8.4.1 (kernel 6.8.12-10-pve)

whjiang, 2025-04-26:
I followed the steps listed in README and I still can't get the 7 VFs. the `dmesg` output is below:
```
root@pve:~# dmesg |grep "00:02"
[    0.792640] pci 0000:00:02.0: [8086:a7a0] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[    0.792692] pci 0000:00:02.0: BAR 0 [mem 0x60e7000000-0x60e7ffffff 64bit]
[    0.792704] pci 0000:00:02.0: BAR 2 [mem 0x4000000000-0x400fffffff 64bit pref]
[    0.792714] pci 0000:00:02.0: BAR 4 [io  0x4000-0x403f]
[    0.792752] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.792765] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.792845] pci 0000:00:02.0: VF BAR 0 [mem 0x60e0000000-0x60e0ffffff 64bit]
[    0.792853] pci 0000:00:02.0: VF BAR 0 [mem 0x60e0000000-0x60e6ffffff 64bit]: contains BAR 0 for 7 VFs
[    0.792865] pci 0000:00:02.0: VF BAR 2 [mem 0x6000000000-0x601fffffff 64bit pref]
[    0.792873] pci 0000:00:02.0: VF BAR 2 [mem 0x6000000000-0x60dfffffff 64bit pref]: contains BAR 2 for 7 VFs
[    0.815054] pci 0000:02:00.0: [1b21:1166] type 00 class 0x010601 PCIe Endpoint
[    0.815147] pci 0000:02:00.0: BAR 0 [mem 0x4f982000-0x4f983fff]
[    0.815166] pci 0000:02:00.0: BAR 5 [mem 0x4f980000-0x4f981fff]
[    0.815175] pci 0000:02:00.0: ROM [mem 0x4f900000-0x4f97ffff pref]
[    0.815198] pci 0000:02:00.0: enabling Extended Tags
[    0.815309] pci 0000:02:00.0: PME# supported from D0 D3hot D3cold
[    0.871956] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    0.871956] pci 0000:00:02.0: vgaarb: bridge control possible
[    0.871956] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.946233] pci_bus 0000:02: resource 1 [mem 0x4f900000-0x4f9fffff]
[    0.948085] pci 0000:00:02.0: Adding to iommu group 0
[    0.949655] pci 0000:02:00.0: Adding to iommu group 11
[    0.955318] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[    0.955814] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[    0.956342] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f6e7000 [fault reason 0x06] PTE Read access is not set
[    1.437434] ahci 0000:02:00.0: SSS flag set, parallel bus scan disabled
[    1.448809] ahci 0000:02:00.0: AHCI 0001.0301 32 slots 32 ports 6 Gbps 0xffffff3f impl SATA mode
[    1.449880] ahci 0000:02:00.0: flags: 64bit ncq sntf stag pm led only pio sxs deso sadm sds apst 
[    5.957944] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[    5.973274] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[    5.989875] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   10.958941] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f6ac000 [fault reason 0x06] PTE Read access is not set
[   10.973493] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   10.990192] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   15.975108] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   15.990554] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   16.007167] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   20.976480] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f6e9000 [fault reason 0x06] PTE Read access is not set
[   20.990784] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   21.007543] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   25.977265] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f794000 [fault reason 0x06] PTE Read access is not set
[   25.990999] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   26.007868] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   30.992975] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   31.008284] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   31.024912] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   35.993793] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   36.008532] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   36.025145] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   40.994522] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f69e000 [fault reason 0x06] PTE Read access is not set
[   41.008744] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   41.025475] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   46.010697] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   46.025797] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   46.042530] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   51.011815] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f6aa000 [fault reason 0x06] PTE Read access is not set
[   51.025942] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   51.042715] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   56.012650] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f77d000 [fault reason 0x06] PTE Read access is not set
[   56.026280] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   56.043002] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   61.028223] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   61.043375] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   61.059967] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   66.028533] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   66.043635] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   66.060269] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[... 5 lines cut ...]
[   76.077409] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   81.045966] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   81.060709] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   81.077606] DMAR: [DMA Read NO_PASID] Request device [00:02.0] fault addr 0x4f600000 [fault reason 0x06] PTE Read access is not set
[   84.411061] xe 0000:00:02.0: Your graphics device a7a0 is not officially supported
[... cut ...]
[   84.733568] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   84.745875] [drm] Initialized i915 1.6.0 20230929 for 0000:00:02.0 on minor 1
[   84.888677] i915 0000:00:02.0: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[   84.946567] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   84.955562] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```

whjiang, 2025-04-29:
I found the problem. I have zfs as my root filesystem. Thus, PVE use systemd-boot instead of grub for bootting. So, I need to modify `/etc/kernel/cmdline` instead of `/etc/default/grub` for boot command line.

After fixing it, I have 00:02.0 to 00:02.7 devices available.

FabioEight, 2025-05-16:
> pve 8.4.1 (kernel 6.8.12-10-pve) grub boot
> 
> root@pve:~# dmesg |grep "00:02" [ 0.605424] pci 0000:00:02.0: [8086:4680] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint [ 0.605438] pci 0000:00:02.0: BAR 0 [mem 0x60e9000000-0x60e9ffffff 64bit] [ 0.605441] pci 0000:00:02.0: BAR 2 [mem 0x4000000000-0x403fffffff 64bit pref] [ 0.605443] pci 0000:00:02.0: BAR 4 [io 0x4000-0x403f] [ 0.605453] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics [ 0.605456] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff] [ 0.605480] pci 0000:00:02.0: VF BAR 0 [mem 0x60e2000000-0x60e2ffffff 64bit] [ 0.605481] pci 0000:00:02.0: VF BAR 0 [mem 0x60e2000000-0x60e8ffffff 64bit]: contains BAR 0 for 7 VFs [ 0.605483] pci 0000:00:02.0: VF BAR 2 [mem 0x6000000000-0x601fffffff 64bit pref] [ 0.605485] pci 0000:00:02.0: VF BAR 2 [mem 0x6000000000-0x60dfffffff 64bit pref]: contains BAR 2 for 7 VFs [ 0.618661] pci 0000:02:00.0: [197b:0585] type 00 class 0x010601 PCIe Legacy Endpoint [ 0.618694] pci 0000:02:00.0: BAR 0 [io 0x3200-0x327f] [ 0.618696] pci 0000:02:00.0: BAR 1 [io 0x3180-0x31ff] [ 0.618698] pci 0000:02:00.0: BAR 2 [io 0x3100-0x317f] [ 0.618700] pci 0000:02:00.0: BAR 3 [io 0x3080-0x30ff] [ 0.618702] pci 0000:02:00.0: BAR 4 [io 0x3000-0x307f] [ 0.618704] pci 0000:02:00.0: BAR 5 [mem 0x85110000-0x85111fff] [ 0.618706] pci 0000:02:00.0: ROM [mem 0x85100000-0x8510ffff pref] [ 0.618767] pci 0000:02:00.0: PME# supported from D3hot [ 0.658320] pci 0000:00:02.0: vgaarb: setting as boot VGA device [ 0.658320] pci 0000:00:02.0: vgaarb: bridge control possible [ 0.658320] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none [ 0.661645] system 00:02: [io 0x1854-0x1857] has been reserved [ 0.689763] pci_bus 0000:02: resource 0 [io 0x3000-0x3fff] [ 0.689764] pci_bus 0000:02: resource 1 [mem 0x85100000-0x851fffff] [ 0.690821] pci 0000:00:02.0: Adding to iommu group 0 [ 0.691143] pci 0000:02:00.0: Adding to iommu group 20 [ 1.074569] ahci 0000:02:00.0: SSS flag set, parallel bus scan disabled [ 1.074606] ahci 0000:02:00.0: AHCI 0001.0301 32 slots 5 ports 6 Gbps 0x1f impl SATA mode [ 1.074610] ahci 0000:02:00.0: flags: 64bit ncq sntf stag pm led clo pmp fbs pio slum part ccc apst boh [ 4.914348] i915 0000:00:02.0: [drm] VT-d active for gfx access [ 4.914351] i915 0000:00:02.0: vgaarb: deactivate vga console [ 4.914383] i915 0000:00:02.0: [drm] Using Transparent Hugepages [ 4.915021] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem [ 4.915792] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915]) [ 4.923201] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1) [ 4.929142] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.36.0 [ 4.929145] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3 [ 4.931788] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads [ 4.932131] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled [ 4.932132] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled [ 4.932439] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled [ 4.932909] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915]) [ 4.933007] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized [ 4.962752] [drm] Initialized i915 1.6.0 20230929 for 0000:00:02.0 on minor 0 [ 4.964057] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes [ 4.987698] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes [ 4.987760] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes [ 5.370905] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs [ 5.370923] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs

I have the same issue, SR-IOV no longer present after rebooting recently (likely kernel upgrade to 6.8.12-10-pve). Re-checked all the configuration and upgraded to the last package here in repo with no luck.

xtsui, 2025-06-22:
> /etc/kernel/cmdline

Hi, could you please tell me how to modify this file configuration? What content do I need to modify and add? Because my open is a line starting with "root=.....". Thanks a lot.