strongtz/i915-sriov-dkms #287: PVE Host (6.8) and Talos VM v1.9.5 (6.12)

mglee72, 2025-04-26:
It's a long shot, but I'm trying to see if I can make Talos VM running on PVE Host (minisforum ms-01) to pick up on a subset of VFs passed through.

Following the guidance here and https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html, PVE host can now see the VFs in `lspci`.

What does not work is when I pass through VFs (4~7) to the talos VM running v1.9.5 with i915 extension included (kernel seems to be 6.12.18)

```
Linux version 6.12.18-talos (@buildkitsandbox) (gcc (GCC) 14.2.0, GNU ld (GNU Binutils) 2.43.1) #1 SMP Tue Mar 11 15:15:12 UTC 2025
```

I also added extra kernel args (`intel_iommu=on i915.enable_guc=3 module_blacklist_xe`).

The dmesg I get via `talosctl -n $NODE_IP 20 dmesg | grep i915`

```
192.168.201.20: kern:    info: [2025-04-26T22:24:08.316076214Z]: Command line: talos.platform=metal talos.config=none console=tty0 init_on_alloc=1 slab_nomerge pti=on consoleblank=0 nvme_core.io_timeout=4294967295 printk.devkmsg=on ima_template=ima-ng ima_appraise=fix ima_hash=sha512 intel_iommu=on i915.enable_guc=3 module_blacklist=xe
192.168.201.20: kern:  notice: [2025-04-26T22:24:08.346593214Z]: Kernel command line: talos.platform=metal talos.config=none console=tty0 init_on_alloc=1 slab_nomerge pti=on consoleblank=0 nvme_core.io_timeout=4294967295 printk.devkmsg=on ima_template=ima-ng ima_appraise=fix ima_hash=sha512 intel_iommu=on i915.enable_guc=3 module_blacklist=xe
192.168.201.20: user: warning: [2025-04-26T22:24:09.918363214Z]: [talos] [initramfs] enabling system extension i915 20250211-v1.9.5
192.168.201.20: kern:    info: [2025-04-26T22:24:10.813884214Z]: i915 0000:06:10.0: [drm] Found ALDERLAKE_P/RPL-P (device ID a7a0) display version 13.00 stepping E0
192.168.201.20: kern:     err: [2025-04-26T22:24:12.814833214Z]: i915 0000:06:10.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
192.168.201.20: kern:     err: [2025-04-26T22:24:12.837735214Z]: i915 0000:06:10.0: [drm] *ERROR* Device initialization failed (-5)
192.168.201.20: kern:     err: [2025-04-26T22:24:12.837893214Z]: i915 0000:06:10.0: probe with driver i915 failed with error -5
192.168.201.20: kern:    info: [2025-04-26T22:24:12.838127214Z]: i915 0000:06:11.0: [drm] Found ALDERLAKE_P/RPL-P (device ID a7a0) display version 13.00 stepping E0
192.168.201.20: kern:     err: [2025-04-26T22:24:14.840891214Z]: i915 0000:06:11.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
192.168.201.20: kern:     err: [2025-04-26T22:24:14.849860214Z]: i915 0000:06:11.0: [drm] *ERROR* Device initialization failed (-5)
192.168.201.20: kern:     err: [2025-04-26T22:24:14.850009214Z]: i915 0000:06:11.0: probe with driver i915 failed with error -5
192.168.201.20: kern:    info: [2025-04-26T22:24:14.850222214Z]: i915 0000:06:1b.0: [drm] Found ALDERLAKE_P/RPL-P (device ID a7a0) display version 13.00 stepping E0
192.168.201.20: kern:     err: [2025-04-26T22:24:16.851108214Z]: i915 0000:06:1b.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
192.168.201.20: kern:     err: [2025-04-26T22:24:16.897656214Z]: i915 0000:06:1b.0: [drm] *ERROR* Device initialization failed (-5)
192.168.201.20: kern:     err: [2025-04-26T22:24:16.897812214Z]: i915 0000:06:1b.0: probe with driver i915 failed with error -5
192.168.201.20: kern:    info: [2025-04-26T22:24:16.898025214Z]: i915 0000:06:1c.0: [drm] Found ALDERLAKE_P/RPL-P (device ID a7a0) display version 13.00 stepping E0
192.168.201.20: kern:     err: [2025-04-26T22:24:18.900224214Z]: i915 0000:06:1c.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
192.168.201.20: kern:     err: [2025-04-26T22:24:18.937675214Z]: i915 0000:06:1c.0: [drm] *ERROR* Device initialization failed (-5)
192.168.201.20: kern:     err: [2025-04-26T22:24:18.938511214Z]: i915 0000:06:1c.0: probe with driver i915 failed with error -5
```

My intuition is that I would have to bake in your kernel module into a new bespoke talos vm image, given the "immutable" nature of Talos (probably https://www.talos.dev/v1.9/advanced/proprietary-kernel-modules/).

I wanted to check here to see if I'm missing anything obvious before getting my hands dirty :)

One quick question I had was - given that this repo claims to pull the code from intel's 6.12 mainline - maybe I should fully expect the talos (which runs 6.12.18) kernel should already include the module.

Any pointer would be appreciated, thank you!

mglee72, 2025-04-26:
I guess I totally confused myself - it's not in linux mainline, it's just in intel's repo with questionable timeline https://github.com/intel/linux-intel-lts/issues/33

Eyeless77, 2025-05-05:
@mglee72 did you manage to get it working in a VM?

mglee72, 2025-05-05:
Hey, not really :( 

Some additional signal suggests that my Minisforum MS-01 box is the culprit. 

Agter removing the dkms, reverting grub confs on my pve host and passing through the whole device, I could get the talos vm working. However after pve hard-backup of the vm where it stopped and rebooted, the vm pesistently was stuck in a state showing the same errors.

What didn't help was re-attaching the pci passthrough, and what worked was host reboot (started to work again)

Some more tests I could do:
1. I could retry with cpu=host to see if it matters. 
2. I could try to replicate on a different intel box with igpu.

isc30, 2025-07-05:
Hi, did anyone get to install it?

Luvieify1, 2025-09-06:
I managed to get this to work, had the same error with Debian 13 generic cloudimage, had to disable secureboot (by using SeaBIOS) and add "iommu=pt" to grub next to the lines in the documentation.

johntdavis84, 2025-09-06:
You can disable secure boot in an OMVF (UEFI)-based VM. Access the VM's UEFI BIOS by hitting escape when the Proxmox splash screen is showing inside the VNC viewer while the VM is booting up (while the white status bar at the bottom is filling). From there, it's under one of the settings options (I think device settings?). Poke around until you find it, disable it, save changes, and then reboot.

In my experience so far, you should always disable Secure Boot inside the VM when you're working with PCIe passthrough. It just makes everything easier.