strongtz/i915-sriov-dkms #291: What is the best way to get audio out of an SRIOV enabled VM under Proxmox?

wouldntyouknow, 2025-05-08:
As per title. A part of the process is to set "Display" to "none" on the VM properties, so audio via spice will stop working, and all 3 audio devices Proxmox lets me add to the VM seem to be spice based. I have to remove the audio device to be able to boot.
What's the best solution to get audio to work via RDP?
Thanks!

johntdavis84, 2025-05-08:
> wouldntyouknow
>  created an issue

What RDP solution are you using now, on what kind of guest OS? You’ll need to use a solution that includes audio in the RDP data stream itself.

wouldntyouknow, 2025-05-08:
> What RDP solution are you using now, on what kind of guest OS? You’ll need to use a solution that includes audio in the RDP data stream itself.
> […](#)
> 
> 
> 
>  wouldntyouknow
>   created an issue
>  ([strongtz/i915-sriov-dkms#291](https://github.com/strongtz/i915-sriov-dkms/issues/291))
>   <[#291](https://github.com/strongtz/i915-sriov-dkms/issues/291)>
>  As per title. A part of the process is to set "Display" to "none" on the VM properties, so audio via spice will stop working, and all 3 audio devices Proxmox lets me add to the VM seem to be spice based. I have to remove the audio device to be able to boot.
>  What's the best solution to get audio to work via RDP?
>  Thanks!

Hi there!
RDP all the way, both host and guest, both Linux (Ubuntu with xrdp) and native Windows. RDP audio works with VMs with Spice - but here display is disabled so no spice, wondering how to do this and what to look into.
IDK, for example, should I add a secondary SPICE display and add audio, would this be fine? Or is there a better way where Display can remain "none".

johntdavis84, 2025-05-08:
This is actually easier to solve if you’re wanting RDP with Ubuntu than it might be otherwise. :)

Completely uninstall the xrdp from the repo. Rip it out like it never happened.

Install this development fork from here: https://github.com/neutrinolabs/xrdp/discussions/2383
To get audio:

> Even though the pulseaudio module is still provided, `pipewire-module-xrdp` should be used instead

So, only use the Pulse-Audio add-on if you don’t have Pipewire.

This is from the official dev team, but it’s a bit more advanced than what’s in the official Ubuntu repos.

I’ve used this with an Ubuntu VM and the iGPU on a 12700T for years with no issues. It’s great.

Make sure you have the non-free Intel video acceleration drivers installed. You need them for encoding video on the server to send down the pipe to your RDP client.

https://launchpad.net/ubuntu/+source/intel-media-driver-non-free

johntdavis84, 2025-05-08:
The above just covers `xrdp`.

Here's my entire process for this with a new Ubuntu VM.
Put the default display device back on for a bit. You're going to need to intact with the Ubuntu VM via NoVNC to do the setup.

1.     Pre-Install/First Boot: Add entropy device and **_disable memory balloon_**;
2.     First system update;
3.     **Disable Xe driver inside the guest** using kernel boot command: `modprobe.blacklist=xe` in `grub` config and updating `grub`.
4.     qemu-agent install;
6.     Set resolution using `xrandr` to whatever you want;
7.     Install SSH server;
8.     install `dkms` and `build-essential` packages inside guest;
10.     Install Ubuntu Extras and Intel iGPU Non-Free Driver (apt install lubuntu-restricted-extras intel-gpu-tools intel-media-va-driver-non-free), install vainfo, install mesa-utils (for glxinfo);
11.     Install special xRDP from https://github.com/neutrinolabs/xrdp/discussions/2383 .
13.     Remove default display device (host).
15.     **VM Hardware Settings:** Add iGPU slice: PCIe, ROM bar enabled, Primary GPU enabled, DO NOT USE ALL FUNCTIONS (uncheck)
16.     Update guest grub with SR-IOV iGPU and update grub/initramfs (see instructions in README)
17.     Install DKMS Driver from this repo using the deb package (download and then `dpkg -i $PACKAGE_NAME`.

wouldntyouknow, 2025-05-08:
Hi!
Thanks for taking the time for the above. I'm still a bit of a noob so maybe it's rather a bit much :)
What I tried so far is to install a minimal xubuntu 25.04, add sriov and install xrdp_vaapi. Render works fine in RDP, but no audio. Is it possible to just add audio to whatever xrdp you are already using?
On a non-gpu accelerated Ubuntu with SPICE I had audio, but I used a different generic installer for that.

johntdavis84, 2025-05-09:
> wouldntyouknow
>  left a comment
> (strongtz/i915-sriov-dkms#291)
> For audio, try installing the Pipewire module I mentioned in the earlier post, if you’re using Pipewire.
> … <x-msg://19/#>
> Thank you. I will try with a fresh install.
> Just one further question if it's ok: what do you mean with this: "Add entropy device"

Sorry for the confusion on that one. It’s not directly relevant to what you’re doing.

When you’re setting up your Virtual Machine, in the hardware section you can add an entropy device, which can avoid slowdowns under certain circumstances (e.g., during boot) for some VMs: https://pve.proxmox.com/wiki/Qemu/KVM_Virtual_Machines

> A RNG (Random Number Generator) is a device providing entropy (randomness) to a system. A virtual hardware-RNG can be used to provide such entropy from the host system to a guest VM. This helps to avoid entropy starvation problems in the guest (a situation where not enough entropy is available and the system may slow down or run into problems), especially during the guests boot process.

You don’t need it, but adding /dev/urandom might, sometimes, theoretically be helpful. :)

wouldntyouknow, 2025-05-11:
> 8\. vainfo, install mesa-utils

Hey again, sorry for bothering. I followed your instructions to the letter and it _seems_ to work. I can RDP in using mstsc from Windows, but it doesn't seem snappy.
The github xrdp build page you referenced mentioned this:

> what codec is used you will see in /var/log/xrdp.log. Look for a line that says
> rdp_encoder_create: starting gfx rfx pro codec session
> or
> xrdp_encoder_create: starting h264 codec session gf

When I cat the logfile there is no such entry (any entry with "encoder" in it), so I assume it's not working. Certainly feels like it. I have checked everything on the "Experience" tab, still the same. Any ideas?

I also checked Guacamole 1.6.0 - I found docker builds that support new gfx options. Here it seems to work!
```
[2025-05-11T14:38:55.167+0200] [INFO ] xrdp_encoder_create: starting gfx rfx pro codec session
[2025-05-11T14:38:55.169+0200] [INFO ] xrdp_encoder_create: using x264 for software encoder
```
Is software encoder OK, is this what is needed or do I need further tweaking for HW encoding or anything?

Vainfo seems to be fine:
```
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
```
vkcube spinney cube seems to work with the below output:
`Selected GPU 0: Intel(R) Graphics (ADL-N), type: IntegratedGpu`

glxinfo:
```
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
Extended renderer info (GLX_MESA_query_renderer):
OpenGL renderer string: Mesa Intel(R) Graphics (ADL-N)
```

Are there any further things I can check that the solution works as optimally (relying on the gpu to the max) as possible?
Thank you!

johntdavis84, 2025-05-11:
> Hey again, sorry for bothering. I followed your instructions to the letter and it _seems_ to work. I can RDP in using mstsc from Windows, but it doesn't seem snappy. The github xrdp build page you referenced mentioned this:

No need to apologize; I'm glad things are moving in the right direction! Getting this going the first time is always a challenge. Once you get it working once the way you want, the next time is much easier.

Is audio working now? That's what you were trying to solve when I pushed you down this rabbit-hole, I think. :)

Have you installed the 'intel-gpu-tools` package in Proxmox?
If not, grab it from the `apt` repo and run `intel_gpu_top -d pci` on the Proxmox host when your VM is active. You should be able to see when the card is actually working there.

As to choppiness ... what version of Ubuntu is your VM running? Older versions of Ubuntu are limited to 25 fps over RDP.

You can see what the RDP server on the Ubuntu thinks its FPS is by running this inside the GUI in the VM:
1. open a terminal window;
2. Type `xrandr`. You'll get a list of all the resolutions the RDP virtual display supports, and it will show you what resolution and frame rate it's actually sending to the RDP client. What do you see? (If you're not sure, just paste the whole output and I'll help.)

The `xrandr` resolution and framerate is what is being sent to your Windows RDP client. Sometimes, it's less than what the guest OS is actually configured to output. Downsampled, if that makes sense. To see what your _guest_ thinks the FPS is, go to the UFO test in a web browser. What's the highest FPS it shows?
 
> When I cat the logfile there is no such entry (any entry with "encoder" in it), so I assume it's not working. Certainly feels like it. I have checked everything on the "Experience" tab, still the [...]

This part always confuses me quite a bit. It's possible to use hardware encoding without using the H264 codec--your Windows RDP client has to ask for the codec it wants, and I think the Windows client can even specify if the server should use hardware encoding or not.

Based on what you said below, I suspect the Windows RDP client isn't requesting a hardware accelerated session.

_**It might be worth it to make a separate issue to ask about the optimal settings for Windows' built-in RDP client.** I don't use it, and don't have any experience in tuning it, unfortunately._

> I also checked Guacamole 1.6.0 - I found docker builds that support new gfx options. Here it seems to work!
>
> ```
> [2025-05-11T14:38:55.167+0200] [INFO ] xrdp_encoder_create: starting gfx rfx pro [...]

That's software decoding or H.264 video streams specifically, think. Ideally, we'd see a notice that it's using a hardware decoder, there.

It's possible you're connected and getting a video stream via a codec other than H264, and the codec you're using is hardware accelerated.

> Vainfo seems to be fine:
>
> ```
> libva info: VA-API version 1.20.0
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
> libva info: Found init function __vaDriverInit_1_20
> libva [...]
The bolded lines above are the correct result, showing that hardware rendering is enabled and working. But just because it's enabled doesn't mean it's always being _used_. If hardware acceleration from the GPU wasn't being used at all, you'd get a different `renderer string` for the software renderer (`LLVM Pipe`)

**I'm not sure what to make of the mentions of software rendering throughout, but again, I think that's due to the Windows RDP client needing to be configured correctly.** And the documentation for doing that from Microsoft is ... terrible. I'm not quite sure how to go about it right this second.

You've got lots of great diagnostic info already.
I'd grab the results of the `xrandr` output to see what the RDP server is giving you, and the results of the UFO test to see what the guest itself thinks its outputing, and take all that diagnostic output and ask over here:
https://github.com/neutrinolabs/xrdp/discussions/2383

Post a message indicating you're trying to make sure you have hardware acceleration with the Windows RDP client configured correctly, but aren't sure because of some contradictory output (mentions of software rendering combined with indicators that the hardware is being used).

I can't boot my Linux VM right now, but when I can, I'll check the log files and see if they match yours and report back.

wouldntyouknow, 2025-05-11:
Hey, thanks for the huge write-up, I really appreciate it.
> Is audio working now? That's what you were trying to solve when I pushed you down this rabbit-hole, I think. :)
Oh yes. All it took was the pipewire-module. I realized I came with the wrong assumption originally - that any sound requires a sound device attached to the VM. Well, now I know it doesn't.

> Have you installed the `intel-gpu-tools` package in Proxmox? If not, grab it from the `apt` repo and run `intel_gpu_top -d pci` on the Proxmox host when your VM is active. You should be able to see when the card is actually working there. I find it easiest to run this command in an SSH session instead of the web console in Proxmox, so I can click around the Proxmox UI and do other things and still have the GPU monitor running.
No I haven't, thanks, I'll check it out. My Proxmox can't locate that package, but I'll look into it.

> As to choppiness ... what version of Ubuntu is your VM running? Older versions of Ubuntu are limited to 25 fps over RDP. Newer versions should be able to hit 60 FPS.
You might be right, I think I saw 25 fps when checking the Display options. It is Ubuntu 24.04.

> You've got lots of great diagnostic info already. I'd grab the results of the `xrandr` output to see what the RDP server is giving you, and the results of the UFO test to see what the guest itself thinks its outputing, and take all that diagnostic output and ask over here: [neutrinolabs/xrdp#2383](https://github.com/neutrinolabs/xrdp/discussions/2383)
> 
> Post a message indicating you're trying to make sure you have hardware acceleration with the Windows RDP client configured correctly, but aren't sure because of some contradictory output (mentions of software rendering combined with indicators that the hardware is being used).
Great, I will do some further checking and experimenting and try to ask over there as well.
Again, thanks for all the info, you have helped me a lot!