strongtz/i915-sriov-dkms #292: No driver bound to device; cannot configure SR-IOV

miRemid, 2025-05-13:
i try to enable sriov for my fresh-new installed pve system (kernel 6.8.12-10-pve) with a 12700H cpu

i follow the steps in README.md, but after rebooting, the sriov seems not working
```shell
root@pve:~# dmesg -T | grep "00:02"
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: [8086:46a6] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: BAR 0 [mem 0x603c000000-0x603cffffff 64bit]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: BAR 2 [mem 0x4000000000-0x400fffffff 64bit pref]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: BAR 4 [io  0x3000-0x303f]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x00ffffff 64bit]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x06ffffff 64bit]: contains BAR 0 for 7 VFs
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0x1fffffff 64bit pref]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0xdfffffff 64bit pref]: contains BAR 2 for 7 VFs
[Tue May 13 19:51:27 2025] pci 0000:02:00.0: [1e0f:0009] type 00 class 0x010802 PCIe Endpoint
[Tue May 13 19:51:27 2025] pci 0000:02:00.0: BAR 0 [mem 0x6c500000-0x6c503fff 64bit]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: vgaarb: bridge control possible
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[Tue May 13 19:51:27 2025] system 00:02: [io  0x1854-0x1857] has been reserved
[Tue May 13 19:51:27 2025] pnp 00:03: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x4020000000-0x40ffffffff 64bit pref]: assigned
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x4010000000-0x4016ffffff 64bit]: assigned
[Tue May 13 19:51:27 2025] pci_bus 0000:02: resource 1 [mem 0x6c500000-0x6c5fffff]
[Tue May 13 19:51:27 2025] pci 0000:00:02.0: Adding to iommu group 0
[Tue May 13 19:51:27 2025] pci 0000:02:00.0: Adding to iommu group 16
[Tue May 13 19:51:28 2025] nvme nvme1: pci function 0000:02:00.0
[Tue May 13 19:51:31 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Tue May 13 19:51:31 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
root@pve:~# dkms status
i915-sriov-dkms/2025.03.27, 6.8.12-10-pve, x86_64: installed
i915-sriov-dkms/2025.03.27, 6.8.12-4-pve, x86_64: installed
```

igem0n, 2025-05-13:
I've got exactly the same issue on i9-12900K with 6.8.12-10-pve

bbaa-bbaa, 2025-05-14:
Check the output of lspci -vv -nn -s 00:02.

If the "Kernel driver in use" is missing, try running `modprobe i915` and check the latest message of `dmesg`.

wouldntyouknow, 2025-05-14:
What's the proper "Kernel driver in use" value here? My output is this:
```
        Kernel driver in use: vfio-pci
        Kernel modules: xe, i915
```
Should it print i915 as driver in use?

igem0n, 2025-05-14:
I've already sorted it out, it was beacause secure boot pevented the driver from loading

miRemid, 2025-05-17:
> Check the output of `lspci -vv -nn -s 00:02`.
> 
> If the "Kernel driver in use" is missing, try running `modprobe i915` and check the latest message of `dmesg`.

after running `modprobe i915`, here is the dmesg
```shell
root@pve:~# modprobe i915
root@pve:~# dmesg -T | grep "00:02"
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: [8086:46a6] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: BAR 0 [mem 0x603c000000-0x603cffffff 64bit]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: BAR 2 [mem 0x4000000000-0x400fffffff 64bit pref]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: BAR 4 [io  0x3000-0x303f]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x00ffffff 64bit]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x06ffffff 64bit]: contains BAR 0 for 7 VFs
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0x1fffffff 64bit pref]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0xdfffffff 64bit pref]: contains BAR 2 for 7 VFs
[Sat May 17 09:55:28 2025] pci 0000:02:00.0: [1e0f:0009] type 00 class 0x010802 PCIe Endpoint
[Sat May 17 09:55:28 2025] pci 0000:02:00.0: BAR 0 [mem 0x6c500000-0x6c503fff 64bit]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: vgaarb: bridge control possible
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[Sat May 17 09:55:28 2025] system 00:02: [io  0x1854-0x1857] has been reserved
[Sat May 17 09:55:28 2025] pnp 00:03: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 2 [mem 0x4020000000-0x40ffffffff 64bit pref]: assigned
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x4010000000-0x4016ffffff 64bit]: assigned
[Sat May 17 09:55:28 2025] pci_bus 0000:02: resource 1 [mem 0x6c500000-0x6c5fffff]
[Sat May 17 09:55:28 2025] pci 0000:00:02.0: Adding to iommu group 0
[Sat May 17 09:55:28 2025] pci 0000:02:00.0: Adding to iommu group 16
[Sat May 17 09:55:29 2025] nvme nvme0: pci function 0000:02:00.0
[Sat May 17 09:55:32 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Sat May 17 09:55:32 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] Found ALDERLAKE_P (device ID 46a6) display version 13.00 stepping D0
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: Running in SR-IOV PF mode
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] VT-d active for gfx access
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: vgaarb: deactivate vga console
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.36.0
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[Sat May 17 09:56:23 2025] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[Sat May 17 09:56:23 2025] [drm] Initialized i915 1.6.0 20230929 for 0000:00:02.0 on minor 0
[Sat May 17 09:56:24 2025] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[Sat May 17 09:56:24 2025] i915 0000:00:02.0: 7 VFs could be associated with this PF
```
the driver seems loaded, but lspci shows only one device
```shell
root@pve:~# lspci | grep 00:02
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics] (rev 0c)
root@pve:~#
```

bbaa-bbaa, 2025-05-17:
After loading the kernel driver module, you need to manually write to sysfs to create the VF.
```
echo 1 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
```
[Creating Virtual Functions (VF)](https://github.com/strongtz/i915-sriov-dkms?tab=readme-ov-file#creating-virtual-functions-vf)

You may need to check whether have a configuration file under /etc/modprobe.d that prevents the i915 module from loading automatically at startup.

miRemid, 2025-05-17:
> After loading the kernel driver module, you need to manually write to sysfs to create the VF.
> 
> ```
> echo 1 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
> ```

after `echo 7 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs` and remove `blacklist i915` from my pve-blacklist.conf, everything works fine

thanks :D