strongtz/i915-sriov-dkms #296: GPU crashing under Wayland?

yvvki, 2025-05-18:
I don't know how to debug further, here's the dmesg of `(i915)|(00:02)`:

```log
[    0.619686] pci 0000:00:02.0: [8086:9a49] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[    0.619702] pci 0000:00:02.0: BAR 0 [mem 0x603c000000-0x603cffffff 64bit]
[    0.619704] pci 0000:00:02.0: BAR 2 [mem 0x4000000000-0x401fffffff 64bit pref]
[    0.619706] pci 0000:00:02.0: BAR 4 [io  0x3000-0x303f]
[    0.619716] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.619718] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.619745] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x00ffffff 64bit]
[    0.619746] pci 0000:00:02.0: VF BAR 0 [mem 0x00000000-0x06ffffff 64bit]: contains BAR 0 for 7 VFs
[    0.619747] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0x1fffffff 64bit pref]
[    0.619748] pci 0000:00:02.0: VF BAR 2 [mem 0x00000000-0xdfffffff 64bit pref]: contains BAR 2 for 7 VFs
[    0.669377] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    0.669377] pci 0000:00:02.0: vgaarb: bridge control possible
[    0.669377] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=mem,locks=none
[    0.670203] pnp 00:03: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.670784] pnp 00:04: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671852] pnp 00:08: disabling [mem 0x000c0000-0x000c3fff] because it overlaps 0000:00:02.0 BAR 6 [mem 0x000c0000-0x000dffff]
[    0.671854] pnp 00:08: disabling [mem 0x000c8000-0x000cbfff] because it overlaps 0000:00:02.0 BAR 6 [mem 0x000c0000-0x000dffff]
[    0.671855] pnp 00:08: disabling [mem 0x000d0000-0x000d3fff] because it overlaps 0000:00:02.0 BAR 6 [mem 0x000c0000-0x000dffff]
[    0.671856] pnp 00:08: disabling [mem 0x000d8000-0x000dbfff] because it overlaps 0000:00:02.0 BAR 6 [mem 0x000c0000-0x000dffff]
[    0.671857] pnp 00:08: disabling [mem 0x00000000-0x0009ffff] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671858] pnp 00:08: disabling [mem 0x000c0000-0x000c3fff disabled] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671859] pnp 00:08: disabling [mem 0x000c8000-0x000cbfff disabled] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671860] pnp 00:08: disabling [mem 0x000d0000-0x000d3fff disabled] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671861] pnp 00:08: disabling [mem 0x000d8000-0x000dbfff disabled] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671862] pnp 00:08: disabling [mem 0x000e0000-0x000e3fff] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671863] pnp 00:08: disabling [mem 0x000e8000-0x000ebfff] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671864] pnp 00:08: disabling [mem 0x000f0000-0x000fffff] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671864] pnp 00:08: disabling [mem 0x00100000-0xa07fffff] because it overlaps 0000:00:02.0 BAR 7 [mem 0x00000000-0x06ffffff 64bit]
[    0.671866] pnp 00:08: disabling [mem 0x00000000-0x0009ffff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671867] pnp 00:08: disabling [mem 0x000c0000-0x000c3fff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671868] pnp 00:08: disabling [mem 0x000c8000-0x000cbfff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671869] pnp 00:08: disabling [mem 0x000d0000-0x000d3fff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671869] pnp 00:08: disabling [mem 0x000d8000-0x000dbfff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671870] pnp 00:08: disabling [mem 0x000e0000-0x000e3fff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671871] pnp 00:08: disabling [mem 0x000e8000-0x000ebfff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671872] pnp 00:08: disabling [mem 0x000f0000-0x000fffff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.671873] pnp 00:08: disabling [mem 0x00100000-0xa07fffff disabled] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    0.688403] pci 0000:00:02.0: VF BAR 2 [mem 0x4020000000-0x40ffffffff 64bit pref]: assigned
[    0.688406] pci 0000:00:02.0: VF BAR 0 [mem 0x4100000000-0x4106ffffff 64bit]: assigned
[    0.690150] pci 0000:00:02.0: Adding to iommu group 2
[    1.119340] i915: loading out-of-tree module taints kernel.
[    1.119439] i915: module verification failed: signature and/or required key missing - tainting kernel
[    1.502610] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
[    1.502612] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
[    1.502614] i915: Module Homepage: https://github.com/strongtz/i915-sriov-dkms
[    1.502763] i915 0000:00:02.0: enabling device (0006 -> 0007)
[    1.503030] i915 0000:00:02.0: [drm] Found TIGERLAKE/UY (device ID 9a49) display version 12.00 stepping C0
[    1.503051] i915 0000:00:02.0: Running in SR-IOV PF mode
[    1.505111] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.505164] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.505218] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.505615] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=mem
[    1.506706] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[    1.519730] i915 0000:00:02.0: [drm] *ERROR* GT0: IOV: Initialization failed (-EIO) GuC
[    1.519733] i915 0000:00:02.0: [drm] PF aborted (-EIO) intel_iov_init_late+0x73/0x80 [i915]
[    1.520622] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.638223] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[    2.650015] fbcon: i915drmfb (fb0) is primary device
[    2.650025] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    2.657886] i915 0000:00:02.0: No VFs could be associated with this PF!
[   17.043363] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   17.045487] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   17.401692] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   32.864034] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[   32.864103] i915 0000:00:02.0: [drm] walker[1868] context reset due to GPU hang
[   32.916841] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:e75ffefe, in walker [1868]
```

bbaa-bbaa, 2025-05-18:
https://github.com/strongtz/i915-sriov-dkms/blob/4de11bfac15ceb2780641b0abc0aac3496abb3d7/drivers/gpu/drm/i915/gt/iov/intel_iov.c#L292-L305

It seem that guc may not running correctly. Have you properly configured the `enable_guc` kernel parameter?

yvvki, 2025-05-18:
I tested with GuC value of both 3 and 0, the dmesg I provided was using the GuC value of 0. It shows flickers with a box of rapidly changing colors at the top-left before the system completely hangs regardless of the GuC value.

Does the dkms explicitly needs GuC/HuC enabled when using i915?