strongtz/i915-sriov-dkms #297: Unable to get it to run on Arrow Lake (proxmox 8.4/Kernel 6.14)

legacyc0de, 2025-05-18:
Having some issues trying to get it to run on Proxmox VE 8.4 with a 6.14 kernel, I'm not sure if it goes wrong somewhere along the way of if there some compatibility issue with newer hardware. So far I've tried various things on 3 different clean installs (with only the kernel upgraded)

I'm Getting a driver not bound error, No vfs showing up either. Confirmed the bios has VT-d and IOMMU enabled, Tried both with secure boot and without.

**Hardware :**
CPU : Intel Core ultra 7 265
MB : MSI Pro B860-p

```
root@pve:~# cat /proc/version
Linux version 6.14.0-2-pve (build@proxmox) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC PMX 6.14.0-2 (2025-04-10T17:57Z)

```
```
GRUB_CMDLINE_LINUX="quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 i915.force_probe=7d67 module_blacklist=xe"
```
```
root@pve:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```
```
root@pve:~# dkms status
i915-sriov-dkms/2025.05.11, 6.14.0-2-pve, x86_64: installed
```
```
root@pve:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Arrow Lake-S [Intel Graphics] (rev 06)
80:14.5 Non-VGA unclassified device: Intel Corporation Device 7f2f (rev 10)
```
```
root@pve:~# dmesg | grep "00:02"
[    0.330778] pci 0000:00:02.0: [8086:7d67] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[    0.330797] pci 0000:00:02.0: BAR 0 [mem 0x6817000000-0x6817ffffff 64bit pref]
[    0.330799] pci 0000:00:02.0: BAR 2 [mem 0x6000000000-0x600fffffff 64bit pref]
[    0.330812] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.330815] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.330840] pci 0000:00:02.0: PME# supported from D0 D3hot
[    0.330857] pci 0000:00:02.0: VF BAR 0 [mem 0x6810000000-0x6810ffffff 64bit pref]
[    0.330858] pci 0000:00:02.0: VF BAR 0 [mem 0x6810000000-0x6816ffffff 64bit pref]: contains BAR 0 for 7 VFs
[    0.358168] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    0.358168] pci 0000:00:02.0: vgaarb: bridge control possible
[    0.358168] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.361763] system 00:02: [io  0x0680-0x069f] has been reserved
[    0.361764] system 00:02: [io  0x164e-0x164f] has been reserved
[    0.378530] pci_bus 0000:02: resource 0 [io  0x9000-0x9fff]
[    0.378531] pci_bus 0000:02: resource 1 [mem 0x80000000-0x85ffffff]
[    0.378531] pci_bus 0000:02: resource 2 [mem 0x6010000000-0x680fffffff 64bit pref]
[    0.379541] pci 0000:00:02.0: Adding to iommu group 0
[    3.829924] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[    3.829937] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
```

bbaa-bbaa, 2025-05-18:
Check the output of lspci -vv -nn -s 00:02.
If the "Kernel driver in use" is missing, try running modprobe i915 and check the latest message of dmesg.
https://github.com/strongtz/i915-sriov-dkms/issues/292#issuecomment-2878353816

legacyc0de, 2025-05-18:
> Check the output of lspci -vv -nn -s 00:02. If the "Kernel driver in use" is missing, try running modprobe i915 and check the latest message of dmesg. [#292 (comment)](https://github.com/strongtz/i915-sriov-dkms/issues/292#issuecomment-2878353816)

It seems to be missing here.

```
root@pve:~# lspci -vv -nn -s 00:02.
00:02.0 VGA compatible controller [0300]: Intel Corporation Arrow Lake-S [Intel Graphics] [8086:7d67] (rev 06) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Micro-Star International Co., Ltd. [MSI] Arrow Lake-S [Intel Graphics] [1462:7e41]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        IOMMU group: 0
        Region 0: Memory at 6817000000 (64-bit, prefetchable) [size=16M]
        Region 2: Memory at 6000000000 (64-bit, prefetchable) [size=256M]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag+ RBE+ FLReset+
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp+ 10BitTagReq+ OBFF Not Supported, ExtFmt+ EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [d0] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Null
        Capabilities: [110 v1] Process Address Space ID (PASID)
                PASIDCap: Exec- Priv-, Max PASID Width: 14
                PASIDCtl: Enable- Exec- Priv-
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable+, Smallest Translation Unit: 00
        Capabilities: [420 v1] Physical Resizable BAR
                BAR 2: current size: 256MB, supported: 256MB
        Capabilities: [320 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration- 10BitTagReq+ Interrupt Message Number: 000
                IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy- 10BitTagReq-
                IOVSta: Migration-
                Initial VFs: 7, Total VFs: 7, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 1, stride: 1, Device ID: 7d67
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000006810000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0
        Capabilities: [400 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Kernel modules: xe

```

The last part of dmesg

```
[67095.598745] i915: module verification failed: signature and/or required key missing - tainting kernel
[67095.780785] Setting dangerous option enable_guc - tainting kernel
[67095.780788] Setting dangerous option force_probe - tainting kernel
[67095.780993] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
[67095.780994] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
[... 12 lines cut ...]

root@pve:~# dmesg -T | grep "00:02"
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: [8086:7d67] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: BAR 0 [mem 0x6817000000-0x6817ffffff 64bit pref]
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: BAR 2 [mem 0x6000000000-0x600fffffff 64bit pref]
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: PME# supported from D0 D3hot
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x6810000000-0x6810ffffff 64bit pref]
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: VF BAR 0 [mem 0x6810000000-0x6816ffffff 64bit pref]: contains BAR 0 for 7 VFs
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: vgaarb: bridge control possible
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[... cut ...]
[Sat May 17 18:39:14 2025] pci_bus 0000:02: resource 2 [mem 0x6010000000-0x680fffffff 64bit pref]
[Sat May 17 18:39:14 2025] pci 0000:00:02.0: Adding to iommu group 0
[Sat May 17 18:39:17 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Sat May 17 18:39:17 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Sun May 18 13:37:43 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[Sun May 18 13:38:38 2025] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV

```

bbaa-bbaa, 2025-05-18:
try https://github.com/bbaa-bbaa/i915-sriov-dkms/commit/a976507083460a647cbd6f9211141b504a873c77
CI artifacts are available here https://github.com/bbaa-bbaa/i915-sriov-dkms/actions/runs/15096049284

bbaa-bbaa, 2025-05-18:
Thanks for your test.

PR #298 (2025-05-19): Add Arrow Lake pciids
Close #297