strongtz/i915-sriov-dkms #300: Update i915 DKMS and/or update PVE host

tomalta, 2025-05-19:
I got this driver fully working on PVE 8.3 but now I have quite a lot of pending updates, that also include kernel headers.
How should I go about updating my system? Do I need to take any preliminary steps? How do I updated the module itself?

I've not found any guide on how to do so. I'm also, unfortunately, running secure boot. Would I need to sign the updated module when installing it like I did the first time? Thanks!

johntdavis84, 2025-05-19:
Did you install via the .deb package that installs the DKMS module?

If so, the DKMS will re-compile itself for the new kernel whenever an updated kernel is installed. You'll be able to see this happening when you're watching the output of the system upgrade if you're doing it from the console with `apt dist-upgrade`.

Afterwards, call `dkms status` and you should see the module listed for every kernel it's installed for.

I'm honestly not sure about the Secure Boot issue. I _think_ you only have to do the MOK key signing once, but I can't be sure.

tomalta, 2025-05-20:
Thanks for answering again my question.
I wanted to ask because I don't remember the MOK password I had set up, but apparently it's just a one time password used to sign the driver so it shouldn't be needed once I upgrade the system. The key is enrolled for the DKMS so it shouldn't ask it I guess.

So to update the driver I should uninstall the previous DKMS and install the new version?

johntdavis84, 2025-05-20:
> Thanks for answering again my question. I wanted to ask because I don't remember the MOK password I had set up, but apparently it's just a one time password used to sign the driver so it shouldn't be needed once I upgrade the system. The key is enrolled for the DKMS so it shouldn't ask it I guess.

That is how it should work, yes. You shouldn't need the MOK password again.

> 
> So to update the driver I should uninstall the previous DKMS and install the new version?

You don't have to uninstall the old one. Download the new .deb and install it with `dpkg -i $debpkgname.deb`. The process will uninstall the old DKMS driver for you, and then compile the new driver for every kernel on the system that supports it. :)

tomalta, 2025-05-20:
> That is how it should work, yes. You shouldn't need the MOK password again.
> 
I need to find more details about this. I'm pretty new to secure boot and I'm struggling to find details about an upgrade to a self signed module. In theory I agree, but in practice I don't know if the module is considered different once I upgrade or not.

> You don't have to uninstall the old one. Download the new .deb and install it with `dpkg -i $debpkgname.deb`. The process will uninstall the old DKMS driver for you, and then compile the new driver for every kernel on the system that supports it. :)
>
Nice, that's handy!

johntdavis84, 2025-05-20:
> I need to find more details about this. I'm pretty new to secure boot and I'm struggling to find details about an upgrade to a self signed module. In theory I agree, but in practice I don't know if the module is considered different once I upgrade or not.

I think you might have to re-sign the new module, but I'm not sure. I don't use Secure Boot because it never worked right with Proxmox on the HP Elite Mini G9 600 I'm using for Proxmox. (And, honestly, stuff like this is a big reason that a lot of people just turn Secure Boot off when they install Proxmox.

Do you have display out/a keyboard on the Proxmox server so you can do the MOK signing if you need to?

> Though I need to add that I installed the module compiling it from source. Is that gonna make a difference? I know it creates a .deb file in the end but yet I'm not sure.

This is how I started out, as well. I was using the module before the .deb existed, so I would download and compile the source using the same guide that you linked. It's great. :)

To switch to the .deb package, I shut down all the VMs using the DKMS i915 driver, and did the manual uninstall as he describes. To be extra cautious, I rebooted Proxmox, and then downloaded the deb file and used it for a clean install. After that I just download and install the updated driver .deb files as they are released.

I'm not sure the reboot is necessary after uninstalling the manually built drivers, but I'm not yet running enough stuff that a couple of extra paranoid.

**You mentioned that you had a bunch of updates pending for Proxmox. I'd do the manual uninstall for the DKMS driver, reboot, check with `dkms status` to make sure it's gone, run the pending updates, and then install the .deb driver file. Since you did a clean install, be prepared to need to set up the MOK password again. When you're done, check `dkms status` again to make sure the driver is installed. I'd check it after every kernel update.**

tomalta, 2025-05-20:
> I think you might have to re-sign the new module, but I'm not sure. I don't use Secure Boot because it never worked right with Proxmox on the HP Elite Mini G9 600 I'm using for Proxmox. (And, [...]
I went through the guide I linked you and seems like an update is just a reinstall as if you were installing from scratch.

> Do you have display out/a keyboard on the Proxmox server so you can do the MOK signing if you need to?
> 
I do.

> This is how I started out, as well. I was using the module before the .deb existed, so I would download and compile the source using the same guide that you linked. It's great. :)
>
> To switch to the [...]

Are you sure I'd be able to boot if I uninstall the module and don't revert GRUB to default? I don't think it's gonna boot because the Intel driver is blacklisted and there are the vfs enabled.
Other than that I don't mind going through the manual process of compiling the module from scratch each time. If that procedure works I'd stick to that and be a little more patience, rather than venturing into the unknown. Even though the simplicity of the .deb package is tempting.

In the meantime I managed to find multiple sources reporting that the password used for the MOK enrollment is just a one time password to make sure you own the system and are not a remote attacker.

johntdavis84, 2025-05-20:
I'm not using GRUB. Since I'm not using secure boot, I'm on `SystemD-Boot`. However, the kernel command on boot is the same.

> `root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 console=ttyS0,115200n8 console=tty0`

**I didn't have to blacklist the xe driver on Proxmox itself, even though I do have to blacklist it in some of the VMs I use. I'm not sure why. Proxmox itself always just used the i915 driver unprompted.** 

When you uninstall the DKMS driver, the original i915 driver in the kernel will kick back in and work. I never had a problem with the system booting after uninstalling the DKMS driver. The default i915 lives in the kernel and can't be uninstalled, so it just loads again.

It's probably worth it to open a separate issue request to ask about whether you can just install the DKMS .deb package over a manual install of the driver and go on with your life, or if that will break something.

As far as how to manage the driver package, there's nothing wrong with doing updates manually. It's just more work, since you have to do it every time there's a kernel update even if there's not a new version of the DKMS driver. Definitely do whatever you're most comfortable with that gets you a stable system.

tomalta, 2025-05-20:
> I'm not using GRUB. Since I'm not using secure boot, I'm on `SystemD-Boot`. However, the kernel command on boot is the same.
> 
> **I didn't have to blacklist the xe driver on Proxmox itself, even though I do have to blacklist it in some of the VMs I use. I'm not sure why. Proxmox itself always just used the i915 driver unprompted.**
> 

I checked and you're right. There's no xe driver to blacklist on Proxmox.
I misread the original guide.

> When you uninstall the DKMS driver, the original i915 driver in the kernel will kick back in and work. I never had a problem with the system booting after uninstalling the DKMS driver. The default i915 lives in the kernel and can't be uninstalled, so it just loads again.
> 

That's good to know. I would hate to have to start back from scratch.

> It's probably worth it to open a separate issue request to ask about whether you can just install the DKMS .deb package over a manual install of the driver and go on with your life, or if that will break something.
> 

I hope it would not be taken as a double issue for the same problem.
I think they should try to detail an upgrade guide for it. If I manage to do it I might try to submit a guide to them.

> As far as how to manage the driver package, there's nothing wrong with doing updates manually. It's just more work, since you have to do it every time there's a kernel update even if there's not a [...]

If the manually compiled DKMS is not recompiled at every upgrade I don't think it's worth going forward with it. So I'm gonna move to the .deb package as you suggested.

Thanks for following me through all my doubts and questions, I very much appreciate it.