strongtz/i915-sriov-dkms #302: host hangs

c0d3z3r0, 2025-05-19:
I tried this on kernel v6.14.2 (custom kernel config). The Windows VM boots up fine but after login everything including the host starts lagging. It feels like boh VM and host graphics lock up but background (non-graphics tasks) still work. Interestingly, I still can move the mouse but cannot click anything (neither on host, nor inside the VM), the keyboard isn't responding as well. After about 20 seconds, things recover for about 5-10 seconds but then locks again.

Any idea what could be wrong? I'll try to provide more details step-by-step.

bbaa-bbaa, 2025-05-20:
It seems that is similar to #296 (Tiger Lake).
Could you upload the kernel logs (`dmesg` or `journalctl -k`) when the host hangs?

c0d3z3r0, 2025-05-20:
Nothing special in dmesg or journalctl, actually nothing at all when it hangs. After some more testing it looks like the host hangs as soon as the VM is utilizing the GPU. opening the start menu or UAC instantly cause hangs.

c0d3z3r0, 2025-05-20:
kernels v6.14.7 and v15-rc7 don't make any difference

c0d3z3r0, 2025-05-21:
Tested linux-intel-lts (6.12) and linux-intel-mainline (6.12) with debian config. Guess what? Same issue.

c0d3z3r0, 2025-05-22:
I got the backport dkms module working for a short time. At least I could log-in via RDP, but after opening a OpenGL app the whole host hang again (I could still use the console via SSH and move the mouse, but nothing else).

Kernel v6.8.8 (vanilla with debian config) + intel dkms tag I915_24WW33.3_950.13_24.4.12_240603.18

bbaa-bbaa, 2025-05-23:
> is the GOP binary really required?

In my environment, the GOP binary is not required for Windows Guest work.

bbaa-bbaa, 2025-06-08:
I have no idea about that. maybe you could try the xe or i915 driver in the latest `linux/v6.15` branch of [intel/mainline-tracking.git](https://github.com/intel/mainline-tracking).

c0d3z3r0, 2025-06-08:
Oh great! I missed that update

pacewicz, 2025-06-27:
It used to happen on my machine quite frequently, especially when starting the VM shortly after host boot - and sometimes still does. When used together with Looking Glass and VDD (Virtual Display Driver), it happened every time a Win11 guest rebooted and rarely on Win10 reboot - but the variables and the frustration were too much to go on testing that.

I guesstimate the hangs went down by at least 90% after I set up calling `virsh nodedev-detach` for all VFs at host boot via a systemd service: https://github.com/strongtz/i915-sriov-dkms/issues/8#issuecomment-1567465036

i5-1350P, 3 VFs, GOP driver does not seem to make much of a difference.
Mint Xia Cinnamon on kernels 6.8.0-55 through 59, then switched to 6.11.0-26 and 28
The "down by at least 90%" finding applies to the last two.

celesrenata, 2025-09-06:
I have been having similar problems, I found that Nvidia and Intel GPUs were not playing nice with each other, stability increased for me by moving to 6.15.11 and only allowing intel or nvidia depending on what's accelerating that node.