strongtz/i915-sriov-dkms #303: SR-IOV Virtual Functions Fail to Initialize on Kernel 6.14.0-2-pve with Intel Alder Lake-P iGPU

keilwill, 2025-05-22:
### **Environment**
- **Proxmox Version:** 8.4.1
- **Kernel:** `6.14.0-2-pve`
- **Hardware:** 
  - **CPU:** Intel 12th Gen i5-12450H (Alder Lake-P)
  - **iGPU:** `8086:46a3` (Intel UHD Graphics)
- **Module Version:** `i915-sriov-dkms/2025.05.18`
- 
The `i915-sriov-dkms` module is installed and loaded, but VFs do not appear.
SR-IOV capabilities exist in `lspci -vv` but remain uninitialized.

#### ~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P GT1 [UHD Graphics] (rev 0c)

#### ~# lspci -vv -nn -s 00:02

00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-P GT1 [UHD Graphics] [8086:46a3] (rev 0c) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Intel Corporation Alder Lake-P GT1 [UHD Graphics] [8086:2212]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 201
        IOMMU group: 0
        Region 0: Memory at 6001000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 4000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+ FLReset+
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
                Address: fee00018  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [d0] Power Management version 2
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D3 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Process Address Space ID (PASID)
                PASIDCap: Exec- Priv-, Max PASID Width: 14
                PASIDCtl: Enable- Exec- Priv-
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable+, Smallest Translation Unit: 00
        Capabilities: [300 v1] Page Request Interface (PRI)
                PRICtl: Enable- Reset-
                PRISta: RF- UPRGI- Stopped+
                Page Request Capacity: 00008000, Page Request Allocation: 00000000
        Capabilities: [320 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration- 10BitTagReq- Interrupt Message Number: 000
                IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy- 10BitTagReq-
                IOVSta: Migration-
                Initial VFs: 7, Total VFs: 7, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 1, stride: 1, Device ID: 46a3
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000004010000000 (64-bit, non-prefetchable)
                Region 2: Memory at 0000004020000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0
        Kernel driver in use: i915
[... 47 lines cut ...]
[   10.260375] i915: You are using the i915-sriov-dkms module, a ported version of the i915 module with SR-IOV support.
[   10.260389] i915: Please file any bug report at https://github.com/strongtz/i915-sriov-dkms/issues/new.
[   10.260392] i915: Module Homepage: https://github.com/strongtz/i915-sriov-dkms
[   10.260890] i915 0000:00:02.0: [drm] Found ALDERLAKE_P (device ID 46a3) display version 13.00 stepping D0
[   10.262108] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   10.262114] i915 0000:00:02.0: vgaarb: deactivate vga console
[   10.262190] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   10.262670] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   10.268407] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[   10.273906] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.36.0
[   10.273918] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[   10.290821] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[   10.291270] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   10.291274] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   10.291669] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[   10.318670] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   10.318845] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   10.318853] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   10.335739] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 0
[   10.352009] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   10.381982] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   10.396978] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

#### ~# echo 7 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: Numerical result out of range

h0m35t4r, 2025-05-23:
Make sure your kernel command line parameters are all in one line. Anything appended with a new line will be ignored

keilwill, 2025-05-23:
Wow, thanks! Maybe there was too much spaces. I add again
intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
then ran 
update-grub
update-initramfs -u
After a reboot, all VFs are there.

closed 2025-05-23 by keilwill