strongtz/i915-sriov-dkms #308: aoostar wtr pro n150 (pve8.4 kernel6.14.5-1-bpo12-pve)-lxc vainfo init failed

DongGamHe, 2025-06-05:
I updated the kernel from 6.8 to 6.14, and

I completed the DKMS build, MOK password input, and reboot to confirm the password.

The dkms status result is confirmed as installed,

The lspci | grep VGA results and dmesg | grep i915 results are as follows.
[image]

[image]

[image]

The pve host seems to be set up well, but now a problem has occurred.

I added an lxc container using a script that many people say was successful (ubuntu 22)

and set the lxcNo.conf file as follows.

[image]

After rebooting the lxc container
apt install ffmpeg
apt install vainfo clinfo intel-opencl-icd intel-media-va-driver-non-free
I installed the necessary things through the two commands and now all that's left is testing

vainfo / clinfo result

[image]

ffmpeg -v verbose -init_hw_device vaapi=va -init_hw_device opencl@va result

[image]

I've tried a lot of Googling and advice but nothing seems to work. Is there anything else I can try?

wouldntyouknow, 2025-06-08:
Did you install SRIOV because of Jellyfin HW transcoding in an LXC? You don't need SRIOV for that, the LXC uses the host HW directly. You would only need this for VMs, really.
You also don't need the cgroup2 stuff in your config. Since 8.4 is out most popular Jellyfin tutorials out there can be considered outdated.
You can pass card0 and renderD128 withe Device -> Device Passthrough in the GUI, nothing else is needed.
Just check groups (getent group) in the LXC. You need groups "video" and "render" and their IDs. Make sure user "jellyfin" is in these groups, also.
For me, video is 44 and render is 104 - probably the same for you as it's ubuntu.
So you add device "/dev/dri/card0" with "GID in CT" being 44, and "/dev/dri/renderD128" with "GID in CT" being 104
Restart the LXC and check, that's it

DongGamHe, 2025-06-08:
I need gpu passthrouth for a total of three lxc. Of course, three LxCs rarely run simultaneously, but three LxCs will always be powered on. Is there any problem with delivering card0 and RenderD128 to all three lxcs as I told you?

wouldntyouknow, 2025-06-09:
This is not the best forum to troubleshoot this exact question, but I'll give it one more go, comparing to mine:
I am running Ubuntu 22.04 (Jammy J) as the host in the LXC, with Jellyfin. If I remember correctly I also used a tteck helper script as you did, to install this. Maybe start from scratch? It can be an unprivileged container!
So after install, I made sure Jellyfin is in the proper groups, if not, added:
```
video:x:44:jellyfin
render:x:104:jellyfin
```

Added the 2 device passthrough objects according to the group IDs:
[image]

In the LXC:
```
root@jellyfin:~# ls -la /dev/dri
total 0
drwxr-xr-x 2 root root         80 Jun  3 22:15 .
drwxr-xr-x 7 root root        500 Jun  3 22:15 ..
crw-rw---- 1 root video  226,   0 Jun  3 22:15 card0
crw-rw---- 1 root render 226, 128 Jun  3 22:15 renderD128
```
So up to this point it looks good on your end also.
My vainfo:
```
root@jellyfin:~# vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
```
Which makes me wonder maybe it's a driver or something that's missing on your end? Other things being equal as it seems. I don't remember what I installed so here is some parts of my apt list that I think might be relevant:
i965-va-driver/jammy,now 2.4.1+dfsg1-1 amd64 [installed,automatic]
intel-gpu-tools/jammy-updates,now 1.26-2ubuntu0.1 amd64 [installed]
intel-media-va-driver/jammy-updates,now 22.3.1+dfsg1-1ubuntu2 amd64 [installed,automatic]
intel-opencl-icd/jammy,now 22.14.22890-1 amd64 [installed]
mesa-va-drivers/jammy-updates,now 23.2.1-1ubuntu3.1~22.04.3 amd64 [installed,automatic]
mesa-vdpau-drivers/jammy-updates,now 23.2.1-1ubuntu3.1~22.04.3 amd64 [installed,automatic]
ocl-icd-libopencl1/jammy,now 2.2.14-3 amd64 [installed]
va-driver-all/jammy,now 2.14.0-1 amd64 [installed]
vainfo/jammy,now 2.12.0+ds1-1 amd64 [installed]
vdpau-driver-all/jammy,now 1.4-3build2 amd64 [installed,automatic]

Hope this helps, out of ideas.

DongGamHe, 2025-06-09:
Thank you so much for your kind reply!!

I know this is not the right place to ask, but I have one last request.

Could you please also tell me the contents of the /etc/pve/lxc/lxcNo.conf file?

wouldntyouknow, 2025-06-09:
Sure here is the relevant content of the conf file:

root@proxmox:/etc/pve/lxc# cat 110.conf
arch: amd64
cores: 2
cpuunits: 120
dev0: /dev/dri/card0,gid=44
dev1: /dev/dri/renderD128,gid=104
features: nesting=1
hostname: jellyfin
memory: 3072
mp0: /mnt/lxc_jellyfin/jellyfin/,mp=/mnt/jellyfin
net0: name=eth0,bridge=vmbr0,gw=192.168.0.1,hwaddr=BC:24:11:BD:96:58,ip=192.168.0.12/24,type=veth
onboot: 1
ostype: ubuntu
rootfs: data:vm-110-disk-0,size=8380M
startup: order=9,up=5
swap: 0
tags:  
unprivileged: 1