strongtz/i915-sriov-dkms #316: Windows 11 Pro (Builtin Remote Desktop Server) VM with Working UHD 770 Passthrough and Microsoft Windows.app (MacOS): Windows Remote Desktop Service Pegged at 32 FPS?

johntdavis84, 2025-07-12:
Hello,

I've got an odd issue here.

I've got a Windows 11 VM set up with a working SR-IOV PCIe device. I can see it using the physical hardware inside and outside the VM.
I'm using Windows 11 Pro's built in RDP server, and the Windows.app (formerly Microsoft Remote Desktop) client on a Mac.

It's all working great, except that the Windows Remote Desktop Service itself seems to be capped at 32 FPS according to the UFO test and the OS itself.

I've adjusted the registry to enable 60 FPS, as per this article: https://learn.microsoft.com/en-us/troubleshoot/windows-server/remote/frame-rate-limited-to-30-fps .

**Note that I've raised the cap to _60 FPS_ and I'm currently seeing a cap imposed at _32 FPS**. So, I'm confident that the registry tweak worked. But something else is clearly wrong._

I also set the Group Policy settings for remote hardware acceleration here: https://learn.microsoft.com/en-us/azure/virtual-desktop/graphics-enable-gpu-acceleration?tabs=group-policy

_**Debug Info**_
Let me know if any other details would be helpful.

Proxmox 8.4 VM Hardware Setup

[image]

```
PS C:\Users\John T Davis\Documents> Get-WmiObject win32_videocontroller

RunspaceId                   : 7b1b960e-cab3-43e2-8f24-c7152fd13fe7
__GENUS                      : 2
__CLASS                      : Win32_VideoController
__SUPERCLASS                 : CIM_PCVideoController
__DYNASTY                    : CIM_ManagedSystemElement
__RELPATH                    : Win32_VideoController.DeviceID="VideoController1"
__PROPERTY_COUNT             : 59
__DERIVATION                 : {CIM_PCVideoController, CIM_VideoController, CIM_Controller, CIM_LogicalDevice…}
__SERVER                     : WENDY-DROMEDA2
__NAMESPACE                  : root\cimv2
__PATH                       : \\WENDY-DROMEDA2\root\cimv2:Win32_VideoController.DeviceID="VideoController1"
AcceleratorCapabilities      :
AdapterCompatibility         : Microsoft
AdapterDACType               :
AdapterRAM                   :
Availability                 : 3
CapabilityDescriptions       :
Caption                      : Microsoft Remote Display Adapter
ColorTableEntries            :
ConfigManagerErrorCode       : 0
ConfigManagerUserConfig      : False
CreationClassName            : Win32_VideoController
CurrentBitsPerPixel          : 32
CurrentHorizontalResolution  : 2048
CurrentNumberOfColors        : 4294967296
CurrentNumberOfColumns       : 0
CurrentNumberOfRows          : 0
CurrentRefreshRate           : 32
CurrentScanMode              : 4
CurrentVerticalResolution    : 1152
Description                  : Microsoft Remote Display Adapter
DeviceID                     : VideoController1
DeviceSpecificPens           :
DitherType                   : 0
DriverDate                   : 20060621000000.000000-000
DriverVersion                : 10.0.22621.5415
ErrorCleared                 :
ErrorDescription             :
ICMIntent                    :
ICMMethod                    :
InfFilename                  : rdpidd.inf
InfSection                   : RdpIdd_Install.NT
InstallDate                  :
InstalledDisplayDrivers      :
LastErrorCode                :
MaxMemorySupported           :
MaxNumberControlled          :
MaxRefreshRate               : 32
MinRefreshRate               : 32
Monochrome                   : False
Name                         : Microsoft Remote Display Adapter
NumberOfColorPlanes          :
NumberOfVideoPages           :
PNPDeviceID                  : SWD\REMOTEDISPLAYENUM\RDPIDD_INDIRECTDISPLAY&SESSIONID_0002
PowerManagementCapabilities  :
PowerManagementSupported     :
ProtocolSupported            :
ReservedSystemPaletteEntries :
SpecificationVersion         :
Status                       : OK
StatusInfo                   :
SystemCreationClassName      : Win32_ComputerSystem
SystemName                   : WENDY-DROMEDA2
SystemPaletteEntries         :
TimeOfLastReset              :
VideoArchitecture            : 5
VideoMemoryType              : 2
VideoMode                    :
VideoModeDescription         : 2048 x 1152 x 4294967296 colors
VideoProcessor               :

RunspaceId                   : 7b1b960e-cab3-43e2-8f24-c7152fd13fe7
__GENUS                      : 2
__CLASS                      : Win32_VideoController
__SUPERCLASS                 : CIM_PCVideoController
__DYNASTY                    : CIM_ManagedSystemElement
__RELPATH                    : Win32_VideoController.DeviceID="VideoController2"
__PROPERTY_COUNT             : 59
__DERIVATION                 : {CIM_PCVideoController, CIM_VideoController, CIM_Controller, CIM_LogicalDevice…}
__SERVER                     : WENDY-DROMEDA2
__NAMESPACE                  : root\cimv2
__PATH                       : \\WENDY-DROMEDA2\root\cimv2:Win32_VideoController.DeviceID="VideoController2"
AcceleratorCapabilities      :
AdapterCompatibility         : Intel Corporation
AdapterDACType               : Internal
AdapterRAM                   : 2147479552
Availability                 : 8
CapabilityDescriptions       :
Caption                      : Intel(R) UHD Graphics 770
ColorTableEntries            :
ConfigManagerErrorCode       : 0
ConfigManagerUserConfig      : False
CreationClassName            : Win32_VideoController
CurrentBitsPerPixel          :
CurrentHorizontalResolution  :
CurrentNumberOfColors        :
CurrentNumberOfColumns       :
CurrentNumberOfRows          :
CurrentRefreshRate           :
CurrentScanMode              :
CurrentVerticalResolution    :
Description                  : Intel(R) UHD Graphics 770
DeviceID                     : VideoController2
DeviceSpecificPens           :
DitherType                   :
DriverDate                   : 20250621000000.000000-000
DriverVersion                : 32.0.101.6913
ErrorCleared                 :
ErrorDescription             :
ICMIntent                    :
ICMMethod                    :
InfFilename                  : oem19.inf
InfSection                   : iADLSD_w11_DS
InstallDate                  :
InstalledDisplayDrivers      : C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_507e62014afcde30\igdumdim64.dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_507e62014afcde30\igd10iumd64
                               .dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_507e62014afcde30\igd10iumd64.dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_507e62014afcde30\igd12
                               umd64.dll
LastErrorCode                :
MaxMemorySupported           :
MaxNumberControlled          :
MaxRefreshRate               :
MinRefreshRate               :
Monochrome                   : False
Name                         : Intel(R) UHD Graphics 770
NumberOfColorPlanes          :
NumberOfVideoPages           :
PNPDeviceID                  : PCI\VEN_8086&DEV_4680&SUBSYS_8952103C&REV_0C\4&23F93F77&0&00E0
PowerManagementCapabilities  :
PowerManagementSupported     :
ProtocolSupported            :
ReservedSystemPaletteEntries :
SpecificationVersion         :
Status                       : OK
StatusInfo                   :
SystemCreationClassName      : Win32_ComputerSystem
SystemName                   : WENDY-DROMEDA2
SystemPaletteEntries         :
TimeOfLastReset              :
VideoArchitecture            : 5
VideoMemoryType              : 2
VideoMode                    :
VideoModeDescription         :
VideoProcessor               : Intel(R) UHD Graphics Family
```

_**Other Notes**_
Using the same client with identical client-side settings, I've confirmed that I can get 60 fps server-side from an Ubuntu VM running the `[egfx-xrdp](https://github.com/neutrinolabs/xrdp/discussions/2383)` RDP server. Interestingly, this VM reports 60 FPS from the OS via the UFO test, even though the `egfx-xrdp` server is capped at 25 FPS (that's not a bug...that Ubuntu VM is old). This is some sort of server-side issue with the Windows 11 VM.

uzumo, 2025-07-30:
All appear to be at 32fps.
If setting the registry doesn't reflect this, then I think the 60fps registry just stopped working.
You need to contact Microsoft if you want to make a request.

Get-WmiObject win32_videocontroller | fl CurrentRefreshRate,MaxRefreshRate,MinRefreshRate

luckman212, 2025-07-30:
Anyone got smooth/high framerates with
- PVE 8.4.5
- kernel 6.14.8-2-bpo12
- dkms 2025.07.22
- MS RDP to a Win11 guest VM

I have a "working" Irix Xe GPU in my Device Manager but framerates are capping at 32fps. I've got an i7-1255U Alder Lake, done all the GPO + registry tweaks, extracted and set the UEFI ROM etc... can't solve it

```
# dkms status
i915-sriov-dkms/2025.07.22, 6.14.8-2-bpo12-pve, x86_64: installed
```

Parsec works great btw.

luckman212, 2025-07-31:
In my case, yes it's the macOS client. Did not know that was artificially capped at 32fps.

johntdavis84, 2025-07-31:
> luckman212

It’s confusing because by default the RDP server in Windows 11 Pro is also capped at 30 FPS. That seems to be the supported config, but Microsoft themselves publish a support article re: enabling 60 FPS RDP.

But the client itself has to ask for a 60 FPS connection, and the Windows.app RDP client does not and is also too under-featured to let us override that setting.

If there’s a hidden way to override that, I haven’t found it yet.

uzumo, 2025-07-31:
It may not change, but try deleting the registry and adding it with the reg add command described.

I added it with regedit, reconnected the session, and it did not change to 66fps, so I deleted it with regedit, added the ms document with reg add, and got 66fps.

There might be a mistake in the description or something.

johntdavis84, 2025-08-15:
I'm closing this, as it seems to be an issue with my particular setup, not a problem with the i915 DKMS driver.

johntdavis84, 2025-09-01:
**Coming back to provide an update on this:**
This weekend, I had to build a new Windows 11 VM from scratch after somehow torching the last one.

The registry hack described here-- https://learn.microsoft.com/en-us/troubleshoot/windows-server/remote/frame-rate-limited-to-30-fps --worked perfectly on the first try. Getting the expected 66 FPS now.

@uzumo Thanks for your help before. Clearly, something was just broken with that VM.