strongtz/i915-sriov-dkms #318: 2 simultaneous VM or VM and LXC using the iGPU monopoly

kraizelburg, 2025-07-13:
Hi, first of all thanks for this great solution and the effort!! 

Now, back to the problem I am having. Whenever i try to use simultaneously 2 igpu resources like 2 vm or lxc and a vm or any kind of combination then one monopolizes the igpu resources and making the other stall. 

Let's say I am transcoding in a linux vm and then on windows vm watching a video, both using the igpu but only one seems to work, as soon as I play a video on the windows vm then the transcoding on linux stops or the other way around.

Has anyone noticed this too? thanks

zz541843, 2025-08-09:
This problem gives me a headache. The performance of Windows through vf1 is almost the same as that of the direct passthrough integrated graphics, but when vf2 is working, Windows is instantly unavailable and has to wait for vf2 to complete its work. No matter how much vf2 is occupied, checking the intel_gpu_tools table under pve, even if vf2 has only a small occupancy, it will cause the Windows screen to freeze and become unusable.

zz541843, 2025-08-09:
This problem gives me a headache. The performance of Windows through vf1 is almost the same as that of the direct passthrough integrated graphics, but when vf2 is working, Windows is instantly unavailable and has to wait for vf2 to complete its work. No matter how much vf2 is occupied, checking the intel_gpu_tools table under pve, even if vf2 has only a small occupancy, it will cause the Windows screen to freeze and become unusable.
etc.
[image]

At this time, the frame rate of 4K video played by Windows through PotPlayer drops from 60fps to 10fps or even lower. When VF2 is suspended, it will rise to 60fps

lutfijw, 2025-08-27:
I have the same issue, Windows VM always stall when it try to process h264 Video but H265 video are working fine even when GPU is utilized simultaneously by two VM or combination of VM and lxc.
My other VM is processing h264 video, Frigate LXC, and when Windows VM try to process h264 video, it stall but GPU is working fine when Windows VM is processing h265 video, even though Frigate LXC is also occupied the GPU.
Maybe the issue only happen when two VM/LXC are processing same video codec.

j3n0v4, 2025-11-20:
Kernel: Linux 6.17.2-1-pve
Proxmox VE: 9.0.18

Hello there,

I believe I am having the same issue. I am currently investigating and hoping to find a solution. I am trying to get two VMs (Frigate and Jellyfin) to use the iGPU simulaniously. Frigate streams crash and report:

frigate  | 2025-11-20 07:19:56.154435493  [2025-11-20 07:19:56] ffmpeg.front.detect            ERROR   : [vist#0:0/h264 @ 0x5649fe572f00] [dec:h264_qsv @ 0x5649fe5d7dc0] Decoding error: Input/output error
frigate  | 2025-11-20 07:19:56.154554347  [2025-11-20 07:19:56] ffmpeg.front.detect            ERROR   : [h264_qsv @ 0x5649fe5d1300] Error during QSV decoding.: GPU Hang (-21)

for all streams while Jellyfin is also (successfully) transcoding streams.

Checkout for what I have tried and my progress:
https://forum.level1techs.com/t/minisforum-ms-01-sriov-i915-igpu-to-vms-for-jellyfin-frigate/241054