strongtz/i915-sriov-dkms #321: i915-sriov-dkms doesn't work on proxmox kernel 6.8.12

kraizelburg, 2025-07-19:
Hi, I just wanted to mention that sr-iov doesn't work after updating to proxmox 8.4.5 from previous version 8.4.1, kernel now is 6.8.12 and I believe before was 6.8.10.

I even reinstalled the .deb package but it does not work anymore.

`root@ms01:~# qm start 100
kvm: -device vfio-pci,host=0000:00:02.1,id=hostpci1,bus=ich9-pcie-port-2,addr=0x0,x-vga=on: vfio 0000:00:02.1: error getting device from group 16: Permission denied
Verify all devices in group 16 are bound to vfio-<bus> or pci-stub and not already in use
start failed: QEMU exited with code 1
root@ms01:~# uname -a
Linux ms01 6.8.12-12-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-12 (2025-07-14T13:20Z) x86_64 GNU/Linux`

luckman212, 2025-07-19:
Same on kernel 6.14.8-1

jfhenriques, 2025-07-19:
In my case, I was only able to start the guest VM again, but the iGPU is not usable. It just hangs as soon I try to use it for any purpose.

jfhenriques, 2025-07-19:
Following is the list of packages that were upgraded at the same time of the kernel

```
Start-Date: 2025-07-19  11:34:28
Commandline: apt dist-upgrade
Install: proxmox-kernel-6.8.12-12-pve-signed:amd64 (6.8.12-12, automatic), proxmox-headers-6.8.12-12-pve:amd64 (6.8.12-12, automatic)
Upgrade: pve-edk2-firmware-ovmf:amd64 (4.2025.02-3, 4.2025.02-4~bpo12+1), proxmox-widget-toolkit:amd64 (4.3.11, 4.3.12), pve-firmware:amd64 (3.15-4, 3.16-3), zfs-zed:amd64 (2.2.7-pve2, 2.2.8-pve1), zfs-initramfs:amd64 (2.2.7-pve2, 2.2.8-pve1), proxmox-headers-6.8:amd64 (6.8.12-11, 6.8.12-12), spl:amd64 (2.2.7-pve2, 2.2.8-pve1), pve-qemu-kvm:amd64 (9.2.0-6, 9.2.0-7), libnvpair3linux:amd64 (2.2.7-pve2, 2.2.8-pve1), libpve-cluster-api-perl:amd64 (8.1.1, 8.1.2), pve-edk2-firmware-legacy:amd64 (4.2025.02-3, 4.2025.02-4~bpo12+1), libproxmox-backup-qemu0:amd64 (1.5.1, 1.5.2), libuutil3linux:amd64 (2.2.7-pve2, 2.2.8-pve1), libzpool5linux:amd64 (2.2.7-pve2, 2.2.8-pve1), proxmox-kernel-6.8:amd64 (6.8.12-11, 6.8.12-12), pve-cluster:amd64 (8.1.1, 8.1.2), proxmox-backup-file-restore:amd64 (3.4.2-1, 3.4.3-1), proxmox-backup-client:amd64 (3.4.2-1, 3.4.3-1), pve-manager:amd64 (8.4.1, 8.4.5), libpve-notify-perl:amd64 (8.1.1, 8.1.2), libzfs4linux:amd64 (2.2.7-pve2, 2.2.8-pve1), pve-edk2-firmware:amd64 (4.2025.02-3, 4.2025.02-4~bpo12+1), zfsutils-linux:amd64 (2.2.7-pve2, 2.2.8-pve1), libpve-cluster-perl:amd64 (8.1.1, 8.1.2)
End-Date: 2025-07-19  11:36:20
```

paulzzh, 2025-07-19:
related #312
The last kernel where SR-IOV was working is **6.14.5-1-bpo12-pve**.

jfhenriques, 2025-07-20:
I tried downgrading the firmware on the host, to try other versions of GuC, but no effect.
I don't know why I keep getting GUC: TLB invalidation response timed out

CaesarBash, 2025-07-21:
> 相关[#312](https://github.com/strongtz/i915-sriov-dkms/issues/312) SR-IOV 运行的最后一个内核是**6.14.5-1-bpo12-pve**。

Unfortunately this may be a kernel issue that can only be pin the kernel version.

CaesarBash, 2025-07-21:
What I know now is that the two and versions are not affected by this problem
1: 6.8.12-11
2: 6.14.5-1-bpo12-pve.

CaesarBash, 2025-07-21:
this problem only affects KVM. LXC using phys passthrough is not affected by this problem. It can be used normally.

rjblake, 2025-07-22:
A fix has been added to Proxmox in the pvetest repo. Confirmed working; or wait for it to hit the pve-no-subscription or enterprise repos