strongtz/i915-sriov-dkms #322: Permission issues when using i915-sriov-dkms 2025.05.18 on Proxmox kernel 6.8.12-12-pve

brokenbuddha, 2025-07-20:
I'm having some issues booting an Ubuntu 24.04 vm after updating the Proxmox kernel to 6.8.12-12-pve as it seems there is a problem with the vm not having permission to use the passed through vGPU.

The issue only occurs for me on the updated 6.8.12-12-pve kernel so I have pinned 6.8.12-11-pve for now.

I'm using the latest i915-sriov-dkms 2025.05.18 version on both the Proxmox node and Ubuntu vm.

The following error is present in the Proxmox console when trying to boot the vm:

```
kvm: -device vfio-pci,host=0000:00:02.1,id=hostpci0,bus=ich9-pcie-port-1,addr=0x0: vfio 0000:00:02.1: error getting device from group 22: Permission denied
Verify all devices in group 22 are bound to vfio-<bus> or pci-stub and not already in use

TASK ERROR: start failed: QEMU exited with code 1
```

johntdavis84, 2025-07-20:
This seems to be an upstream issue in the 6.8.12-12-pve kernel/VFIO.

Over on the Proxmox forum, similar errors are also hitting NICs and other cards with SR-IOV support. Rolling back to the 6.8.12-11-pve kernel and pinning it seems to be the only way to fix it until something gets patched.

ohnggni, 2025-07-20:
Exactly same issue from 6.14.5-1-bpo12-pve to 6.14.8-1-bpo12-pve. I finally rolled back to 6.14.5-pve kernel and fixed it.

rjblake, 2025-07-22:
> by the way, this problem only affects KVM. LXC using phys passthrough is not affected by this problem. It can be used normally.

Indeed, noticed this same behaviour on my setup. Oddly I first saw this trying to backup a VM but that kept failing. Have a few LXCs that are all running fine. Just my Windows VMs won't even start. Guess I'll go back to previous kernel for now.

uzumo, 2025-07-22:
It should be fixed in kernel 6.8.12-13

rjblake, 2025-07-22:
A fix has been added to Proxmox in the pvetest repo. Confirmed working; or wait for it to hit the pve-no-subscription or enterprise repos

wouldntyouknow, 2025-07-25:
Same on 6.14.8-2, all good as expected

feerlessleadr, 2025-08-05:
now that PVE 9 final has dropped, am I good to assume the fix has hit the pve-no-subscription repo? I've been holding off updating until then.

johntdavis84, 2025-08-07:
You're welcome. :)

PBS 4 is out today. I'd wait at least a week to let them iron out the killer glitches.
There's still an issue with PBS 4 and PVE 9 both sometimes wiping out people's EFI partitions due to having to change how SystemD-Boot is installed. A lot of people got bricked.