strongtz/i915-sriov-dkms #325: How to uipdate to a current release from previous one

kraizelburg, 2025-07-25:
Hi, sorry this is not an issue rather a question. 

How can I update from the previous release to the latest one?

Thanks

johntdavis84, 2025-07-25:
If you used the .deb file, just download the new one and install it with `dpkg -i $debfilename`.
It will automatically uninstall the old version and build and install the new version.
Its output will show you what it's doing.
When it's done, reboot and check `dkms status` to make sure you're running the latest version.

Did you build manually from source or use some other method?

Sauryc, 2025-07-26:
> 如果您使用的是 .deb 文件，只需下载新版本并使用 进行安装`dpkg -i $debfilename`。 它会自动卸载旧版本，并构建和安装新版本。 其输出将显示正在执行的操作。 完成后，请重新启动并检查`dkms status`以确保您运行的是最新版本。
> 
> 您是从源代码手动构建还是使用了其他方法？

I would also like to know how to switch from building from source code to installing with .DED packages.

kraizelburg, 2025-07-26:
> If you used the .deb file, just download the new one and install it with `dpkg -i $debfilename`. It will automatically uninstall the old version and build and install the new version. Its output [...]

Thanks, all good now, I did not know it was so easy. I used the .deb package. Thanks again.

johntdavis84, 2025-07-26:
> Thanks, all good now, I did not know it was so easy. I used the .deb package. Thanks again.

Awesome! This is the first DKMS .deb package I've ever used; I wasn't sure how to update it either. We should probably get that into the README. :)

> Is it important? I just checked and I for example seemed to had built it from source. Can I just grab the new deb?

I wouldn't. It _should_ work fine, but if there's any difference at all in how the manual install and the .deb build and install the driver, it could lead to issues.

The safer path is to deactivate the VMs and LXCs using the driver, manually uninstall the driver from the host (leaving all your settings in place), and run the .deb.
For manual uninstall instructions, see: https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

```
rm -rf /usr/src/i915-sriov-dkms-*
rm -rf /var/lib/dkms/i915-sriov-dkms
rm -rf ~/i915-sriov-dkms*
find /lib/modules -regex ".*/updates/dkms/i915.ko" -delete
```

After that, install the .deb package and you should be fine.