strongtz/i915-sriov-dkms #326: PMU Support

lutfijw, 2025-08-04:
Hi All,
Follow the guidance to enable SRIOV on my Intel Arrow Lake GPU. Everything seem to be working, however I no longer able to check my GPU usage using intel_gpu_top.
Seem like the driver is not supporting PMU. Is there any plan to introduce PMU support for this driver?

  30.953990] i915 0000:00:02.1: [drm] PMU not supported for this GPU.

johntdavis84, 2025-08-04:
> Seem like the driver is not supporting PMU. Is there any plan to introduce PMU support for this driver?
I think that fix would have to come from upstream.

Have you tried `intel_gpu_top -d sriov` from the host PC?
That gets me live monitoring of the iGPU on my 12700T-based machine when handing your VFs to virtual machines in Proxmox.

ojsef39, 2025-08-11:
> Have you tried `intel_gpu_top -d sriov` from the host PC? That gets me live monitoring of the iGPU on my 12700T-based machine when handing your VFs to virtual machines in Proxmox.

I tried that in the vm (there it says no sriov device) and on the host for some reason if i filter for card0 or use -d sriov it will just freez but not show any stats, maybe i have set something up wrong (testing it out on pve 9)

johntdavis84, 2025-08-11:
> ojsef39
>  left a comment
> (strongtz/i915-sriov-dkms#326)
> intel_gpu_top -L works btw (5 is mounted to a vm):
>
> intel_gpu_top -L
> card1                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=0
> └─renderD128
> card0                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=1
> └─renderD129
> card2                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=2
> └─renderD130
> card3                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=3
> └─renderD131
> card4                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=4
> └─renderD132
> card6                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=5
> └─renderD134
> card7                    Intel Raptorlake_s (Gen12)        pci:vendor=8086,device=A780,card=6
> └─renderD135

I didn’t know that was a thing. Neat. :)

The intel_gpu_tools package was updated in Debian 13, which is the package repo for PVE 9.

You could try apt removing intel_gpu_tools and reinstalling it. If that doesn’t work, this sounds like a bug that needs to be reported the intel_gpu_tools repo.

If you end up doing that, you’ll probably want to include all the diagnostic information you grabbed earlier.